libplctag/libplctag #220: PLC-5 Comm Issues

parksma, 2020-12-23:
We have been using libplctag with ControlLogix for a while now and recently landed a project with PLC-5 processors.  I have been pulling my hair out trying to establish comms with the PLC for the past couple of weeks.  I have a PLC-5/80E with ethernet side card.  The PLC is on our office network currently while we develop the user interface.  We CAN connect via RSLogix 5, RSLinx, and from a Panelview HMI (via RSLinx) with no issues.  Has anyone else experienced issues like this?

More information:
- Developing in Win 10 64 Bit using Visual Studio 2019 Community.
- The dev environment is in a VirtualBox VM; host is also Win 10 - 64 Bit.
- We've wrapped the library in C# and can connect to ControlLogix no problem.
- I loaded the libplctag files in Visual Studio 2019 and have been working with the PLC5.c example to debug this issue.
- I have tried the current release as well as v1.5 and v2.0
- Firewall is turned off in the VM.
- There are a couple of network switches between my dev PC and the PLC; one is unmanaged and the other is connected directly to the PLC with the port set to 10 Mbs / full duplex (not auto-negotiate).

It is confusing that we can connect with the Rockwell software, but not with any version of libplctag.  I cannot imagine there is an issue with all of the versions I tried, so I am thinking there must be some external issue... maybe with the network, VM, or host PC?

Any suggestions would be greatly appreciated.

xinthose, 2020-12-23:
Can you post the connection string you are using?

parksma, 2020-12-23:
As mentioned in my post, I am using the Plc5.c example for testing.  The only change I made to the connection string is the IP address:
#define TAG_PATH "protocol=ab_eip&gateway=192.168.2.253&cpu=PLC5&elem_size=4&elem_count=5&name=F8:10&debug=4"

I forgot to mention in my original post that the detailed debug messages (debug=4) appears to indicate that socket did not connect.  Below is the entire output after running:

2020-12-23 11:27:44.217 thread(1) tag(0) INFO find_tag_create_func:99 Matched protocol=ab_eip
2020-12-23 11:27:44.219 thread(1) tag(0) INFO ab_tag_create:165 Starting.
2020-12-23 11:27:44.219 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from ab_tag_create:172
2020-12-23 11:27:44.219 thread(1) tag(0) INFO rc_alloc_impl:130 Done
2020-12-23 11:27:44.219 thread(1) tag(0) DETAIL rc_alloc_impl:135 Returning memory pointer 00000142A0341A50
2020-12-23 11:27:44.220 thread(1) tag(0) DETAIL ab_tag_create:178 tag=00000142A0341A50
2020-12-23 11:27:44.220 thread(1) tag(0) DETAIL get_plc_type:886 Found PLC/5 PLC.
2020-12-23 11:27:44.220 thread(1) tag(0) DETAIL get_plc_type:886 Found PLC/5 PLC.
2020-12-23 11:27:44.220 thread(1) tag(0) DETAIL session_find_or_create:230 Starting
2020-12-23 11:27:44.220 thread(1) tag(0) DETAIL session_find_or_create:254 Creating new session.
2020-12-23 11:27:44.220 thread(1) tag(0) INFO session_create_unsafe:446 Starting
2020-12-23 11:27:44.220 thread(1) tag(0) DETAIL session_create_unsafe:451 Session should not use connected messaging.
2020-12-23 11:27:44.220 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from session_create_unsafe:454
2020-12-23 11:27:44.220 thread(1) tag(0) INFO rc_alloc_impl:130 Done
2020-12-23 11:27:44.220 thread(1) tag(0) DETAIL rc_alloc_impl:135 Returning memory pointer 00000142A0341CC0
2020-12-23 11:27:44.221 thread(1) tag(0) DETAIL cip_encode_path:346 Starting
2020-12-23 11:27:44.221 thread(1) tag(0) DETAIL cip_encode_path:435 IOI size before 0
2020-12-23 11:27:44.221 thread(1) tag(0) DETAIL cip_encode_path:456 Done
2020-12-23 11:27:44.221 thread(1) tag(0) DETAIL add_session_unsafe:310 Starting
2020-12-23 11:27:44.221 thread(1) tag(0) DETAIL add_session_unsafe:320 Done
2020-12-23 11:27:44.221 thread(1) tag(0) INFO session_create_unsafe:553 Done
2020-12-23 11:27:44.221 thread(1) tag(0) INFO session_init:569 Starting.
2020-12-23 11:27:44.221 thread(1) tag(0) DETAIL mutex_create:632 Starting.
2020-12-23 11:27:44.221 thread(1) tag(0) DETAIL mutex_create:659 Done.
2020-12-23 11:27:44.221 thread(1) tag(0) DETAIL thread_create:790 Starting.
2020-12-23 11:27:44.222 thread(1) tag(0) DETAIL thread_create:824 Done.
2020-12-23 11:27:44.222 thread(1) tag(0) INFO session_init:584 Done.
2020-12-23 11:27:44.222 thread(1) tag(0) DETAIL session_find_or_create:300 Done
2020-12-23 11:27:44.222 thread(1) tag(0) DETAIL ab_tag_create:266 using session=00000142A0341CC0
2020-12-23 11:27:44.222 thread(1) tag(0) DETAIL ab_tag_create:272 Setting up PLC/5 tag.
2020-12-23 11:27:44.223 thread(1) tag(0) DETAIL plc5_encode_tag_name:169 Starting.
2020-12-23 11:27:44.224 thread(1) tag(0) DETAIL parse_pccc_logical_address:665 Starting.
2020-12-23 11:27:44.224 thread(1) tag(0) DETAIL parse_pccc_file_type:840 Done.
2020-12-23 11:27:44.226 thread(1) tag(0) DETAIL parse_pccc_file_num:851 Starting.
2020-12-23 11:27:44.226 thread(1) tag(0) DETAIL parse_pccc_file_num:866 Done.
2020-12-23 11:27:44.227 thread(1) tag(0) DETAIL parse_pccc_elem_num:877 Starting.
2020-12-23 11:27:44.228 thread(1) tag(0) DETAIL parse_pccc_elem_num:895 Done.
2020-12-23 11:27:44.228 thread(1) tag(0) DETAIL parse_pccc_subelem_num:906 Starting.
2020-12-23 11:27:44.229 thread(1) tag(0) DETAIL parse_pccc_subelem_num:919 No subelement in this name.
2020-12-23 11:27:44.230 thread(1) tag(0) DETAIL parse_pccc_logical_address:689 Starting.
2020-12-23 11:27:44.230 thread(1) tag(0) DETAIL plc5_encode_tag_name:212 Done.
2020-12-23 11:27:44.231 thread(1) tag(0) INFO tag_read_start:190 Starting
2020-12-23 11:27:44.232 thread(1) tag(0) DETAIL session_create_request:2334 Starting.
2020-12-23 11:27:44.232 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from session_create_request:2343
2020-12-23 11:27:44.233 thread(1) tag(0) INFO rc_alloc_impl:130 Done
2020-12-23 11:27:44.234 thread(1) tag(0) DETAIL rc_alloc_impl:135 Returning memory pointer 00000142A0334D90
2020-12-23 11:27:44.234 thread(1) tag(0) DETAIL session_create_request:2357 Done.
2020-12-23 11:27:44.235 thread(1) tag(0) INFO session_add_request:912 Starting. sess=00000142A0341CC0, req=00000142A0334D90
2020-12-23 11:27:44.236 thread(1) tag(0) DETAIL session_add_request_unsafe:877 Starting.
2020-12-23 11:27:44.237 thread(1) tag(0) DETAIL session_add_request_unsafe:896 Total requests in the queue: 1
2020-12-23 11:27:44.237 thread(1) tag(0) DETAIL session_add_request_unsafe:898 Done.
2020-12-23 11:27:44.238 thread(1) tag(0) INFO session_add_request:918 Done.
2020-12-23 11:27:44.239 thread(1) tag(0) INFO tag_read_start:304 Done.
2020-12-23 11:27:44.239 thread(1) tag(0) INFO ab_tag_create:446 Done.
2020-12-23 11:27:44.240 thread(1) tag(0) DETAIL mutex_create:632 Starting.
2020-12-23 11:27:44.246 thread(1) tag(0) DETAIL mutex_create:659 Done.
2020-12-23 11:27:44.247 thread(1) tag(0) DETAIL mutex_create:632 Starting.
2020-12-23 11:27:44.248 thread(1) tag(0) DETAIL mutex_create:659 Done.
2020-12-23 11:27:49.039 thread(2) tag(0) INFO session_handler:979 Starting thread for session 00000142A0341CC0
[... cut ...]
2020-12-23 11:28:06.342 thread(1) tag(0) INFO mb_teardown:1901 Starting.
2020-12-23 11:28:06.342 thread(1) tag(0) DETAIL mb_teardown:1905 Destroying Modbus mutex.
2020-12-23 11:28:06.343 thread(1) tag(0) INFO mb_teardown:1911 Done.
2020-12-23 11:28:06.344 thread(1) tag(0) INFO lib_teardown:152 Tearing down library.
2020-12-23 11:28:06.344 thread(1) tag(0) INFO lib_teardown:177 Done.
2020-12-23 11:28:06.345 thread(1) tag(0) DETAIL plc_tag_unregister_logger:914 Starting
2020-12-23 11:28:06.345 thread(1) tag(0) DETAIL plc_tag_unregister_logger:918 Done.

kyle-github, 2020-12-23:
Thanks for reaching out.  This sounds like a weird problem!

I just tried plc5.c with a valid IP address locally and it Works For Me(tm).   This is with the release branch which is version 2.1.21.  My network is fairly simple with a router and two switches between my laptop and the PLC.   It is a bad network, but simple.  Laptop -> WiFi router -> powerline Ethernet -> switch -> switch -> PLC/5.  Flat, single IP space.

I suspect that there is something about your firewall or the IP address is blocking the communication.   In VirtualBox (that is where I run my Windows tests too) you may want to check how the network is set up.  I use bridged networking so that the VB VM looks like a different computer on the network.   I wonder if NAT mode will cause problems with the _host_ PC's firewall?

Because of the need for raw packet support, I believe that RSLinx and similar Rockwell tooling is installed to run as Administrator or otherwise tweaks the Windows firewall.  So if the firewall is a problem, the Rockwell tools will probably still work but other things won't.

Can you ping the PLC from within a CMD windows on the VM?   Also make sure that you are using the latest release version.  If you are still on 2.0.x there are many fixes that you do not have.   Try running the plc5 test from a CMD window run as Administrator.   I would pay particular attention to both the host and guest firewalls.   Normally there is not much of a problem as the port number is really high (44818).

Let me know if any of this helps!

parksma, 2020-12-23:
Thanks for the suggestions.

I have been using the current release branch of the library with the exception of one time attempting the v1.5 and v2.0 branches to see if there was any difference.  Ping is always successful.

I have done the following based on your recommendations and testing at each action:
1. Run Plc5.exe from an elevated command prompt. 
2. Disabled all firewalls and antivirus in both the VM and host.
2. Removed the connection to the main office router from the network switch in my office.
3. Connected directly from my PC to the PLC.

None of the above had any impact on the results.

As a sanity check, I downloaded a trial version of the CimQuest InGear Net ABLink driver and installed it into my dev VM.  I ran one of the sample programs and was able to connect to the PLC-5 without issue and read tags (same addresses I was attempting to read with the Plc5 example).  I am assuming that driver uses the same underlying protocol, so I would think I can now eliminate the PLC itself or the ethernet side card as the issue.

I will keep working on it and post any additional information that might be useful.

Thanks again for the help.

kyle-github, 2020-12-24:
How old is that PLC-5?

I have a PLC-5/30E.   It has updated firmware to support the CIP protocol on port 44818.   However, older PLC-5 firmware only supported another protocol, CSP, that Allen-Bradley never publicly documented.  It was on port 2222.   The protocol is almost certainly based on the DF1 protocol (the CIP version is just a wrapper around that), which _is_ documented, but the wrapper or any other changes do not appear in any literature I have found.

Rockwell describes some aspects of PLC/5 Ethernet support in the following document:

https://literature.rockwellautomation.com/idc/groups/literature/documents/um/1785-um012_-en-p.pdf

According to that, there may be a web server running on your 1785 Ethernet module (see chapter 11).  I do not have one running on my Ethernet module.   If there is, you may be able to find out what is supported (CSP vs. CIP).

Is there any chance you can get a Wireshark dump of the CimQuest driver talking to the PLC/5?  If it is connecting on port 44818, then the library should work.   If it is connecting on a different port, it may be using a different protocol.

A PLC-5/80 should be new enough that the firmware should be able to support CIP.

I want to solve this.   I have not run into a PLC-5 that would not work before!

kyle-github, 2020-12-24:
Another thing to try is to do a port scan on the PLC-5.

I did one against mine:

```text
$> sudo nc -v -z 10.206.1.38 22-65535 | grep succeed
Connection to 10.206.1.38 2222 port [tcp/*] succeeded!
Connection to 10.206.1.38 44818 port [tcp/*] succeeded!
```

That was on Linux.  On Windows you may be able to find the netcat utility (that is what "nc" is) and use that.   At least on my system, both port 2222 (CSP) and port 44818 are open.   The library uses CIP which is on port 44818.

I did a little more research and found an open source library that claims to support CSP from about 1999!  I guess I have a little project for the holiday :-)

parksma, 2020-12-25:
I'm sure that this PLC-5 is quite old... PLC-5s were probably not manufactured after the 90's.  That said, the 5/80 is newest of the models, so I would think that it should have the firmware revisions you are referring to.  For what it's worth, the processor is a Series E, Revision J.  From what I read yesterday however, it appears to be the firmware revision of the ethernet side card rather than the PLC itself that matters.  The link you provided appears to apply to an ethernet PLC with built-in RJ45 port (not requiring a side card).  The ethernet settings are actually channel 2 for those PLCs rather than channel 3A like I have.  See the following:

[https://literature.rockwellautomation.com/idc/groups/literature/documents/in/1785-in019_-en-p.pdf](url)

The module depicted at the beginning of this document is [I believe] newer than the one I have.  It has a built-in RJ45 connection while mine has a d-shell 25-pin [or so] that requires a transceiver to convert to RJ45.  That said, I do believe that the web page you referred to is supported on the one I have, but it looks like I need to go into the office to literally cycle power on the PLC.  I was able to enable HTTP on the channel 3A settings page, but the documentation explicitly states that power must be cycle... a download is not enough.  So, I won't be able to do that until after Christmas.

I found a version of NetCat for Windows.  It was taking way too long to scan the entire range of 22-65535 ports, so I ran it twice targeting each port in question:

[image]

I was also able to load WireShark and capture the InGear communications.  I have not used WireShark much in the past, so it took some time to figure out how to capture this specific traffic.  I captured the following from a sample InGear app:

- Connect to PLC
- Read F8:10, Len 5 (values = 1.9, 2.9, 3.9, 4.9, 5.9)
- Write F8:10, Len 5 (values = 2.1, 3.2, 4.3, 5.4, 6.5)
- Disconnect from PLC

Note that I am connected to the office via a VPN, which is why my IP is 10.8.0.2 and the PLC is 192.168.2.253.  Here is the capture:

[InGear Capture.zip](https://github.com/libplctag/libplctag/files/5742250/InGear.Capture.zip)

As you will see, it appears that InGear is using port 2222 for communications.  Can we simply change the port # or is it an entirely different protocol on that port?

Merry Christmas!

kyle-github, 2020-12-26:
Merry Christmas!

I did a bit more fiddling with my PLC and... swing and a miss.   It turns out that if I connect any program to port 2222, the PLC immediately closes the connection.     So while port 2222 is open, it really is not.

I do have the Ethernet sidecar with the old transceiver module as well.   The trasceiver falls off at least once a week when I walk by and snag the Ethernet cable :frowning_face:.

The protocols are similar, but not the same.   There is a protocol, DF1 (often called PCCC), that is the core.   That is either encapsulated for transmission over serial lines (DH+), Ethernet (CSP), or CIP (over Ethernet or ControlNet for two levels of encapsulation).   The latter is what the library currently supports over Ethernet.      But that encapsulation includes handshakes and headers that are unique for each lower level protocol.   Sadly that means that this is not a matter of changing the port number.   AB seems to go out of their way to make this hard.

I found a library by Ron Gage from the late 1990s that supports CSP (port 2222).  I tried it against my PLC and got the same result as above.  I grabbed the packets with Wireshark and can see that the PLC simply sends a FIN (close) immediately after it receives an attempted TCP open.  I found a forum thread on MrPLC from 2010 by Ken Roach that says that this is how RSLinx determines whether to use the older CSP or the newer PCCC-over-CIP encapsulation.   PLC-5 systems (and possibly SLC 500?) that were updated apparently leave port 2222 open, but immediately close all attempts to connect on it to signal that the EthernetIP port 44818 should be used instead.

Unfortunately, that means that I do not have an easy way to test this.  I was going to take Ron's library, add in some packet dumps and watch it bring up and down connections.  The protocol to read and write tags is the same, I just need the handshake information.

Thanks for providing the packet dump.  I might be able to get some traction using that.   I am trying to figure out how I can get a test system.  Only certain models and revisions of PLC-5 and SLC 500 supported CSP.   Most either never supported Ethernet at all or were updated to support PCCC-over-CIP (EthernetIP) later.  I do not know if AB even provides upgrade kits any more.

parksma, 2020-12-30:
Hi Kyle... sorry for the delayed response.

I have to bite the bullet and purchase InGear for this PLC-5 project since I don't know what protocol will be supported on the plant PLC / side-card.  That said, we would prefer to continue using libplctag for Logix PLCs, and for PLC-5s / SLCs as needed in the future.  I would love to be part of the upgrade to support both the 2222 and 44818 ports.  I'm mostly a .NET GUI guy, so the best way I can help is to get you access to my PLC somehow.  I can probably setup a laptop connected to the PLC-5 on our guest wifi and provide TeamViewer remote control if you could work with that.  If you have another peer-to-peer VPN suggestion that would allow you to work from your hardware, then let me know.  I probably won't be able to do this until sometime in January though.

kyle-github, 2020-12-30:
Hi Mike,

No problem. You need to get your system working.  That is the highest priority!

Unfortunately, I have not been able to make my PLC-5 respond on port 2222.   I have some code from Ron Gage from 1998 or so that will talk with PLCs on that port, so I have a protocol model, but I do not have a PLC that will speak that protocol.

I will keep trying to find a way to do this.   The packet dumps are a large step.   The problem with eBay is finding a PLC that has Ethernet, but is not completely upgraded to Ethernet/IP-CIP!   I have reached out to some other contacts to see if anyone has an old PLC sitting around in the shop.

kyle-github, 2021-01-08:
I am going to rename this one to note the needed support for CSP and call it an enhancement request.