libplctag/libplctag #221: Support for ISO connection

alireza4474, 2021-01-02:
Does anybody know how to communicate with S5 PLCs via ISO connection. I ve searched and found a lot of libraries and sources like this, but all of them were communicating the PLC via tcp protocol but I need to use ISO protocol.

kyle-github, 2021-01-02:
Hi, ISO still needs to go over something physical.  From Wikipedia, the S5 PLCs are fairly old.  What standard networking do they normally have?  

If you are looking for ISO over TCP, the [Snap7](http://snap7.sourceforge.net/) library does this for many models of Siemens PLC.  It is widely used and has been out for many years.  The library sets up an ISO connection over TCP and then communicates using that.  

I hope that helps!

alireza4474, 2021-01-03:
> Hi, ISO still needs to go over something physical. From Wikipedia, the S5 PLCs are fairly old. What standard networking do they normally have?
>
> If you are looking for ISO over TCP, the [...]

alireza4474, 2021-01-03:
> Hi, ISO still needs to go over something physical. From Wikipedia, the S5 PLCs are fairly old. What standard networking do they normally have?
>
> If you are looking for ISO over TCP, the [...]

First of all thank you for your answer
Second sorry that I closed the comment. I am new in github and dont know what to do.

As you said ISO is so old and I have to say I am doing a project to communicate with S5 PLCs which are using cp 1430 tf that support ISO connection and I want to how it is possible. I need to communicate with it however it is possible.
Thank you

kyle-github, 2021-01-04:
I think perhaps I was not clear.   ISO is a specification of a networking protocol stack.   It does not specify how the bits actually get transmitted.   It is just an alternate to things like TCP.   You can run an ISO protocol stack over a serial port, a radio, or a couple of cans and string :-)

There is a RFC to do ISO over TCP.  That is what the S7 Siemen's PLC's use.   In that case, it is generally ISO over TCP over Ethernet where Ethernet is the lowest level (and even that has multiple types of hardware).  The Snap7 library support that for many models of Siemens PLC.

Here is a place to start with ISO over TCP: [RFC 1006](https://tools.ietf.org/html/rfc1006)

I found a video of someone using the same 1430 Ethernet module that you noted: https://www.youtube.com/watch?v=y-bBY-ZbMt4.

Interestingly it looks like the data block is simply transmitted without any formatting at all.  Just a TCP connection and then dumping the data.   That is quite low level!

I would reach out to the author of the Snap7 package.   He may have more insight.  I have never used a Siemens PLC, so it is all just guesswork on my part!

alireza4474, 2021-01-04:
Your answer was so helpful. I will read all the links you send and watch the video and inform you the result. I thank you once more

alireza4474, 2021-01-04:
Hello again

alireza4474, 2021-01-04:
Hello again
I have read the documents and watched the video but I think there is a misunderstanding.
"ISO" connection is not "ISO over TCP". There is no IP in ISO connection and to addressing other devices you have to use their mac address. So this connection is not routable and I can not use socket programming or arp or anything that use with IP.
It is my problem that made me to think about creating a network protocol.

kyle-github, 2021-01-15:
Have you had any luck on this?

kyle-github, 2021-02-20:
Please reopen if you have additional problems.

closed 2021-02-20 by kyle-github