libplctag/libplctag #222: Reading from Logix Based Alarms

darko-liberati, 2021-01-05:
I have ran into an issue where I am trying to read logix based alarms from a control logix plc.
The format of the tag is as follows:
motor_current_left@Alarms.Motor_Current_Left_LO.Severity

When I try to read this tag (datatype dint) using the library I just get a -32 as a result.

Trying to figure out if I need to do something different or if the libplctag api doesn't handle this type of requests yet.

Thanks,

kyle-github, 2021-01-06:
Hi,

Thanks for reaching out.   I have not tried to access alarms yet.   The format of the tag above will definitely not work as the "@" symbol is not considered a valid character for a tag name (apart from the special @tags name) in the library.

I cannot find out much about these tags.  I think they must be implemented as tags.

Can you run the tag listing program from the examples against a PLC with some of these alarms and point out if any of the alarms show up in the listing?   Perhaps if I see what the PLC thinks they look like I can tweak the library to work with them.

I do not have any PLCs that use tag-based alarms and I have never used them, so unfortunately I have no examples.   I cannot find a lot of documentation on them online.   

Do you have any good sources of information?

Thanks for bringing this up!

GitHubDragonFly, 2021-01-06:
There is some info on tag-based alarms here:

https://literature.rockwellautomation.com/idc/groups/literature/documents/rm/ftae-rm001_-en-e.pdf

It does show this tip on page 33:

"Logix tag-based alarms are supported only on Compact GuardLogix 5380, CompactLogix 5380, CompactLogix 5480, ControlLogix 5580, and GuardLogix 5580 controller."

GitHubDragonFly, 2021-01-06:
Besides for the mentioned libplctag tag listing program, there is another free tag browser available at the AdvancedHMI forum website (free registration is required to access it). See this message:

https://www.advancedhmi.com/forum/index.php?topic=2670.msg16843#msg16843

It is currently in the form of a standalone Windows package which should provide a little bit more UDT related info.

GitHubDragonFly, 2021-01-06:
Tag alarm syntax is also shown on page 46 here:

https://www.reynoldsonline.com/ASSETS/DOCUMENTS/CMS/EN/Studio+5000+Modular+%26+Advanced+Programming+Techniques.pdf

darko-liberati, 2021-01-06:
So I was able to finally get a value back although it is incorrect. When trying to read from one of the alarms I get a result of -4 when using client.ReadTag(). I looked at the function but not really sure what that error code (assuming it is one) means.

Also the format of the alarm is TagWithAlarmDefined.@Alarms.AlarmName.AlarmAttribute

kyle-github, 2021-01-06:
Hi, thanks for getting back so quickly.

Thanks to @GitHubDragonFly, I was able to make an Alarm in my L5581.   Didn't even know that existed!   After a good hour of trying different things, I realized that I needed to mark the alarm as used.   Sigh.

I made a DINT called AlarmLevelTest.   I created an alarm on it called AlarmTest that was a HI alarm triggered at >= 42.

First write the tag to 40.

```text
kyle@kylesdell:~/Documents/Projects/libplctag/libplctag-org/libplctag/build/bin_dist$ ./tag_rw2 --type=sint32 '--tag=protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=controllogix&name=AlarmLevelTest' --write=40
Library version 2.2.0.
Processing argument 0 "./tag_rw2".
Warning: Unknown argument "./tag_rw2"!
Processing argument 1 "--type=sint32".
Processing argument 2 "--tag=protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=controllogix&name=AlarmLevelTest".
Processing argument 3 "--write=40".
Number of write args 1.
data[0]=43 (2b)
New values written to tag.
data[0]=40 (28)
```

Now check the alarm.

```text
kyle@kylesdell:~/Documents/Projects/libplctag/libplctag-org/libplctag/build/bin_dist$ ./tag_rw2 --type=bit '--tag=protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=controllogix&name=AlarmLevelTest.@Alarms.AlarmTest.InAlarm'
Library version 2.2.0.
Processing argument 0 "./tag_rw2".
Warning: Unknown argument "./tag_rw2"!
Processing argument 1 "--type=bit".
Processing argument 2 "--tag=protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=controllogix&name=AlarmLevelTest.@Alarms.AlarmTest.InAlarm".
data=0
```

OK.   The alarm is off.   That is correct.

Now change the input value to 42.

```text
kyle@kylesdell:~/Documents/Projects/libplctag/libplctag-org/libplctag/build/bin_dist$ ./tag_rw2 --type=sint32 '--tag=protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=controllogix&name=AlarmLevelTest' --write=42
Library version 2.2.0.
Processing argument 0 "./tag_rw2".
Warning: Unknown argument "./tag_rw2"!
Processing argument 1 "--type=sint32".
Processing argument 2 "--tag=protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=controllogix&name=AlarmLevelTest".
Processing argument 3 "--write=42".
Number of write args 1.
data[0]=40 (28)
New values written to tag.
data[0]=42 (2a)
```
What is the alarm now?

```text
kyle@kylesdell:~/Documents/Projects/libplctag/libplctag-org/libplctag/build/bin_dist$ ./tag_rw2 --type=bit '--tag=protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=controllogix&name=AlarmLevelTest.@Alarms.AlarmTest.InAlarm'
Library version 2.2.0.
Processing argument 0 "./tag_rw2".
Warning: Unknown argument "./tag_rw2"!
Processing argument 1 "--type=bit".
Processing argument 2 "--tag=protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=controllogix&name=AlarmLevelTest.@Alarms.AlarmTest.InAlarm".
data=1
```

Bingo. `data=1`

I will make a release with this as soon as possible.   It might be in an hour or two and it might be tomorrow as work is crazy and we have family staying with us right now.

The alarms are some sort of UDT.  If you get rid of the `InAlarm` part and print it out as uint8, you can see that there is quite a bit of data there.  108 bytes.  When I trigger the alarm and turn it off, several bytes change, so I am not sure where the `InAlarm` flag is in that UDT.

kyle-github, 2021-01-06:
BTW, it looks like you are using a wrapper around the C core library.   Which one?   Unless you are comfortable updating the core library independently, you may need to wait until the wrapper takes up the new core library.

darko-liberati, 2021-01-07:
I am using the c# wrapper.

Wow, that's awesome man. Really appreciate you looking into this. Let me know if there is anything I can do to help.

commit 4877647a (2021-01-08): Support alarms on tags.  Fix #222.
closed 2021-01-08 by kyle-github