libplctag/libplctag #223: MicroLogix PID Question

GitHubDragonFly, 2021-01-06:
How does the library itself treat the PID addresses PD10:0.0 or PD10:0.CVH?

Just for clarification, requesting up to 23 words of the address PD10:0 does provide all the values (elem_size=2 and elem_count=(any value from 1 to 23)). This approach does allow reading bits as well. 

Would elem_size need to be specified as 46, with elem_count=1, in case of PID so to read it all in and then parse?

GitHubDragonFly, 2021-01-08:
It looks like the current solution could be reading all 46 bytes (elem_size=2, elem_count=23), parsing, changing and writing them all back.

closed 2021-01-08 by GitHubDragonFly

kyle-github, 2021-01-08:
The map for the elements I see in the DF1 doc is:

PD Fields
0, 1 control
2 SP
4 Kp
6 Ki
8 Kd
26 PV

The document says that all fields are floats, so 4 bytes each.   But word 26 means 52 bytes at least and since PV is a float, it is two words, so we'd have 54 bytes total.   Are the docs wrong for MicroLogix?

Slightly confused here...

kyle-github, 2021-01-08:
When a tag like a PID or a string is ready, the library reads all the bytes.  Unless you have specified a specific field.   Only some fields can be specified and it depends on the type.  The underlying protocol will refuse unsupported/unnamed fields.

With something like a PID, it pays to be careful because when you write it back, you will be overwriting a bunch of internal state while the PID might be live.

kyle-github, 2021-01-08:
I just looked at the library code, I should be converting `Kp` etc. to the correct offsets.

GitHubDragonFly, 2021-01-08:
The attached picture shows what these look like for MicroLogix 1100.

[image]

RSLogix help file has a lot of info for the instruction itself and parameters for different models of MicroLogix.
This online manual also has PID covered starting at page 279:

https://literature.rockwellautomation.com/idc/groups/literature/documents/rm/1763-rm001_-en-p.pdf

One needs to be careful when writing anything back to the PLC.

GitHubDragonFly, 2021-01-09:
PLC5 seems to be rather tricky for PID, since it can use either Integer Control Block (23 words) or PD file (44 words).
That's what I understood from their 1785-rm001_-en-p manual.

I have included PID in the android PhoneTest app but it will show for MicroLogix only.

kyle-github, 2021-01-11:
I was able to find something in the docs and it does say 88 bytes for PID on PLC-5.   Yikes.   I had not considered that these could be different sizes.  I will need to poke at my PLC-5 and see what I can do.   Unfortunately, I am having problems getting RSLogix 5 to start (the VM got upgraded while I was not paying attention).

I could make the user tell me the type somehow since that is not something that is simple to introspect.   Hmm.  I am really going to have to think about how to handle this.  I guess elem_size is enough to tell, but I am trying to make that optional across the board.