libplctag/libplctag #224: communicate with the compactlogix l33er

leowoki, 2021-01-13:
Is it possible to communicate with the compactlogix l33er?

kyle-github, 2021-01-13:
If the tags are not protected, you should be fine.

kyle-github, 2021-01-15:
Oops, rereading that, you might need a path of "1,0".   I do not have a CompactLogix handy but I think they have a static path.   You can leave out the elem_size if you want, it is optional.

leowoki, 2021-01-16:
> Oops, rereading that, you might need a path of "1,0". I do not have a CompactLogix handy but I think they have a static path. You can leave out the elem_size if you want, it is optional.

Thank you. I will do the test next week and will send the results in the meantime.

kyle-github, 2021-01-24:
Were you able to test this?

leowoki, 2021-02-01:
> Were you able to test this?

Sorry, I was delayed by other things for a long time, but I finally got a chance to test it and it can communicate with the compactlogix l33er.

kyle-github, 2021-02-01:
Good to hear that it is working for you.   Please reopen this if there are other problems.  Thanks for reaching out!

closed 2021-02-01 by kyle-github