libplctag/libplctag #225: Problem listing tags, debug return "Port Not Available"

elbaghee, 2021-01-21:
Hello,

I'm trying to list tags in CLX L61, but I always have the same result in debug "port not available", however, if I read a single tag it works fine (with the same IP and port of course).

Here a little part of debugs
**Listing Tags:**
`2021-01-21 16:01:57.668 thread(3) tag(0) INFO session_handler:980 Starting thread for session 07FAC498
2021-01-21 16:01:57.684 thread(3) tag(0) DETAIL session_handler:999 in SESSION_OPEN_SOCKET state.
2021-01-21 16:01:57.684 thread(3) tag(0) INFO session_open_socket:602 Starting.
2021-01-21 16:01:57.684 thread(3) tag(0) DETAIL socket_create:1014 Starting.
2021-01-21 16:01:57.684 thread(1) tag(0) DETAIL setup_tag_listing:2393 Starting.
2021-01-21 16:01:57.684 thread(1) tag(0) DETAIL setup_tag_listing:2398 Tag is a controller tag listing request.
2021-01-21 16:01:57.684 thread(1) tag(0) DETAIL setup_tag_listing:2500 Done.
2021-01-21 16:01:57.684 thread(1) tag(0) DETAIL get_tag_data_type:614 Done.
2021-01-21 16:01:57.684 thread(1) tag(0) INFO tag_read_start:239 Starting
2021-01-21 16:01:57.684 thread(1) tag(0) INFO build_tag_list_request_connected:461 Starting.
2021-01-21 16:01:57.684 thread(1) tag(0) DETAIL session_create_request:2335 Starting.
2021-01-21 16:01:57.684 thread(3) tag(0) DETAIL socket_create:1033 Done.
2021-01-21 16:01:57.699 thread(3) tag(0) DETAIL session_open_socket:637 Using default port 44818.
2021-01-21 16:01:57.699 thread(3) tag(0) DETAIL socket_connect_tcp:1053 Starting.
2021-01-21 16:01:57.699 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from session_create_request:2344
2021-01-21 16:01:57.699 thread(1) tag(0) INFO rc_alloc_impl:130 Done
2021-01-21 16:01:57.699 thread(1) tag(0) DETAIL rc_alloc_impl:135 Returning memory pointer 02DF4BC8
2021-01-21 16:01:57.699 thread(1) tag(0) DETAIL session_create_request:2358 Done.
2021-01-21 16:01:57.699 thread(1) tag(0) INFO session_add_request:913 Starting. sess=07FAC498, req=02DF4BC8
2021-01-21 16:01:57.699 thread(1) tag(0) DETAIL session_add_request_unsafe:878 Starting.
2021-01-21 16:01:57.699 thread(1) tag(0) DETAIL session_add_request_unsafe:897 Total requests in the queue: 1
2021-01-21 16:01:57.699 thread(1) tag(0) DETAIL session_add_request_unsafe:899 Done.
2021-01-21 16:01:57.699 thread(1) tag(0) INFO session_add_request:919 Done.
2021-01-21 16:01:57.699 thread(1) tag(0) INFO build_tag_list_request_connected:581 Done
2021-01-21 16:01:57.699 thread(1) tag(0) INFO tag_read_start:268 Done.
2021-01-21 16:01:57.699 thread(1) tag(0) INFO ab_tag_create:446 Done.
2021-01-21 16:01:57.699 thread(1) tag(0) DETAIL mutex_create:633 Starting.
2021-01-21 16:01:57.699 thread(3) tag(0) DETAIL socket_connect_tcp:1102 Found numeric IP address: 192.120.7.63
2021-01-21 16:01:57.699 thread(1) tag(0) DETAIL mutex_create:660 Done.
2021-01-21 16:01:57.714 thread(1) tag(0) DETAIL mutex_create:633 Starting.
2021-01-21 16:01:57.714 thread(1) tag(0) DETAIL mutex_create:660 Done.
2021-01-21 16:01:57.731 thread(3) tag(0) DETAIL socket_connect_tcp:1194 Done.
2021-01-21 16:01:57.731 thread(3) tag(0) INFO session_open_socket:652 Done.
2021-01-21 16:01:57.731 thread(3) tag(0) DETAIL session_handler:1016 in SESSION_REGISTER state.
2021-01-21 16:01:57.731 thread(3) tag(0) INFO session_register:665 Starting.
2021-01-21 16:01:57.731 thread(3) tag(0) INFO send_eip_request:1764 Starting.
2021-01-21 16:01:57.731 thread(3) tag(0) INFO send_eip_request:1777 Sending packet of size 28
2021-01-21 16:01:57.731 thread(3) tag(0) INFO send_eip_request:1778  65 00 04 00 00 00 00 00 00 00
2021-01-21 16:01:57.731 thread(3) tag(0) INFO send_eip_request:1778  00 00 00 00 00 00 00 00 00 00
2021-01-21 16:01:57.731 thread(3) tag(0) INFO send_eip_request:1778  00 00 00 00 01 00 00 00
2021-01-21 16:01:57.731 thread(3) tag(0) INFO send_eip_request:1812 Done.
2021-01-21 16:01:57.731 thread(3) tag(0) INFO recv_eip_response:1832 Starting.
2021-01-21 16:01:57.793 thread(3) tag(0) INFO recv_eip_response:1896 request received all needed data (28 bytes of 28).
2021-01-21 16:01:57.793 thread(3) tag(0) INFO recv_eip_response:1898  65 00 04 00 00 15 02 12 00 00
2021-01-21 16:01:57.793 thread(3) tag(0) INFO recv_eip_response:1898  00 00 00 00 00 00 00 00 00 00
2021-01-21 16:01:57.793 thread(3) tag(0) INFO recv_eip_response:1898  00 00 00 00 01 00 00 00
2021-01-21 16:01:57.793 thread(3) tag(0) INFO recv_eip_response:1905 Done.
2021-01-21 16:01:57.793 thread(3) tag(0) INFO session_register:732 Done.
2021-01-21 16:01:57.793 thread(3) tag(0) DETAIL session_handler:1031 in SESSION_SEND_FORWARD_OPEN state.
2021-01-21 16:01:57.793 thread(3) tag(0) INFO send_forward_open_request:1943 Starting
2021-01-21 16:01:57.793 thread(3) tag(0) DETAIL send_forward_open_request:1953 Setting initial payload size guess to 4002.
2021-01-21 16:01:57.793 thread(3) tag(0) INFO send_extended_forward_open_request:2069 Starting
2021-01-21 16:01:57.793 thread(3) tag(0) INFO send_eip_request:1764 Starting.
2021-01-21 16:01:57.793 thread(3) tag(0) INFO send_eip_request:1777 Sending packet of size 92
2021-01-21 16:01:57.793 thread(3) tag(0) INFO send_eip_request:1778  6f 00 44 00 00 15 02 12 00 00
2021-01-21 16:01:57.793 thread(3) tag(0) INFO send_eip_request:1778  00 00 55 62 00 00 00 00 00 00
2021-01-21 16:01:57.793 thread(3) tag(0) INFO send_eip_request:1778  00 00 00 00 00 00 00 00 01 00
2021-01-21 16:01:57.793 thread(3) tag(0) INFO send_eip_request:1778  02 00 00 00 00 00 b2 00 34 00
2021-01-21 16:01:57.793 thread(3) tag(0) INFO send_eip_request:1778  5b 02 20 06 24 01 0a 05 00 00
2021-01-21 16:01:57.793 thread(3) tag(0) INFO send_eip_request:1778  00 00 78 1d 00 00 af 46 3d f3
2021-01-21 16:01:57.793 thread(3) tag(0) INFO send_eip_request:1778  45 43 50 21 01 00 00 00 40 42
2021-01-21 16:01:57.793 thread(3) tag(0) INFO send_eip_request:1778  0f 00 a2 0f 00 42 40 42 0f 00
2021-01-21 16:01:57.793 thread(3) tag(0) INFO send_eip_request:1778  a2 0f 00 42 a3 03 00 20 02 24
2021-01-21 16:01:57.793 thread(3) tag(0) INFO send_eip_request:1778  01 00
[... 147 lines cut ...]
2021-01-21 15:47:16.340 thread(3) tag(0) INFO send_eip_request:1778  0f 00 fc 43 40 42 0f 00 fc 43
2021-01-21 15:47:16.340 thread(3) tag(0) INFO send_eip_request:1778  a3 03 01 00 20 02 24 01
2021-01-21 15:47:16.340 thread(3) tag(0) INFO send_eip_request:1812 Done.
2021-01-21 15:47:16.340 thread(3) tag(0) INFO send_old_forward_open_request:2056 Done
2021-01-21 15:47:16.340 thread(3) tag(0) INFO send_forward_open_request:1970 Done
2021-01-21 15:47:16.340 thread(3) tag(0) DETAIL session_handler:1037 Send Forward Open succeeded, going to SESSION_RECEIVE_FORWARD_OPEN state.
2021-01-21 15:47:16.340 thread(3) tag(0) DETAIL session_handler:1043 in SESSION_RECEIVE_FORWARD_OPEN state.
2021-01-21 15:47:16.340 thread(3) tag(0) INFO receive_forward_open_response:2141 Starting
2021-01-21 15:47:16.340 thread(3) tag(0) INFO recv_eip_response:1832 Starting.
2021-01-21 15:47:16.402 thread(3) tag(0) INFO recv_eip_response:1896 request received all needed data (70 bytes of 70).
2021-01-21 15:47:16.402 thread(3) tag(0) INFO recv_eip_response:1898  6f 00 2e 00 00 14 02 12 00 00
2021-01-21 15:47:16.402 thread(3) tag(0) INFO recv_eip_response:1898  00 00 5b 70 00 00 00 00 00 00
2021-01-21 15:47:16.402 thread(3) tag(0) INFO recv_eip_response:1898  00 00 00 00 00 00 00 00 01 00
2021-01-21 15:47:16.402 thread(3) tag(0) INFO recv_eip_response:1898  02 00 00 00 00 00 b2 00 1e 00
2021-01-21 15:47:16.402 thread(3) tag(0) INFO recv_eip_response:1898  d4 00 00 00 02 21 51 00 e4 6d
2021-01-21 15:47:16.402 thread(3) tag(0) INFO recv_eip_response:1898  00 00 5c 57 3d f3 45 43 50 21
2021-01-21 15:47:16.402 thread(3) tag(0) INFO recv_eip_response:1898  40 42 0f 00 40 42 0f 00 00 00
2021-01-21 15:47:16.402 thread(3) tag(0) INFO recv_eip_response:1905 Done.
2021-01-21 15:47:16.402 thread(3) tag(0) INFO receive_forward_open_response:2203 ForwardOpen succeeded with our connection ID 6de4 and the PLC connection ID 512102 with packet size 508.
2021-01-21 15:47:16.402 thread(3) tag(0) INFO receive_forward_open_response:2208 Done.
2021-01-21 15:47:16.402 thread(3) tag(0) DETAIL session_handler:1062 Send Forward Open succeeded, going to SESSION_IDLE state.`

Can someone help me? I don't know why it says that the default port is not available.

Thanks in advance, and sorry for my English.

GitHubDragonFly, 2021-01-22:
While waiting for kyle-github to decipher all the info you provided, besides for the libplctag tag listing program there is another free tag browser available at the AdvancedHMI forum website (free registration is required to access it). See this message:

https://www.advancedhmi.com/forum/index.php?topic=2670.msg16843#msg16843

It is currently in the form of a standalone Windows package which should provide a little bit more UDT related info.

If this tag browser cannot access your tags either then it is either that something is blocking that service in your PLC or the PLC just doesn't support that service (your output shows this line: "Forward Open command failed, response code: Unsupported service").

kyle-github, 2021-01-22:
Hi, thanks for bringing this up.

Are you using the same path for both the tag listing and tag read attempts?   When working with a ControlLogix system, you need to have an IP address for the network module and a path.   A path is a set of elements describing the path through the system to the CPU module.   For instance, if your CPU is in slot 0, the path would be "1,0".

When listing tags you need to supply the path as well as a second argument.   Suppose you have a L61 CPU module in slot 0 of your ControlLogix chassis.   Suppose your PLC is at address 192.168.42.21.   You would list tags with the following:

```text
> ./list_tags '192.168.42.21'  '1,0'
```

(assuming you are using a Linux-like operating system, if using Windows you will want to use double quotes)

The error returned is:

```text
This error occurs when a port number specified in a port segment is not available. Consult the documentation of the device to verify the correct port number.
```

This usually means that the path is incorrect.   Looking at the attempted Forward Open packet, it looks like something is malformed as the path gets turned into the following:

`03 00 20 02 24 01 00`

The first two bytes are a count of 16-bit words in the path.   The part with `20 02 24 01` is a CIP object reference, specifically to the Message Router, if I remember correctly.   Normally I would expect the encoded path to be something like this:

`03 01 00 20 02 24 01`

The `01 00` part corresponds to the path I described above. 

In the working example output, the last part of the forward open request is indeed `03 01 00 20 02 24 01`.

Please double check your path on the tag listing attempt and make sure that it has "1,0" as the last argument (as a single string!).

Let me know if that helps!

elbaghee, 2021-01-22:
> 
> 
> Hi, thanks for bringing this up.
> 
> Are you using the same path for both the tag listing and tag read attempts? When working with a ControlLogix system, you need to have an IP address for the network module and a path. A path is a set of elements describing the path through the system to the CPU module. For instance, if your CPU is in slot 0, the path would be "1,0".
> 
> When listing tags you need to supply the path as well as a second argument. Suppose you have a L61 CPU module in slot 0 of your ControlLogix chassis. Suppose your PLC is at address 192.168.42.21. You would list tags with the following:
> 
> ```
> > ./list_tags '192.168.42.21'  '1,0'
> ```
> 
> (assuming you are using a Linux-like operating system, if using Windows you will want to use double quotes)
> 
> The error returned is:
> 
> ```
> This error occurs when a port number specified in a port segment is not available. Consult the documentation of the device to verify the correct port number.
> ```
> 
> This usually means that the path is incorrect. Looking at the attempted Forward Open packet, it looks like something is malformed as the path gets turned into the following:
> 
> `03 00 20 02 24 01 00`
> 
> The first two bytes are a count of 16-bit words in the path. The part with `20 02 24 01` is a CIP object reference, specifically to the Message Router, if I remember correctly. Normally I would expect the encoded path to be something like this:
> 
> `03 01 00 20 02 24 01`
> 
> The `01 00` part corresponds to the path I described above.
> 
> In the working example output, the last part of the forward open request is indeed `03 01 00 20 02 24 01`.
> 
> Please double check your path on the tag listing attempt and make sure that it has "1,0" as the last argument (as a single string!).
> 
> Let me know if that helps!

Thanks for your quick answer!!
You are right, I had a mistake in listing Tags' path and only write the slot 0 and not 1,0 like in the other process, I'm very stupid. Really I spent a lot of time looking for the problem before open this issue, and in the end, the solution is really easy.
Sorry for disturbing you

kyle-github, 2021-01-22:
The only bad question is one that is not asked!

In the future, can you please try to make sure that the debugging output is formatted?   It is very hard to read this way!

I see this error:

```text
CIP read failed with status: 0x4 IOI could not be deciphered or tag does not exist. [snip] The path segment identifier or the segment syntax was not understood by the target device.
```

This indicates that something about the tag is not matching what the PLC expects.

Are you trying to read the tags yourself?   If you are, please be aware that the `list_tags.c` program already does the work of getting program tags.    However, if the tags are protected they will not be visible.   Also note that some parts of the fake tag name are case-sensitive.

Can you share the command lines you are using?

elbaghee, 2021-01-22:
> 
> 
> The only bad question is one that is not asked!
> 
> In the future, can you please try to make sure that the debugging output is formatted? It is very hard to read this way!
> 
> I see this error:
> 
> ```
> CIP read failed with status: 0x4 IOI could not be deciphered or tag does not exist. [snip] The path segment identifier or the segment syntax was not understood by the target device.
> ```
> 
> This indicates that something about the tag is not matching what the PLC expects.
> 
> Are you trying to read the tags yourself? If you are, please be aware that the `list_tags.c` program already does the work of getting program tags. However, if the tags are protected they will not be visible. Also note that some parts of the fake tag name are case-sensitive.
> 
> Can you share the command lines you are using?

Hello again, sorry for copy debugging output without format, I only copied and pasted it, in the future, I will be more careful.

I'm using the `list_tag.c` program to list tags (not reading them by myself), I know that debugging says that the tag doesn't exist, but I don't write any tag name, I want to list all tags in PLC, the library founds the different programs in the main task, but in all of them return the same and list nothing, it only list controller tags.

The PLC is a non-Safety PLC, it's completely empty, it only has some tags in Controler and 2 programs (Mainprogram and Prog1) completely empty and only with 2 tags inside each one to test library.

Sorry but I don't understand what you want in your last question. I'm using your library in C++ Builder IDE and I'm calling your functions from my Form...

kyle-github, 2021-01-23:
I tested the current version (2.1.22) against a L55, L61 and L81.   I get the following kind of output:

```text
libplctag/build/bin_dist$ ./list_tags 10.206.1.40 1,4
Getting controller tags.
Getting tags for program: Program:MainProgram.
Tag "Program:MainProgram.one_second_pulse" (00c1): protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=ControlLogix&elem_size=1&elem_count=1&name=Program:MainProgram.one_second_pulse
Tag "Program:MainProgram.Count" (8f83): protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=ControlLogix&elem_size=12&elem_count=1&name=Program:MainProgram.Count
Tag "AlarmLevelTest" (00c4): protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=ControlLogix&elem_size=4&elem_count=1&name=AlarmLevelTest
Tag "TestUDTMultiLevel2[3]" (aac7): protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=ControlLogix&elem_size=600&elem_count=3&name=TestUDTMultiLevel2
Tag "TestUDTMultiLevel1[3]" (a314): protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=ControlLogix&elem_size=56&elem_count=3&name=TestUDTMultiLevel1
Tag "TestUDTMultiField[3]" (a685): protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=ControlLogix&elem_size=16&elem_count=3&name=TestUDTMultiField
Tag "TestUDTBoolArray[3]" (a86a): protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=ControlLogix&elem_size=4&elem_count=3&name=TestUDTBoolArray
Tag "TestLINTArray[10]" (20c5): protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=ControlLogix&elem_size=8&elem_count=10&name=TestLINTArray
Tag "TestMultiDimArray[10,10,10]" (60c4): protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=ControlLogix&elem_size=4&elem_count=1000&name=TestMultiDimArray
Tag "stop_barcodes" (00c1): protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=ControlLogix&elem_size=1&elem_count=1&name=stop_barcodes
Tag "one_second_timer" (8f83): protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=ControlLogix&elem_size=12&elem_count=1&name=one_second_timer
Tag "barcode_index" (8f82): protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=ControlLogix&elem_size=12&elem_count=1&name=barcode_index
Tag "barcodes[11]" (afce): protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=ControlLogix&elem_size=88&elem_count=11&name=barcodes
Tag "barcode" (8fce): protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=ControlLogix&elem_size=88&elem_count=1&name=barcode
Tag "barcode_processed" (00c1): protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=ControlLogix&elem_size=1&elem_count=1&name=barcode_processed
Tag "new_barcode" (00c1): protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=ControlLogix&elem_size=1&elem_count=1&name=new_barcode
Tag "TestSINTArray[1000]" (20c2): protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=ControlLogix&elem_size=1&elem_count=1000&name=TestSINTArray
Tag "TestSSTRING" (8fce): protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=ControlLogix&elem_size=88&elem_count=1&name=TestSSTRING
Tag "TestINTArray[1000]" (20c3): protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=ControlLogix&elem_size=2&elem_count=1000&name=TestINTArray
Tag "TestBOOL" (00c1): protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=ControlLogix&elem_size=1&elem_count=1&name=TestBOOL
Tag "TestBOOLArray[4]" (20d3): protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=ControlLogix&elem_size=4&elem_count=4&name=TestBOOLArray
Tag "TestBigArray[1000]" (20c4): protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=ControlLogix&elem_size=4&elem_count=1000&name=TestBigArray
Tag "TestDINTArray[10]" (20c4): protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=ControlLogix&elem_size=4&elem_count=10&name=TestDINTArray
Tag "__CONTAINER" (00c4): protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=ControlLogix&elem_size=4&elem_count=1&name=__CONTAINER
```

I apparently do not have any PLCs with more than a single program.  You can see that the listing above shows tags in `MainProgram` though.

Can you show the command line used and the output and the rest of the output?   It is almost as if there is a tag that is causing the program to think it is a program name and then we get an error.   The program looks for tags that start with `Program:`.

Can you build the program yourself?  If you can, please uncomment the printout at about line 185 then change it to 

```c
            printf("Found program: %s\n", tag_name);
```

This will print out as the program runs and I can see what it is finding.  I am suspicious that there is tag that it thinks is a program but is not.

elbaghee, 2021-01-23:
Hi, how I said before, I am not using the command line to execute this library, I'm calling its functions from a Form code.
[image]

These are the outputs for list_tags (ascending order):

> 
	23/01/2021 3:20:23	" (00c3): protocol=ab-eip&gateway=192.120.7.63&path=1,0&plc=ControlLogix&elem_size=2&elem_count=1&name=INT

	23/01/2021 3:20:23	Tag "INT
	23/01/2021 3:20:23	" (00c2): protocol=ab-eip&gateway=192.120.7.63&path=1,0&plc=ControlLogix&elem_size=1&elem_count=1&name=SINT

	23/01/2021 3:20:23	Tag "SINT
	23/01/2021 3:20:23	" (8920): protocol=ab-eip&gateway=192.120.7.63&path=1,0&plc=ControlLogix&elem_size=108&elem_count=1&name=UDT

	23/01/2021 3:20:23	Tag "UDT
	23/01/2021 3:20:23	" (00c4): protocol=ab-eip&gateway=192.120.7.63&path=1,0&plc=ControlLogix&elem_size=4&elem_count=1&name=DINT

	23/01/2021 3:20:23	Tag "DINT
	23/01/2021 3:20:23	" (00c4): protocol=ab-eip&gateway=192.120.7.63&path=1,0&plc=ControlLogix&elem_size=4&elem_count=1&name=TotalCount

	23/01/2021 3:20:23	Tag "TotalCount
	23/01/2021 3:20:23	" (8fce): protocol=ab-eip&gateway=192.120.7.63&path=1,0&plc=ControlLogix&elem_size=88&elem_count=1&name=String

	23/01/2021 3:20:23	Tag "String
	23/01/2021 3:20:23	" (00c1): protocol=ab-eip&gateway=192.120.7.63&path=1,0&plc=ControlLogix&elem_size=1&elem_count=1&name=Bool

	23/01/2021 3:20:23	Tag "Bool
	23/01/2021 3:20:23	" (20c4): protocol=ab-eip&gateway=192.120.7.63&path=1,0&plc=ControlLogix&elem_size=4&elem_count=20&name=DINTArray

	23/01/2021 3:20:23	[20]
	23/01/2021 3:20:23	Tag "DINTArray
	23/01/2021 3:20:23	" (00c4): protocol=ab-eip&gateway=192.120.7.63&path=1,0&plc=ControlLogix&elem_size=4&elem_count=1&name=String_DINT

	23/01/2021 3:20:23	Tag "String_DINT
	23/01/2021 3:20:23	" (00c4): protocol=ab-eip&gateway=192.120.7.63&path=1,0&plc=ControlLogix&elem_size=4&elem_count=1&name=CartonSize

	23/01/2021 3:20:23	Tag "CartonSize
	23/01/2021 3:20:22	Unable to read tag!  Return code PLCTAG_ERR_NOT_FOUND

	23/01/2021 3:20:21	Unable to open tag!  Return code PLCTAG_ERR_NOT_FOUND

	23/01/2021 3:20:21	Getting tags for program: Program:MainProgram.

	23/01/2021 3:20:20	Unable to read tag!  Return code PLCTAG_ERR_NOT_FOUND

	23/01/2021 3:20:18	Unable to open tag!  Return code PLCTAG_ERR_NOT_FOUND

	23/01/2021 3:20:17	Getting tags for program: Program:Prog1.

	23/01/2021 3:20:17	Found program: Program:Prog1

	23/01/2021 3:20:17	Found program: Program:MainProgram

	23/01/2021 3:20:16	Getting controller tags.

And this is the complete debug output (with format :-) )

> 
2021-01-23 03:20:16.319 thread(1) tag(0) INFO plc_tag_create:559 Starting
2021-01-23 03:20:16.912 thread(1) tag(0) INFO initialize_modules:177 Starting.
2021-01-23 03:20:16.912 thread(1) tag(0) INFO initialize_modules:186 Creating library mutex.
2021-01-23 03:20:16.912 thread(1) tag(0) DETAIL mutex_create:633 Starting.
2021-01-23 03:20:16.912 thread(1) tag(0) DETAIL mutex_create:660 Done.
2021-01-23 03:20:16.912 thread(1) tag(0) INFO initialize_modules:207 Initialized library modules.
2021-01-23 03:20:16.912 thread(1) tag(0) INFO lib_init:122 Setting up global library data.
2021-01-23 03:20:16.912 thread(1) tag(0) INFO lib_init:124 Creating tag hashtable.
2021-01-23 03:20:16.912 thread(1) tag(0) INFO hashtable_create:75 Starting
2021-01-23 03:20:16.912 thread(1) tag(0) INFO hashtable_create:99 Done
2021-01-23 03:20:16.912 thread(1) tag(0) INFO lib_init:130 Creating tag hashtable mutex.
2021-01-23 03:20:16.912 thread(1) tag(0) DETAIL mutex_create:633 Starting.
[... 2130 lines cut ...]
2021-01-23 03:20:23.005 thread(1) tag(0) WARN lookup_tag:2853 Tag with ID -19 not found.
2021-01-23 03:20:23.005 thread(1) tag(0) WARN plc_tag_get_int8:2511 Tag not found.
2021-01-23 03:20:23.005 thread(1) tag(0) WARN lookup_tag:2853 Tag with ID -19 not found.
2021-01-23 03:20:23.005 thread(1) tag(0) WARN plc_tag_get_int8:2511 Tag not found.
2021-01-23 03:20:23.005 thread(1) tag(0) WARN lookup_tag:2853 Tag with ID -19 not found.
2021-01-23 03:20:23.005 thread(1) tag(0) WARN plc_tag_get_int8:2511 Tag not found.
2021-01-23 03:20:23.005 thread(1) tag(0) WARN lookup_tag:2853 Tag with ID -19 not found.
2021-01-23 03:20:23.005 thread(1) tag(0) WARN plc_tag_get_int8:2511 Tag not found.
2021-01-23 03:20:23.005 thread(1) tag(0) WARN lookup_tag:2853 Tag with ID -19 not found.
2021-01-23 03:20:23.005 thread(1) tag(0) WARN plc_tag_get_int8:2511 Tag not found.
2021-01-23 03:20:23.005 thread(1) tag(0) WARN lookup_tag:2853 Tag with ID -19 not found.
2021-01-23 03:20:23.005 thread(1) tag(0) WARN plc_tag_get_int8:2511 Tag not found.
2021-01-23 03:20:23.005 thread(1) tag(0) WARN lookup_tag:2853 Tag with ID -19 not found.
2021-01-23 03:20:23.005 thread(1) tag(0) WARN plc_tag_get_int8:2511 Tag not found.
2021-01-23 03:20:23.005 thread(1) tag(0) WARN lookup_tag:2853 Tag with ID -19 not found.
2021-01-23 03:20:23.005 thread(1) tag(0) WARN plc_tag_get_int8:2511 Tag not found.
2021-01-23 03:20:23.005 thread(1) tag(0) WARN lookup_tag:2853 Tag with ID -19 not found.
2021-01-23 03:20:23.005 thread(1) tag(0) WARN plc_tag_get_int8:2511 Tag not found.
2021-01-23 03:20:23.005 thread(1) tag(0) WARN lookup_tag:2853 Tag with ID -19 not found.
2021-01-23 03:20:23.005 thread(1) tag(0) WARN plc_tag_get_int8:2511 Tag not found.
2021-01-23 03:20:23.005 thread(1) tag(0) WARN lookup_tag:2853 Tag with ID -19 not found.
2021-01-23 03:20:23.005 thread(1) tag(0) WARN plc_tag_get_int8:2511 Tag not found.
2021-01-23 03:20:23.005 thread(1) tag(0) WARN lookup_tag:2853 Tag with ID -19 not found.
2021-01-23 03:20:23.005 thread(1) tag(0) WARN plc_tag_get_int8:2511 Tag not found.
2021-01-23 03:20:23.005 thread(1) tag

GitHubDragonFly, 2021-01-23:
If you have a Windows based computer then I would still suggest you try using the AdvancedHMI tag browser, which would be equivalent of getting a second opinion from another doctor.

Besides for having the MainProgram I did create the second program (Prog1 with 2 tags like yours) and the AdvancedHMI tag browser lists all of them at the same time.

GitHubDragonFly, 2021-01-23:
And if you have an Android tablet or phone then you could also try one of the apps from my repos.

They do feature "Get Tags" button and a program name can be changed.

elbaghee, 2021-01-23:
Thanks GitHubDragonFly, I will check it. But I was looking for a code in C++ or C to build it join my code.

About your apps in Android, this is very interesting because my end target is can build my application in Windows and Android and for the moment libplctag doesn't support this platform, but I sure that in the future it will do it.

kyle-github, 2021-01-23:
@elbaghee, libplctag does support Android, thanks to @GitHubDragonFly.   It will build on Android.  I have not had time in the past few months to really polish it up and make a clean release, but it compiles just fine.  There is also a project for Java using that C library.  All these are in the main libplctag project on GitHub.

GitHubDragonFly, 2021-01-24:
@elbaghee, my apps are using libplctag and if you are to test either of them and get the same results as you are getting now, then that would indicate that the library has a bug which needs to be fixed. Otherwise it would indicate that your code has a bug which you need to fix.

AdvancedHMI tag browser is just another tool that can also be used for quick troubleshooting.

The sooner you are to try either one of these, the sooner your issue might get resolved.

elbaghee, 2021-01-24:
@GitHubDragonFly, OK I go to test libplctag4Android in Android Studio, I never before used this IDE, but I hope I won't have problems... When I'll finish with this, I will do the same with AdvanceHMI...

GitHubDragonFly, 2021-01-24:
@elbaghee, It is sort of wrong order and wrong repository.

AdvancedHMI should take you approximately 5 minutes to complete free registration and download.

Testing with Android apps might take several hours since you need to have Android Studio installed.
My apps are not in libplctag organization, you need to go to my github account to get them.
libplctag4android was the very first Android app that was created, is fairly basic and doesn't have the "Get Tags" button.

GitHubDragonFly, 2021-01-24:
You just need to find time to read, since these are the build instructions for my apps:

- Download and install Android Studio.
- Download and extract the zip file of the project.
- Open this as an existing project in Android Studio and, on the menu, do a Build/Rebuild Project.
- Locate created APK file in this folder: app/build/outputs/apk/free/debug.
- Copy the APK over to your phone and install it (you might need to enable the "Install from Unknown Sources" option).

elbaghee, 2021-01-24:
@GitHubDragonFly, yes I installed Visual Studio and test AndvancedHMI easily. I tested to list tags and works fine, all tags are read (controller tags and program tags), so in PLC everything is OK.
I'have tried to test libpltag4android, I didn't know that there were other codes, or I misunderstand it. But I have a problem with AVD manager, when I start it immediately the process is killed. I have tried different "solutions" from Google but they don't fix this problem to me...
I go to search your github and I will try again... but before I want to fix AVD problem...

elbaghee, 2021-01-25:
I just execute Phone Test apk in BlueStacks emulator (I tried everything in Android studio but AVD doesn't work) and Controler Tags and MainProgram Tags are listed. In my C++ project I'm using the libplctag v2.1.22 and I'm using list_tag function to list tags but with program tags doesn't work, however in Phone Test works... I understand nothing :-(

GitHubDragonFly, 2021-01-25:
It looks like your C++ code that is wrapping list_tag function is not correct.

As for the AVD Manager in Android Studio, this page lists minimum requirements:

https://developer.android.com/studio/run/emulator#requirements

GitHubDragonFly, 2021-01-25:
Thanks for mentioning the BlueStacks emulator.

I just tried it and it is definitely another tool that can be used for testing APKs.

Also added some new screenshots of this emulator to my repos.

elbaghee, 2021-01-25:
> 
> 
> It looks like your C++ code that is wrapping list_tag function is not correct.
> 
> As for the AVD Manager in Android Studio, this page lists minimum requirements:
> 
> https://developer.android.com/studio/run/emulator#requirements

Yes, seems that there is something wrong, but I don't understand what, I only call the example list_tags function with argument number and one array with IP and path... and there is no extra code between listing controller tags and program tags, all code is from the library...

Anyway, I don't go to spend more time with this at the moment. I can read and write tags in controller tags and in all program tags... so for the moment is enough for me...

@kyle-github & @GitHubDragonFly thanks a lot for your help.

GitHubDragonFly, 2021-01-25:
I am not a C++ person and cannot really understand why your config[0] is set as an empty string.

If you don't have any more need for this issue to remain open then close it.

elbaghee, 2021-01-26:
I set config[0] as an empty string because this index is not used in the library, IP is in [1] and Path in [2] (list_tags function is the same as the main function in the example, only fprintf were changed to show them in a grid in the form)

[image]

OK, I close it. If in the future I found something or have more issues I will open a new one.

Thanks again.

closed 2021-01-26 by elbaghee

GitHubDragonFly, 2021-01-27:
Just so you know, another test could be performed in Windows:

It is sufficient to use Visual Studio, clone the libplctag library and build it.
Then open Command Prompt, change to the directory that has list_tags.exe file and run it with your IP and Path.

kyle-github, 2021-01-27:
Thanks, @GitHubDragonFly, I assumed that @elbaghee had already done that.   Yes, what @GitHubDragonFly says is good advice.   I have run the `list_tags.c` code on Windows with no problems.   I used Windows 10, 64-bit x86, with a relatively recent Visual Studio (free edition) version.

Based on the output, it looks like there may be memory corruption due to either use-after-free or pointers into stack memory after a function return.