libplctag/libplctag #228: Python Wrapper

GitHubDragonFly, 2021-01-28:
This is sort of an instructional post of how to use python wrapper in python3 on Windows and with a tkinter GUI.

The folder structure looks like in the attached picture, where the corresponding plctag libraries have been copied to their own folders.

[image]

The libplctag.py code was slightly modified for loading the C library on different systems (see the attached).
It works in Windows but I'm not sure about the other. The logic suggests that Android precompiled library folders could also be included but Android support for python seems to be tricky.

[libplctag.py.zip](https://github.com/libplctag/libplctag/files/5890502/libplctag.py.zip)

The GUI interface/script was derived from the updated pylogix GUI example, which I recently completed.
A picture of the preliminary GUI looks like in the attached, where the DINT value was extracted from the received bytes with hard coded plcTagGetInt32() function.

[image]

It is my intention to eventually create a repository and include fully functional plctag_gui.py file.

kyle-github, 2021-01-30:
Wow!  This is nice. I will try this on Linux this weekend and see if it works there too.   Do you want me to set up a libplctag.py repo in the libplctag project for you?   You have done so many great contributions, it would be good to have them more visible!  I am still underwater with the new event core and will be for a while.

GitHubDragonFly, 2021-01-31:
Try not to take what I am to state too seriously.

I still intend to create my own repository with this since it makes it easier for me to maintain.
My repositories are generally intended as a self-serve place that only requires people to read instructions and use whatever they can.

On the other hand, you should create separate repositories within your organization, for each language, so you can have a better control and keep the relevant posts within each repository.

Neither of the following is your competition: AdvancedHMI, pylogix or me.
I contribute to libplctag and pylogix and AdvancedHMI with sole purpose of trying to help users understand their choices and hopefully solve their issues and all for FREE, in hopes to "make this world a better place" (this last part is from the Silicon Valley show on TV).

Promoting any of these can easily be done just by mentioning them in the main repository page, take a look at my TabletTest and PhoneTest repos with the "Useful Resources" paragraph on the bottom.

GitHubDragonFly, 2021-01-31:
In case if you would like to experiment with GUI file, it is attached and so far it has a code to get tags.

[plctag_gui.zip](https://github.com/libplctag/libplctag/files/5899623/plctag_gui.zip)

Also, in case if you might be interested, I experimented with pycomm3 to get and list UDT members.
There is an issue in that repo, which I opened and closed, with a file that is using pylogix for the Device Discovery and pycomm3 at the same time.
The tag listing produced by the pycomm3 looks like in the attached file.

[image]

I will go ahead and close this issue.

kyle-github, 2021-01-31:
Great stuff!   No worries if you want to keep going in your own repo.   

Can I keep this open as a reminder to update the README to point to your repo?  

I am still working with the new core and it is fighting back.   Just got raw CIP to work by pulling some tag definitions.   Next up is testing getting UDT definitions.

GitHubDragonFly, 2021-01-31:
Keep this open for as long as you want.

Whenever I manage to create a new repo then I will eventually post the link here.

GitHubDragonFly, 2021-02-05:
Pydroid 3 app on Android devices can run this project with your library.
GUI file has to be opened in the IDE and run from there.

I have updated the folder structure picture and the libplctag.py file in my initial post.

GitHubDragonFly, 2021-02-05:
There also seems to be another way of making things work and not only on Android:

https://github.com/dmroeder/pylogix/issues/137

With respect to using your library on Android, I still find my apps to currently have the best approach.

kyle-github, 2021-02-05:
Ooh, thanks for the link!

Yeah, there has to be a better way.   I just found out that JFrog is going to wind down the bintray project and thus all my hosting for an Android package.   I think I may just give up in disgust trying to get it fully into the Android ecosystem and just build my own packages as you said :-/

GitHubDragonFly, 2021-02-05:
The libplctag4android project is already set to clone the latest "prerelease" branch.

If you change it to point to the "release" branch then you can easily get the packages ready.

I still think that you could create another Android project for creating AAR, just like it was discussed in the libplctag4android project. I usually update it in my repos so you can always get it from there instead.

GitHubDragonFly, 2021-02-09:
Even though this is still work in progress, I did create a new repo with current GUI file:

https://github.com/GitHubDragonFly/PythonTest

Also, check the libplctag.py file in the first post since I did update it to include string data accessor functions, for which I am not completely sure if everything is correct and how they need to be used.

GitHubDragonFly, 2021-02-09:
There is also a new repo with VB Net app for Windows, that can be used for testing:

https://github.com/GitHubDragonFly/AB-Modbus-Master

kyle-github, 2021-02-09:
Ooh, nice work!

kyle-github, 2021-02-09:
Unfortunately, I am not a Python programmer, so I am not sure if the Python for the string handling is right or not :-(

GitHubDragonFly, 2021-02-10:
I did update libplctag.py again and do get all the string functions return correct values with this python code (just in case if somebody might need it):

'
stringLength = plcTagGetStringLength(tagID, 0) 
stringCapacity = plcTagGetStringCapacity(tagID, 0) 
stringTotalLength = plcTagGetStringTotalLength(tagID, 0) 

stringValue = str('\0' * (stringLength + 1))
if int(pythonVersion[0]) >= 3:
    stringValue = stringValue.encode('utf-8')

response = plcTagGetString(tagID, 0, stringValue, stringLength + 1)

tagValue['text'] = ''.join(map(chr, stringValue))
'

GitHubDragonFly, 2021-02-15:
Here is a few more bit/bool functions that I missed to put in the libplctag.py file, approximately at line 338 just above the #String:

# Bit/Bool
def plc_tag_get_bit(tag, offset):
    return plcTagGetBit(tag, offset)

def plc_tag_set_bit(tag, offset, value):
    return plcTagSetBit(tag, offset, value)

# String

You can probably update it yourself so I don't have to put a PR in.

GitHubDragonFly, 2021-02-18:
I consider this done currently so you can close this issue whenever you are ready.

The app is finished so give it a try (and also try not to report a bug).

kyle-github, 2021-02-20:
Nice work on that app!   That is really a good test/demonstrator! 

I am getting pretty close to supporting UDT definitions now too.   See issue #231 for tracking.

I'll close this now.

closed 2021-02-20 by kyle-github

GitHubDragonFly, 2021-02-27:
@kyle-github just in case if you might be interested, there is a special version of this running on the iMac PowerPC and with the library v2.1.17 compiled from the source:

https://github.com/GitHubDragonFly/PythonTest_PPC