libplctag/libplctag #230: OMRON NX1P2 Communication failed

hussainkadori, 2021-02-04:
Hi, 
I tried this library to communicate with omron nx1p2 plc but it failed. Does this library support this type?

kyle-github, 2021-02-05:
Omron NJ/NX are the only ones that I know of.   The support for Omron is only for those models that support the Allen-Bradley/Rockwell CIP protocol.   

What version of the library are you using?   Are you using the C DLL directly or a wrapper like libplctag.NET?   

How are you connecting to the Omron PLC?   Can you share your tag string or the call into the wrapper so that we can see what arguments you are sending?

Thanks for reach out!

hussainkadori, 2021-02-05:
Hi, I connect PLC with EthernetIP using CIP Protocol c# sample and it worked but when I used the libplctag library the output as shown in this image [https://ibb.co/7nxGrMK](url) , I used Python wrapper and build this repo from source.

PLC IP=192.168.1.180 Port=2

GitHubDragonFly, 2021-02-05:
You have used an incomplete wrapper. Get Tags button/function is only set for ControlLogix PLCs.

If you think it might work for Omron as well then change the code.

kyle-github, 2021-02-20:
The Python wrapper has been updated significantly.   Please get the latest version v2.3.3.

Please reopen this issue if you still have problems.  Thanks!

closed 2021-02-20 by kyle-github

hamduc7, 2022-07-19:
Hi @hussainkadori 
I have the same problem like you. I need to write codes to get list of tags in OMRON NX1P2, write/read tag value. 
Have you solved this issue?

TQK0209, 2022-11-12:
@kyle-github I connect using ethernet, and python language. I connected but when I read the cell in it I get 1 wrong result. Can you help me

kyle-github, 2022-11-12:
Hi @TQK0209, this is a closed issue.  Please join the [forum](https://groups.google.com/g/libplctag) and ask questions there, thanks!