libplctag/libplctag #231: Support for pulling both tag and UDT definition data from a Rockwell PLC

kyle-github, 2021-02-06:
This ticket is to gather information and requirements about how this should be represented to users.   From the user perspective, something along the lines of the `@tags` magic tag would be nice, but where should the line be drawn.

Some considerations:

1. Doing too much in C takes away choices from the application and wrapper.   For instance if we add a magic tag to get both the tags and UDTs in one pass, then the application does not get the option to avoid that much memory allocation.   Admittedly even thousands of tags in a PLC is not that much memory on today's PCs.
2. Doing too much in one step makes it harder to trim specific functionality when needed for embedded use.   Perhaps tag listing is something that deeply embedded clients won't be doing anyway.
3. What format would such data be stored in?   Currently `@tags` returns the same data the PLC returns with no interpretation beyond simply getting it all.   The Rockwell "native" format for UDTs is not easy to process.

jkoplo, 2021-02-06:
Some thoughts after reading the email chain and above:

1) Because type can't be determined until runtime, high level languages are going to have some very different ways of dealing with this. The C# way is going to be quite different than the python way.

My initial thought in C# is to represent a runtime UDT as a json document. I'm sure that is making the memory conservative C developer cringe... :-)

2) Could we maybe collaborate on a way to parse UDT in pseudo code? Then we could all (java, c#, etc) program to that. And maybe identify functions that could be pushed into C and reused by the higher languages?

I don't think embedded needs to be able to list and dynamically create tags. With the cost of cheap arm chips I don't have a strong use case for embedded.

kyle-github, 2021-02-06:
Let me expand on the embedded points to see if we can come to some agreement on targets.   I think there are three categories of client/application here:

1. Tiny.   Really tiny.   Arduino UNO?   Stretch goal: PIC.   There is no way that systems like this are going to do more than read/write a small number of simple tags.   Listing tags?   Nope.
2. Small but not too crazy.  Systems with ~1MB of RAM.   ESP32 and similar maybe?   With some care, and pure C coding, you could list some tags here, but mostly just reading and writing tags.
3. PC-class.   RPi or greater.   256MB or RAM or more.  OK my definition of PC class is a bit dated :-)   No specific limits.

All the C# and other wrappers except C++ fall into category 3.   Right now, the library is only tested on category 3.   I am trying to make it work realistically on category 2.   Category 1 is not really a goal unless I get particularly lucky with the design of the code.

Interesting thought about dynamically creating tags.   I wonder if I can get the build system to calculate most of the constant data for a tag.   I probably can.  That would put category 2 in reach more easily and might even allow for category 1.   Seems like a bit of an abuse of CMake, but that is what build systems are for.   I'll shelve that idea for now.   I need to focus on getting the new core done.

timyhac, 2021-02-06:
Just wanted to copy Kyle's words from [here](https://github.com/orgs/libplctag/teams/maintainers/discussions/20?from_comment=9#discussion-20-comment-9) that seem pertinent to this discussion:

> There are four different things that need to be done in order to really list tags with all associated structural descriptions:
>
> 1. List the tags themselves. That's mostly covered by the existing @tags magic tag. However, I still make the user code build the tag names to fetch the program tags. This requires multiple round trips to the PLC internally if you have more than a handful of tags.
> 2. Get the metadata about each UDT. This includes the number of fields and the total size of a single instance of the UDT. Call is one per UDT.
> 3. Get the medatadata about each field in a UDT. Call is one per UDT, but due to packet size, may require multiple requests for a single UDT. Not fun to automate outside the library.
> 4. Call some special CIP call that returns some values. You store those. Call this magic CIP function again every few minutes. Compare the values. Did they change? If so, then you need to do steps 1-3 again because someone edited the tags. This apparently is the only way to tell if something changed.

timyhac, 2021-02-06:
Another dimension to this is what "dynamic" means.

In statically typed languages, it could be helpful to have a tool to query a PLC for UDT definitions, and generate code files that can be used by a developer as part of their application.
In the context of libplctag.NET, we would generate Mapper classes such as [this](https://github.com/libplctag/libplctag.NET/blob/master/src/Examples/CSharp%20DotNetCore/SequencePlcMapper.cs).

This scenario is dynamic at (or before) compile time.

A second meaning is where it is dynamic at run time.

Example scenario: A web application that has a drag-and-drop User Interface designer. The person using this software is able to get a list of tags (and subtags) available in a target PLC, and are able to bind a User Interface element to the data in the selected tag. When they view the User Interface they have designed, the tags would be read periodically and the UI would animate. No prior knowledge of which tags are available on a PLC would be required.

To implement this, listing "standard" tags as well as "subtags" within UDT instances would be required.
It would also need all other information required to create a tag object (e.g. name=MyUdtArray[0,0].Enabled, elem_size=1, etc) and its type (which would inform which Tag Data Accessors should be used).

timyhac, 2021-02-06:
To give some opinions on considerations 1 and 2 posted in [this post](https://github.com/libplctag/libplctag/issues/231#issue-802569115).

I'm familiar with browsing tags using PLC programming tools that present tags as a tree list. When the user expands a UDT instance, it shows the subtags for that instance. In case I'm not explaining myself properly, tejaswi posted an image showing this in [this post](https://groups.google.com/g/libplctag/c/3rQ-8yNFQ7M/m/kkAxJJU0DAAJ).
If someone was to develop an application with a similar browsing feature, they would only need get the top level tags, only getting the UDT definitions when required.

For this use case, my opinion is that it would be totally acceptable to have the subtags returned on demand rather than upfront. To make the API convenient, there would need to be some link between the @tags result and the UDT structure query: "for this UDT tag, get its structure".
For the other use cases I've mentioned in the previous post it would also be acceptable to have the subtags returned on demand.

For consideration 3 in [that post](https://github.com/libplctag/libplctag/issues/231#issue-802569115), jumping on the magic tag idea, libplctag could present a special syntax to return the list of fields for one specific UDT definition. The UDT ID comes from the Type field in `@tags`.

`plc_tag_create("@udt=<UDT_ID>", 0)`.

Each field would have
* Name (Variable Length ASCII String + its Length) 
* Offset (1 byte?)
* Type (2 bytes)

kyle-github, 2021-02-06:
The information I can get for a UDT is this:

- The name of the UDT (there are some confusing parts because the names of UDTs are _NOT_ counted and _NOT_ zero terminated!).   
- The total size in bytes of one instance of the UDT.
- The number of fields, but this includes invisible fields!
- The name of each field (see the comment above about the name of the UDT, same kind of weirdness applies, only slightly different).
- The type of each field, but, thanks Rockwell, not in the same binary format as the type of a tag.   For instance 0x2000 (a single bit) being set means that the field is an array.  This is a 16-bit value and can indicate an atomic type like DINT, a bit value, or a UDT ID.
- The array length.   There is nothing about the dimensions of the array.   Just the total length.   This is a 16-bit value.
- The offset in bytes where the field starts.
- If the field is a bit (0xC1 type), then the metadata field otherwise used for the array length contains the bit number of the bit field.

Any field that starts with "ZZZZZZZ" (I forget how many), is considered a hidden field.   These are used for packing booleans at the front of UDTs.

I am leaning toward the "magic tag" approach on this.   @timyhac's point about the usual use case of drilling down rather than getting all of the UDTs at once is a good one.

One thing that is fairly ugly is just how much data this is.   It can take a lot of packets and requests back and forth to the PLC to get this data.   Getting the UDTs is particularly slow as it is one at a time and two requests per UDT.

I am going to take a swing at two things today:

1. See if I can use the same CIP commands to get lists of UDT information as I do for lists of tag information.  If I can do this, then I can have a magic "@udts" tag and get them all at once with far less network traffic.  
2. If not, I will cobble together a magic "@udt=<id>" tag type.

timyhac, 2021-02-06:
Something that could explain why individual array aren't available is this from page 33 of 1756-PM020G-EN-P 

Each member can be an:
• Atomic data type.
• Another structured data type.
• **A single dimension array of an atomic or structure data type.**

jkoplo, 2021-02-06:
RSLogix v19 - yikes! ;-)

We've done a lot of populating tags from AB using a commercial driver. It can only populate ALL tags/UDTs and it's a long process - 30+ seconds depending on how loaded up the PLC is. Also, it's a single blocking call so I can't provide any progress indicator to the user. And there's no auto-detection of change. If a PLC is edited while we're connected we give the user a button to trigger a FULL tag re-scan.

I haven't read the technical doc or really taken the time to think through the details above, admittedly. But being able to only scan smaller portions of the PLC or load UDT definitions individually would be very much appreciated.

kyle-github, 2021-02-06:
@timyhac Yep, that's the only data you get.  I had not looked at the UI to see if Rockwell limits it, but I guess that's good?  I have a L55 CPU sitting around here somewhere and I think that only will run version 18 firmware.   Maybe earlier?  12?

@jkoplo the magic tag approach seems pretty solid so far for listing tags.  I am building a bit more infrastructure around CIP PLCs right now to redo the magic tag approach for listing tags.  I am doing that first as it is a known quantity.  There will be no functional change from before.   Then I will do `@udt=xyz` magic tags.   Then I will do another, name TBD, to read out the special values that Rockwell says will let you know if the tags changed.   The units you will be able to get are:

1. list controller tags. `name=@tags`
2. list each program's tags (one call per).  `name=Program:NiftyProg@tags`
3. get UDT details, one at a time. `name=@udt/xyz`.  I might need to change the separator to `/`.
4. get the special values that indicate when tags got changed. `name=@wtfRockwell`.

I will also set up (eventually as the old code has this hidden away) the ability to open a second connection to a PLC.   You can run the tag listing tags on that and not impact the main data connections too much.  If you have a L8x and 1Gbps networking.

timyhac, 2021-02-06:
Looking further into this, [1756-pm004](https://literature.rockwellautomation.com/idc/groups/literature/documents/pm/1756-pm004_-en-p.pdf) at the top of pg 36 says 

> If you include an array as a member, limit the array to a single dimension. Multi-dimension arrays are not permitted in a user defined data type.

timyhac, 2021-02-07:
In my view, a useful feature is how to determine what xyz should be in `@udt/xyz`. Ideally this can somehow be automatically inferred from information returned by `@tags` or some other listing service, so that you don't need to know beforehand what UDTs are available and what their name/id is.

In [this post](https://github.com/libplctag/libplctag/issues/231#issuecomment-774472492) I'd naively assumed that the Type field (from `@tags`) could be used for this, but I don't understand what services are available in Rockwell controllers.

kyle-github, 2021-02-07:
@timyhac it is the UDT ID and is determined as per page 45 in the Rockwell guide you referenced.   You take the 16-bit tag type field.  If bit 15 is set, then you mask and take the lower 12 bits of the value.  That is the UDT/Template ID. There can only be 4096 UDT and system types.  Somewhat weird that it is so limited.

There is also a structure handle.  If you read in a tag that is a UDT or an array of UDTs, then the type field is a two INT sequence.  The first one is something like 0x02A2.   That indicates that you have a struct or array coming.   The next INT is a CRC calculated across the structure's fields.   It is not an ID.     I have yet to see any way to use that value anywhere.   Given that it is only 16 bits, it is fairly likely that you will get collisions.  It is definitely not unique.

It would have been great if Rockwell had returned the UDT ID with the tag value when it was returning the type.  But no.  That would have been far too easy.

If there is a hard way to do something in a protocol, Rockwell has done it at least twice.

jkoplo, 2021-02-07:
Is the CRC calculated across the data in the fields or the types in the fields? 

Can it be cached and used to determine if the data in a UDT changed without pulling the whole UDT?

kyle-github, 2021-02-07:
IIRC, it is the names and types.   It might not include the names.   It does not include the values, so you cannot use it to determine whether the data in a UDT changed.

But you have an interesting point about the CRC.  You could look at the CRC to determine if a specific UDT changed definition.   I.e. I could save the UDT CRC, then when you pull it again, you could potentially tell if that specific UDT was the one that changed without pulling all the field metadata.   When repulling all the UDT definition data, you could potentially use this to skip one of the parts.   

Definitely an optimization that might be worth it in the future.

kyle-github, 2021-02-07:
I will add the CRC field to the list and store it in the results.

jkoplo, 2021-02-07:
Yeah I think that makes it so you could check a PLC for changes while you were connected by:

1) Listing tags at controller/program level and make sure they match previous.
2) For all UDTs found (and recursively) - check the UDT CRC against previous.

That doesn't seem like it would be too many calls (obviously depending on how many tags and UDTs are used) and would be an improvement for us against the current library we use that has no way to check if things changed.

Of course, this does start to pull me back toward having a separate class to represent the controller. 😁

From the C# wrapper side, I think it's going to make sense to have a `libplctag.rockwell` or `libplctag.clogix` library to handle caching and UDT discovery/parsing.

kyle-github, 2021-02-07:
Oh, you can check for edits fairly easily.   There is (yet another) a special CIP call that gets a few attributes of some object.   If those attributes change, then you know that the tag definitions have changed.    Rockwell does not specify what the fields mean.  They just say that if they change, you know that you need to re-pull everything.

I have the CRC field in now.   I need to:

- add the "check for changes" magic tag.   This should be straightforward.
- get the latest changes working with the new implementation of various magic tags.  I.e. fix up list_tags.c to show UDTs too.
- get Windows working.

Once I get these, I would really like some help testing.   I can only do some much on the limited hardware I have.   And, unfortunately, one of the companies that used to give me old hardware from time to time has gone out of business (more fallout from Boeing being run by spreadsheet jockeys instead of engineers, and COVID).  So it is unlikely I will get more cheap hardware in the future :-/

One interesting challenge with these special CIP calls is that only some of them have the ability to do offsets.   I am not sure what happens if you pack them in with other requests and get back partial data.   I will need to try to simulate that or make these requests non-packable.

jkoplo, 2021-02-07:
Would it help with testing if we made a C# based app that people could grab and run? Maybe have it do a bunch of different permutations of calls and save the results to a file (CSV, json log, whatever) that could be emailed?

kyle-github, 2021-02-08:
That sounds like a good idea.

kyle-github, 2021-02-14:
Closing in on the first cut of the "magic" tag solution.   I have `@tags` working as before, but on the new event-based core.  I have `@udt/xxx` working now.   That pulls the UDT info.   It builds a special four field header that Rockwell does not provide, but the rest is just the Rockwell dump of UDT field information.   Rockwell made it such that you have to do one CIP call to get the total number of fields and the total size of the field information.   You need that in order to query for the field information.   So two round trips to the PLC for every UDT.

I built a new version of `list_tags.c` using these tags.

I also have the code written for the special call that you need to do to find out if the tag definitions changed.  I have not tested that yet.

Note that the information returned by the `@tags` magic tag is identical to that you got before.   Same data, same use.   Existing code should continue to work, apart from the need to use `protocol=abeip2` at the start of the tag string.

kyle-github, 2021-02-15:
I got the tag definition detection tag created.   This is creatively called `@change`.   I need a different name.   

It returns 38 bytes.   I tried adding a tag and one of the bytes changed.  I tried modifying an existing UDT and several more bytes changed.   It is not clear if a simple (but good) hash of all 38 bytes would be sufficient.

I also have basic tag read/write working with the new library core.   No bit support for Logix-class PLCs yet and no packet requests yet, but the new architecture makes that a standalone thing rather than invasive as it was in the old code.

Laro88, 2021-02-15:
I can't build "event" on MSVC (vs2019), seems like "__sync_fetch_and_or", "release" branch build fine.

kyle-github, 2021-02-15:
Yeah, that is, hopefully, one of today's tasks.  There are a lot of small parts that are not ported to Windows yet.   Mostly the atomic integer stuff.   These will be supported in Windows with the various InterlockedXYZ() functions.

Laro88, 2021-02-15:
I will check in the repo in the morning then (CET)
Btw what is the timeframe for the event based solution? I have a CompactLogix from a partner company on my table that I have to get UDT data out of so I can test your builds with live hardware on VS2019.

kyle-github, 2021-02-19:
I am still working on getting this to work on Windows.   I got the UDT data to dump out on Linux, but getting the whole new event-based core just compiling on Windows took a lot longer than planned.   Now I am working my way through it in a debugger in Windows.   

Once that is working, I will start having people test it even though it is mostly alpha level code.

kyle-github, 2021-02-19:
Huh, so you have to create sockets before you use them on Windows.  Picky, picky.

Anyone that is desperate enough can check out the `event` branch.   There are a number of things missing:

- multiple request packing.
- bit tags.
- general debugging.

In return you get:

- bugs!
- gratuitous and confusing debugging outputs!
- list_tags.c will list tags and UDTs, and it gets the UDTs recursively until there are no undefined ones left.  This uses the magic tags `@tags` and `@udt/xyz`.
- test_change_tag.c which pulls the "magic" information that Rockwell recommends to determine whether the tag definitions or UDT definitions in the PLC have changed.   I have tested that the data changes when either tags change or UDT definitions change.  This uses the magic tag `@change`.   Warning, that name is going to be different before this goes to release!

The only change you need to make to your code to ~find more bugs~use this fine code is to use `protocol=ab-eip2` in the tag string.

Laro88, 2021-02-19:
@kyle-github - where would you like feedback, do you have a slack channel somewhere?
I just tried x64 Debug on windows and got:
```
C:\GitProjects\github\libplctag\cmakeout\bin_dist\Debug>list_tags.exe 10.121.1.1 1,0
Starting with library version 2.3.2.
Unexpected UDT name length: 308!
Usage: list_tags <PLC IP> <PLC path>
Example: list_tags 10.1.2.3 1,0
2021-02-19 14:14:24.160 thread(1) tag(76) WARN plc_module_teardown:1808 PLC list not empty!
2021-02-19 14:14:24.165 thread(1) tag(76) WARN thread_join:185 null thread pointer.
2021-02-19 14:14:24.166 thread(1) tag(76) ERROR event_loop_teardown:130 Unable to join event loop thread, got error PLCTAG_ERR_NULL_PTR!
2021-02-19 14:14:24.166 thread(1) tag(76) WARN mutex_destroy:235 null mutex pointer.
2021-02-19 14:14:24.167 thread(1) tag(76) WARN plc_module_teardown:1815 Error, PLCTAG_ERR_NULL_PTR, destroying plc list mutex!
```

kyle-github, 2021-02-19:
Hi @Laro88, this is a good channel.

I have a check for all names that looks to see if they are longer than 256 characters.   However, the UDT name is really, really weird (even for Rockwell).   The name is a zero-terminated string, but you read until you find the first semicolon and that is the UDT name.  There is data after that to the zero terminator that Rockwell uses for ... something.   

So this check is hitting a problem with that data.   308 bytes is probably not a real violation of the 256 rule if we consider cutting the string off at the first semicolon.  I will rejigger that code to check two things:

1. Just print a warning but do not punt if we see a really long UDT name.
2. Check for 256 bytes _after_ I trim to the first semicolon.

I will try to have another drop later today.   Thanks for the report and the testing!

kyle-github, 2021-02-19:
There is another drop of the `event` branch now with the above code changed into just a warning.

Laro88, 2021-02-20:
New dump - this time into a file, attached. This looks quite promising!
[dump.txt](https://github.com/libplctag/libplctag/files/6015630/dump.txt)

The output (error stream is here)
```
C:\GitProjects\github\libplctag\cmakeout\bin_dist\Debug>list_tags.exe 10.121.1.1 1,0 > dump.txt
Unexpected raw UDT name length: 308!
2021-02-20 17:39:09.088 thread(1) tag(82) WARN plc_module_teardown:1808 PLC list not empty!
2021-02-20 17:39:09.092 thread(1) tag(82) WARN thread_join:185 null thread pointer.
2021-02-20 17:39:09.248 thread(1) tag(82) ERROR event_loop_teardown:130 Unable to join event loop thread, got error PLCTAG_ERR_NULL_PTR!
2021-02-20 17:39:09.251 thread(1) tag(82) WARN mutex_destroy:235 null mutex pointer.
2021-02-20 17:39:09.251 thread(1) tag(82) WARN plc_module_teardown:1815 Error, PLCTAG_ERR_NULL_PTR, destroying plc list mutex!

C:\GitProjects\github\libplctag\cmakeout\bin_dist\Debug>
```

kyle-github, 2021-02-20:
Looks like it is working!   The errors you still see are due to some incorrect checks running during library teardown, so they are not a problem for the tag and UDT output.   They are not as high a priority as the missing functionality and overall performance.   They will be fixed before this code is considered beta quality, though!

Thanks for testing this!

Laro88, 2021-02-20:
I will test with c++ against the lib to see how it behaves, thanks for the work so far.
Tag me when there is something new to test 🙂

fluisgirardi, 2021-03-08:
Hi @kyle-github!

I'll test this new feature and update the Pascal wrapper. Should I do my changes on the prerelease branch or keep the changes on this branch (multi_request_2)?

kyle-github, 2021-03-09:
Hi Fabio,

The new UDT support is in the *event* branch.  That is not stable yet.   If
you want to update the Pascal wrapper for that, please do it against that
branch.   The branch builds against Linux.  I think it works against
Windows.   It used to!

The other branch, *multi_request_2*, is for adding multiple request packing
to Logix-class PLCs.   I have not been able to work on that very much the
last couple of weeks.

There are some other test programs that come with the event branch.   The
most important is test_change_tag.c.   That is the last part of the tag
listing support.  It gets a set of attributes from the PLC, about 38 bytes
IIRC.   These bytes change when tags are added or changed or UDTs are added
or changed.   So you poll that tag and save the data you get, then when it
changes, you need to list the tags and get all the UDT definitions again.

You could probably use SHA256 or something similar to hash the results to
use fewer bytes.

Best,
Kyle

fluisgirardi, 2021-03-10:
Hi @kyle-github!

I'm having trouble trying to read some big UDT's. Getting PLCTAG_ERR_OUT_OF_BOUNDS. Probably is because it's a big UDT and it exceeds the payload (64k) limit? I'm seeking how to process this in two or more messages.

kyle-github, 2021-03-11:
Can you turn on debugging (level 4) and capture the logs?   I do not have much code in the UDT tag support yet to handle really big UDTs.

fluisgirardi, 2021-03-11:
Yes, I'll do it tomorrow. I'm going to sleep...

kyle-github, 2021-03-11:
Thanks!   Reading through this is confusing.  It looks like the code has pulled all the tag listing data (almost 64k!) successfully, but then there appears to be a crash (are you running this in a debugger?) as the socket is being closed.   That is likely to be a bug in my code, but I am not sure where.   

Is there any chance you can run this in a debugger and find out where this is crashing?   I tried briefly on my systems and I am not duplicating this crash.  

One other issue I see is that it takes almost 2 seconds just to get the tag listing part!   I think I need to change the default timeouts to handle such a long time.

This is very helpful!

kyle-github, 2021-03-11:
BTW, according to the log, this is not even trying to read the UDTs yet.   Whatever the crash bug you found is, it is in the code that shuts down a TCP connection.

The code in the current `list_tags` program is really dumb.   It closes each tag after use and that closes the connection to the PLC completely.  So in this case, the TCP connection to the PLC is being closed after all the tags have been listed but before any attempt has been made to get the UDT.

fluisgirardi, 2021-03-12:
@kyle-github yes. The example appears to crash before starts the UDT listing...

Here is the callstack of the list_tags.c error

*** buffer overflow detected ***: terminated

Thread 1 "list_tags" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50	../sysdeps/unix/sysv/linux/raise.c: Arquivo ou diretório inexistente.
(gdb) br
Ponto de parada 1 at 0x7ffff7d8d18b: file ../sysdeps/unix/sysv/linux/raise.c, line 50.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff7d6c859 in __GI_abort () at abort.c:79
#2  0x00007ffff7dd73ee in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff7f0107c "*** %s ***: terminated\n") at ../sysdeps/posix/libc_fatal.c:155
#3  0x00007ffff7e79b4a in __GI___fortify_fail (msg=msg@entry=0x7ffff7f01012 "buffer overflow detected") at fortify_fail.c:26
#4  0x00007ffff7e783e6 in __GI___chk_fail () at chk_fail.c:28
#5  0x00007ffff7e79a8b in __fdelt_chk (d=<optimized out>) at fdelt_chk.c:25
#6  0x00007ffff7f79559 in socket_close () from /usr/local/lib/libplctag.so.2.3
#7  0x00007ffff7f75fb9 in reset_plc () from /usr/local/lib/libplctag.so.2.3
#8  0x00007ffff7f7761f in plc_rc_destroy () from /usr/local/lib/libplctag.so.2.3
#9  0x00007ffff7f72d79 in rc_dec_impl () from /usr/local/lib/libplctag.so.2.3
#10 0x00007ffff7f69eb4 in magic_list_tags_tag_destroy () from /usr/local/lib/libplctag.so.2.3
#11 0x00007ffff7f72d79 in rc_dec_impl () from /usr/local/lib/libplctag.so.2.3
#12 0x00007ffff7f474e8 in plc_tag_destroy () from /usr/local/lib/libplctag.so.2.3
#13 0x0000555555555771 in main (argc=3, argv=0x7fffffffdea8) at list_tags.c:174
(gdb)

fluisgirardi, 2021-03-13:
[stderr_pascal.txt](https://github.com/libplctag/libplctag/files/6134573/stderr_pascal.txt)
[stdout_pascal.txt](https://github.com/libplctag/libplctag/files/6134574/stdout_pascal.txt)

Same debug level, Pascal program, loading UDT's immediately after getting tag list. This program doesn't crash because I removed the line plc_tag_destroy. But it freezes at some point that I don't know...

kyle-github, 2021-03-15:
Sorry for the radio silence.   Work is really killing me right now.

fluisgirardi, 2021-03-15:
Don't worry, I'm too busy now...

kyle-github, 2021-03-17:
I found a problem and updated the `event` branch.   Can you please give that a try?

fluisgirardi, 2021-03-18:
Yes. I'll report soon

fluisgirardi, 2021-03-18:
Working with a charm! The unique thing that isn't working is when I request a UDT that is too big and the response exceeds the payload limit...

fluisgirardi, 2021-03-18:
An observation: 

the first line of my Pascal program is:

  plc_tag_set_debug_level(PLCTAG_DEBUG_DETAIL);

but no debug info printed on the output after the last updates...

kyle-github, 2021-03-19:
Hmm...

There should be output.  Is the program terminating normally?   Are you on Windows?  This seems odd.   You are seeing the tags and UDT information?

You can also get debugging output by adding a "&debug=4" to the tag string.

kyle-github, 2021-03-19:
I will need to work on the UDT size problem.  I thought I was handling status 6, but perhaps not.

fluisgirardi, 2021-03-19:
@kyle-github No, I'm using Linux. I can read almost everything, with the exception of big UDTs that can't be read from PLC and the plc_tag_set_debug_level that don't produce any output. But I'll rebuild everything again to ensure that is all ok.

fluisgirardi, 2021-03-19:
@kyle-github I have something wrong with my environment. Now plc_tag_set_debug_level is producing the expected output.

timyhac, 2021-04-02:
Hey there - I just got access to a controller today so thought I would try this out. I'm running on Windows 10 x64.

The issue I ran into was a `PLCTAG_ERR_OUT_OF_BOUNDS` error. Please see attached logs:
[list_tags.log](https://github.com/libplctag/libplctag/files/6248038/list_tags.log)

I noticed that list_tags_old.c was using the ab-eip2 protocol so thought I would run that too, in case it is useful. It seems to generate the tag list OK.
[list_tags_old.log](https://github.com/libplctag/libplctag/files/6248033/list_tags_old.log)

This binaries were build from code on commit: 3ddad036c1b646a700770a7c187132af70c6fd94

kyle-github, 2021-04-02:
That's the latest commit.  Hmm, I wonder if I did not push something?  No.   

Hmm, that's not good.   

Can you turn on debugging to level 4 and capture the output for the first attempt?   I am fairly sure that the point at which things are not working is when the code tries to get the UDT definition.  The old list tags code does not try to get the UDT definitions (though it should not be using ab-eip2!).

timyhac, 2021-04-02:
Yep no problems - see attached.
[list_tags.DETAIL.log](https://github.com/libplctag/libplctag/files/6251359/list_tags.DETAIL.log)

Interestingly, when using this logging level, it does output some information!

Just to make sure I'm sane, what I changed was the parameter at [this line](https://github.com/libplctag/libplctag/blob/event/src/examples/list_tags.c#L138) - which I changed to `PLCTAG_DEBUG_DETAIL` - is this correct?
I also changed it back to `PLCTAG_DEBUG_WARN` to double-check that there was no output - indeed the output was the same as in the previously posted logs.

P.S. the reason I think that old tag listing app used ab-eip2 is because of [these lines](https://github.com/libplctag/libplctag/blob/event/src/examples/list_tags_old.c#L78-L80).

kyle-github, 2021-04-07:
Is that log file corrupt?   I am getting a >5MB file and it changes into ... Chinese (?) about 5500 lines into the file.   Any chance you can ZIP the file and reattach?  I do not even get past the point where it is getting the tags.  It is not into UDTs yet.

timyhac, 2021-04-07:
Sorry the file encoding with UTF16 LE without BOM - must be the default encoding used by powershell.

Here it is in UTF8 without Byte Order Mark:
[list_tags.DETAIL.log](https://github.com/libplctag/libplctag/files/6268754/list_tags.DETAIL.log)

kyle-github, 2021-04-07:
Ah, thanks.  That looks better.   Interesting, the docs do not really specify how to calculate the offset when requesting another slice of data for a UDT definition.   Either I have the wrong offset, or the total length is not right.   The length calculation is really weird.  You get back a value for the total length, but that is in DINT equivalents (4-byte chunks).  Then there is an odd amount of 23 you need to remove.   The doc's explanation sounds suspiciously like "multiply by zero and add the right answer" to me. 

I will have to create a UDT with a sufficiently large definition and try this using a PLC accessed via small packets (your PLC can only do ~540 byte request packets).  

Sorry for the delays on this.   Lots of outside things going on right now.  Things should return to a more normal schedule in about two weeks.

jkoplo, 2021-06-16:
@kyle-github we're in a major next-gen development phase at work right now and looking at shifting our codebase to libplctag. But with that said, listing tags/UDTs at runtime is a required feature for us.

It looks like this branch might be mostly working other than very large UDTs? Anything we can do to help test/develop this?

kyle-github, 2021-09-08:
This is in the add_udt branch.   I'll close this as soon as there is more testing (and the bugs found get fixed!).

kyle-github, 2021-09-11:
This is getting close.  Thanks to @timyhac for doing some testing.  Looks like some system types (in particular MESSAGE) have non-documented (perish the thought) encodings.   The docs say that you should not assume how things are encoded for system types.  And the docs appear to be correct.   Rather than filter out all system types, I will see if I can make the code at least not break.

Note that this is in list_tags.c not in the library.  The library appears to be reading all tags and UDT definitions just fine.

[1 more reply]

closed 2021-09-24 by kyle-github