libplctag/libplctag #233: Release build dll naming convention

Laro88, 2021-02-08:
I have just started working with libplctag and got the version checking (plc_tag_check_lib_version) to work.
I am using the release builds provided for x86 of 2.2.0 , however the lib file is named "libplctag.lib" and the corresponding dll is named "plctag.dll" which I think is making Visual Studio 2019 unable to automatically copy the dll to the output directory - hand copying works fine though.

Any reason for this mismatch? The provided outputs seem to be a fairly new addition, see #138 for ref.

kyle-github, 2021-02-08:
Hi, thanks for raising this.  I am not at all a Windows programmer.   Is VS expecting a different name?  If so, do you know what name that is?

The build on VS is different than the others because VS takes over the release/debug build process and does both.   It ends up building debug builds in one directory and release builds in another, but the files all have the same names.  With all the other platforms, CMake is called with just release or just debug and there is no duplication.

kyle-github, 2021-03-21:
Any update here?   I can modify the output names, but I do not know what they should be.  Can you please provide an example name?

Laro88, 2021-03-24:
1. I would prefer "lib" as a prefix for the library name (libplctag.dll), however this is not a breaking thing - it just makes life a bit easier.
2. There is a rule of sorts to put a "d" after the library name (before extension) when it is a debug build (there are a couple below), but things works out eventually. 

If the git project name = project name from make = output name(s) then it is as helpful as can be for users down the road.

I just took a screenshot of one of my build output folders, they are mostly external libraries (except for the ~UT.dlls and LSLSim.dll) and it is not consistency :-)  there are a couple not doing it, and sthe casing is also off on PQ

[image]

So, if you change it then think of it as a small improvement, but everyone should be able adapt no matter what, myself included.

closed 2021-03-24 by Laro88