libplctag/libplctag #236: String API merge breaks non-byte accesses with Modbus

kyle-github, 2021-02-13:
Thanks to Atif Shoukat Ali for noticing this embarrassing miss!   I forgot to set the byte order information for Modbus tags.

closed 2021-02-13 by kyle-github

atifali, 2021-02-14:
I can confirm this has fixed the segfault on reads with non-byte accessors, at least with my setup! 

I do not have a physical device to test with as well, but I am very certain that the simulated protocol is representative of an actual PLC.

kyle-github, 2021-02-14:
Thanks!

Do you have any simulators that simulate strings in a Modbus PLC?   The
only thing I have is a really simple Python simulator that simulates coils
and registers and nothing else.

Strings are currently turned off for Modbus simply because I have no idea
or documentation on how they work.  Are they PLC specific?

Thanks again for testing!

Best,
Kyle

GitHubDragonFly, 2021-02-16:
From my understanding of Modbus, one has to designate the number of Holding Registers to store strings.
Since this is arbitrary number, a user would have to specify the starting address as well as this number in order to read/write strings.

If they choose to designate 10 registers then any string of up to 20 characters can be stored there.
Any time they are to read/write, a string length of 20 would have to be specified so all the registers get overwritten.
Once designated then they shouldn't be using those registers for anything else.

Here is a picture of ModRSSim2 simulator for Windows, which can show chars. The original code seems to be in C++ language so somebody would have to try modifying it for use in other operating systems.

[image]

atifali, 2021-02-16:
@kyle-github I believe on the protocol level all 'string' type data would be represented with some fixed registers dedicated for that purpose, exactly as @GitHubDragonFly pointed out. At least that is how I have mostly came across it during my short time working with the Modbus TCP protocol. 

Some PLCs might have some specific implementations based on fixed register sizes at specific addresses representing something predefined (vendor names, module names etc...). But I believe trying to support that would be out of scope for our purposes here. 

This means that any 'simulator' that can have a series registers should be perfectly capable of dealing with strings since that would just be a bunch of chars, with each register supporting 2 chars each. 

I can definitely help with this, once I have some extra bandwidth.

kyle-github, 2021-02-17:
Hi @atifali, thanks for the offer!

The storage of strings is not usually the problem (though it is necessary to know if 16-bit words are in big or little endian order for bytes).   The part that is challenging and often per PLC type is how strings are laid out.  Are they counted?  Are they zero terminated?  Is there padding?

For instance, Allen-Bradley PLCs use strings in several formats:

- DF1/PCCC PLCs like PLC/5 use strings with a 16-bit count word, 82 bytes of character data in big-endian byteswapped word (INT) format.
- CIP-based PLCs like Control Logix use strings with a 32-bit count word, 82 bytes of character data and two padding bytes.
  - except when they do not such as the string types you get back with tag listings in which case they are counted with a 16-bit count word, no explicit length, and no padding.
  -  or when you get fields in a UDT description in which case they are zero-terminated C-style strings.

Is it that description that makes strings difficult on Modbus PLCs.   If there is a common description for many Modbus PLCs, then I can set up a default.  If not, then I will need to make a table of string definitions by PLC type and slowly fill that in as people report it.

I am still working on solid code to allow the tag description/attribute string to describe all the parts of a string definition.

atifali, 2021-02-18:
@kyle-github really appreciate your explanation! 

I am not aware of any standard string descriptions for Modbus PLCs, but if I do come across any I will let you know. 

However, If I were to guess it definitely would be by PLC types (as with most other industrial protocols), in which case I would agree with your plan for having a table of string definitions for different PLC types.

kyle-github, 2021-02-19:
I will be adding more tag string attributes that will let the user override the PLC-specific settings.   That will let users test until they get the right settings for their specific PLC and then I can start adding those configurations to the table.

commit dced404e (2021-02-13): Fix #236, Modbus segfault.
commit d035dadd (2021-02-13): New raw byte API.  Fix #236.   Update Python wrapper.  Update Pascal wrapper. (#237)

* Fix #236.

* Add updated Python wrapper.

* Add updated Pascal wrapper.  Add raw byte accessors.

* Bump version for new raw byte API.