libplctag/libplctag #239: Read and Write to custom "Data Type" fields

Laro88, 2021-02-15:
Since this is my first experiments with libplctag then please bear with me.

Is there a way to get the "Data Type" for variables so I programmatically can an "AU" type? I need to drill into some special UDTs that are of "AU" type.

I have tried "list_tags.exe" program, the content of tag_type for the tag_name = "_21480_3E_3011_Out" is 36491.

I am looking at #231 that is evolving by the hour - so this might help me....

[image]

kyle-github, 2021-02-15:
Hi @Laro88, I think there might be a word missing in "It should be enough if I can get the "AU" out of the tag in some form."?

If you already know the path you need, then you can just access it via the tag name, "name=_21480_3E_3011_Out.Data", for instance.   If you want to know that a tag is of a certain type, the tag type is the right thing.   Even for UDTs like "AU" in your example, if you look at the lowest 12 bits of the type, that will be the UDT ID.   Every tag of that type (or of an array of that type) will have the same 12 bit ID (in this case 0xE8B).   You look at the high bit (16-bit word).  If that is "1" then you have either a system definition or a UDT.   You then look at bit 12 (counting from zero).  If it is zero, then you have a UDT.  Bits 13-14 are the array dimension count if it is an array.

If the type is not a UDT, then the lowest 8 bits gives you the atomic type.  Except (of course there is an except, this is Rockwell!) for single bit/bool tags.   Those are of type 0xC1 in the lowest 8 bits and the next 3 bits give the bit number within the byte.

If you want to get the UDT definition via the library, that is coming.   As you note issue #231 is tracking that.  However, that is tied to the new event-based core version of the library.   I plan to release that alongside the old, stable, version.   That said, it is likely to be a while before the event-based core is as stable as the current release version.   It just takes time to find all the places where unexpected results cause problems.   Once it is ready for testing I would love to have people use it to find those rough edges!

I hope this helps.

Laro88, 2021-03-09:
HI @kyle-github 
The latest event branch has a UDT name problem, or so it seems.

I just tried the 2.3.3 event branch on a production grade program (with IOcards missing though) - i am using ab-eip - not ab-eip2 as that fails. The program works fine on hardware with cards, and downloads nicely to the PLC.
One of the UDT names seems to be causing problems, here is a snippet of it.
Btw. I compiled with static int debug_level = PLCTAG_DEBUG_INFO;  in case it might help.

```
2021-03-09 08:21:26.061 thread(1) tag(24) INFO session_close_socket:757 Starting.
2021-03-09 08:21:26.061 thread(1) tag(24) INFO socket_close:557 Starting.
2021-03-09 08:21:26.061 thread(1) tag(24) INFO socket_close:654 Done.
2021-03-09 08:21:26.061 thread(1) tag(24) INFO socket_destroy:663 Starting.
2021-03-09 08:21:26.061 thread(1) tag(24) INFO socket_close:557 Starting.
2021-03-09 08:21:26.061 thread(1) tag(24) INFO socket_close:654 Done.
2021-03-09 08:21:26.061 thread(1) tag(24) INFO socket_destroy:684 Done.
2021-03-09 08:21:26.061 thread(1) tag(24) INFO session_close_socket:765 Done.
2021-03-09 08:21:26.061 thread(1) tag(24) INFO session_destroy:863 Done.
2021-03-09 08:21:26.061 thread(1) tag(24) INFO refcount_cleanup:445 Done.
2021-03-09 08:21:26.067 thread(1) tag(24) INFO ab_tag_destroy:854 Finished releasing all tag resources.
2021-03-09 08:21:26.067 thread(1) tag(24) INFO ab_tag_destroy:856 done
2021-03-09 08:21:26.067 thread(1) tag(24) INFO refcount_cleanup:445 Done.
2021-03-09 08:21:26.067 thread(1) tag(24) INFO plc_tag_destroy:1143 Done.
Using tag string "protocol=ab-eip&gateway=10.121.1.1&path=1,0&plc=ControlLogix&name=@udt/2867".
2021-03-09 08:21:26.074 thread(1) tag(0) INFO plc_tag_create:579 Starting
2021-03-09 08:21:26.074 thread(1) tag(0) INFO initialize_modules:192 Starting.
2021-03-09 08:21:26.074 thread(1) tag(0) INFO initialize_modules:262 Done.
2021-03-09 08:21:26.074 thread(1) tag(0) INFO find_tag_create_func:108 Matched protocol=ab-eip
2021-03-09 08:21:26.074 thread(1) tag(0) INFO ab_tag_create:176 Starting.
2021-03-09 08:21:26.074 thread(1) tag(0) INFO rc_alloc_impl:288 Starting, called from ab_tag_create:183
2021-03-09 08:21:26.074 thread(1) tag(0) INFO rc_alloc_impl:307 Done
2021-03-09 08:21:26.074 thread(1) tag(0) INFO session_create_unsafe:449 Starting
2021-03-09 08:21:26.074 thread(1) tag(0) INFO rc_alloc_impl:288 Starting, called from session_create_unsafe:457
2021-03-09 08:21:26.074 thread(1) tag(0) INFO rc_alloc_impl:307 Done
2021-03-09 08:21:26.074 thread(1) tag(0) INFO session_create_unsafe:556 Done
2021-03-09 08:21:26.074 thread(1) tag(0) INFO session_init:572 Starting.
2021-03-09 08:21:26.074 thread(1) tag(0) INFO session_init:587 Done.
2021-03-09 08:21:26.074 thread(1) tag(0) INFO setup_tag_listing:2539 Tag is not a tag listing request.
2021-03-09 08:21:26.074 thread(18) tag(0) INFO session_handler:982 Starting thread for session 0000019EE43CF890
2021-03-09 08:21:26.075 thread(1) tag(0) WARN cip_encode_tag_name:838 Unexpected character at position 4 in name string @udt/2867!
2021-03-09 08:21:26.075 thread(18) tag(0) INFO session_open_socket:605 Starting.
2021-03-09 08:21:26.075 thread(1) tag(0) WARN cip_encode_tag_name:844 Bad tag name format.  Tag must end with a bit identifier if one is present.
2021-03-09 08:21:26.075 thread(1) tag(0) WARN check_tag_name:984 parse of CIP-style tag name @udt/2867 failed!
2021-03-09 08:21:26.075 thread(1) tag(0) INFO ab_tag_create:457 Bad tag name!
2021-03-09 08:21:26.075 thread(1) tag(0) INFO set_tag_byte_order:3378 Starting.
2021-03-09 08:21:26.075 thread(1) tag(0) INFO set_tag_byte_order:3679 Done.
2021-03-09 08:21:26.075 thread(1) tag(0) WARN plc_tag_create:744 Error PLCTAG_ERR_BAD_PARAM while trying to create tag!
2021-03-09 08:21:26.075 thread(1) tag(0) INFO refcount_cleanup:433 Starting
2021-03-09 08:21:26.075 thread(1) tag(0) INFO ab_tag_destroy:813 Starting.
2021-03-09 08:21:26.075 thread(1) tag(0) INFO refcount_cleanup:433 Starting
2021-03-09 08:21:26.075 thread(1) tag(0) INFO session_destroy:776 Starting.
2021-03-09 08:21:26.075 thread(1) tag(0) INFO mem_move:199 Number of bytes to move is zero.
2021-03-09 08:21:26.076 thread(1) tag(0) INFO session_destroy:787 Session sent 0 packets.
2021-03-09 08:21:26.077 thread(18) tag(0) INFO session_open_socket:654 Done.
2021-03-09 08:21:26.082 thread(1) tag(0) INFO session_close_socket:757 Starting.
2021-03-09 08:21:26.082 thread(1) tag(0) INFO socket_close:557 Starting.
2021-03-09 08:21:26.083 thread(1) tag(0) INFO socket_close:654 Done.
2021-03-09 08:21:26.083 thread(1) tag(0) INFO socket_destroy:663 Starting.
2021-03-09 08:21:26.083 thread(1) tag(0) INFO socket_close:557 Starting.
2021-03-09 08:21:26.083 thread(1) tag(0) INFO socket_close:654 Done.
2021-03-09 08:21:26.083 thread(1) tag(0) INFO socket_destroy:684 Done.
2021-03-09 08:21:26.083 thread(1) tag(0) INFO session_close_socket:765 Done.
2021-03-09 08:21:26.083 thread(1) tag(0) INFO session_destroy:863 Done.
2021-03-09 08:21:26.083 thread(1) tag(0) INFO refcount_cleanup:445 Done.
2021-03-09 08:21:26.083 thread(1) tag(0) INFO ab_tag_destroy:854 Finished releasing all tag resources.
2021-03-09 08:21:26.083 thread(1) tag(0) INFO ab_tag_destroy:856 done
2021-03-09 08:21:26.083 thread(1) tag(0) INFO refcount_cleanup:445 Done.
Unable to open tag!  Return code PLCTAG_ERR_BAD_PARAM
Usage: list_tags <PLC IP> <PLC path>
Example: list_tags 10.1.2.3 1,0
2021-03-09 08:21:26.090 thread(1) tag(0) INFO ab_teardown:145 Releasing global AB protocol resources.
[... cut ...]
2021-03-09 08:21:26.095 thread(1) tag(0) WARN thread_join:185 null thread pointer.
2021-03-09 08:21:26.095 thread(1) tag(0) ERROR event_loop_teardown:130 Unable to join event loop thread, got error PLCTAG_ERR_NULL_PTR!
2021-03-09 08:21:26.095 thread(1) tag(0) INFO plc_module_teardown:1805 Starting.
2021-03-09 08:21:26.095 thread(1) tag(0) WARN mutex_destroy:235 null mutex pointer.
2021-03-09 08:21:26.095 thread(1) tag(0) WARN plc_module_teardown:1815 Error, PLCTAG_ERR_NULL_PTR, destroying plc list mutex!
```

kyle-github, 2021-03-09:
There is no support for the special UDT tags in `ab-eip`.  You must use `ab-eip2`.  What is failing with that?

Laro88, 2021-03-09:
Sorry for the confusion @kyle-github , I was & am using ab-eip2 in this test.
Here is a snippet (fresh build of list_tags.exe & run) with the problem as above. The PLC programmer in charge does not know what the @udt/2867 is, and I could not find t in a program export from Studio 5000.
```
2021-03-09 17:13:42.897 thread(1) tag(0) INFO setup_tag_listing:2539 Tag is not a tag listing request.
2021-03-09 17:13:42.897 thread(18) tag(0) INFO session_handler:982 Starting thread for session 0000029F22F94B30
->FIRST WARNING LINE 2021-03-09 17:13:42.897 thread(1) tag(0) WARN cip_encode_tag_name:838 Unexpected character at position 4 in name string @udt/2867!
2021-03-09 17:13:42.897 thread(18) tag(0) INFO session_open_socket:605 Starting.
2021-03-09 17:13:42.897 thread(1) tag(0) WARN cip_encode_tag_name:844 Bad tag name format.  Tag must end with a bit identifier if one is present.
2021-03-09 17:13:42.897 thread(1) tag(0) WARN check_tag_name:984 parse of CIP-style tag name @udt/2867 failed!
2021-03-09 17:13:42.897 thread(1) tag(0) INFO ab_tag_create:457 Bad tag name!
```

The last couple of lines of the program run is here:
````
2021-03-09 17:13:42.917 thread(1) tag(0) INFO event_loop_teardown:123 Starting.
2021-03-09 17:13:42.917 thread(1) tag(0) WARN thread_join:185 null thread pointer.
2021-03-09 17:13:42.917 thread(1) tag(0) ERROR event_loop_teardown:130 Unable to join event loop thread, got error PLCTAG_ERR_NULL_PTR!
2021-03-09 17:13:42.917 thread(1) tag(0) INFO plc_module_teardown:1805 Starting.
2021-03-09 17:13:42.917 thread(1) tag(0) WARN mutex_destroy:235 null mutex pointer.
2021-03-09 17:13:42.917 thread(1) tag(0) WARN plc_module_teardown:1815 Error, PLCTAG_ERR_NULL_PTR, destroying plc list mutex!
```

kyle-github, 2021-03-11:
It looks like you are using the protocol `ab-eip` instead of `ab-eip2` for the UDT calls.   Or perhaps I am somewhere!  I will try to investigate this weekend when I get a break from work.

The errors above are coming from the old code and it does not understand the `@udt/xyz` tag names.

kyle-github, 2021-03-17:
I have updated the `event` branch with a fix in the code when closing a socket.   This might help here too if you have a chance to try it.

Please use the `list_tags` C program for now.

kyle-github, 2021-09-28:
This has been released in the main release branch with [release 2.4.0](https://github.com/libplctag/libplctag/releases/tag/v2.4.0)!

closed 2021-09-28 by kyle-github