libplctag/libplctag #247: Garbage values being written - MicroLogix 1400

xinthose, 2021-02-22:
Version: 2.3.3
OS: Ubuntu 20.04 LTS
PLC: MicroLogix 1400
Data Type: Float

Did something change recently in the library concerning writing floats?  I was using an old version of the library with no issues, then upgraded, now I'm seeing garbage values being written to my PLC:

[image]

**Code**
`fvec` is a vector of floats, `element_size` is 4, `tag` is a vector of tags

```
		for (size_t i = 0; i < fvec.size(); i++)
		{
			status = plc_tag_set_float32(tag.at(tag_num), (i * element_size), fvec.at(i)); // tag, offset, value
			if (status != PLCTAG_STATUS_OK)
			{
				ERROR << "tag.at(" << tag_num << ") >> plc_tag_set_float32 >> error = " << plc_tag_decode_error(status);
				throw -2;
			}
		}

		// write data
		status = plc_tag_write(tag.at(tag_num), timeout); // timeout (ms)
		if (status != PLCTAG_STATUS_OK)
		{
			ERROR << "tag.at(" << tag_num << ") >> plc_tag_write >> error = " << plc_tag_decode_error(status);
			throw -3;
		}
```

kyle-github, 2021-02-23:
Hmm, that was about when the byte order code went in.   I had an interesting time getting it working correctly for PLC-5 and had someone testing MicroLogix since I do not have one.   Frustrating if I have this wrong now!

There are new attributes that can control the byte order of the individual types.   Can you please try using the extra tag attribute `float32_byte_order`?   It would be like this:

`protocol=ab-eip&gateway=....&elem_size=4&elem_count=X&name=F8:0&float32_byte_order=0123`

The string `0123` defines the byte order.   For PLC-5, it is in a weird mishmash of big and little endian, `2301`.

Please try different values.   The most likely ones are `0123` to match ControlLogix or `2301` to match PLC-5.   

Each digit in the string denotes which byte in the data will be used for that byte in the result.  "Pure" little-endian will be `0123`.

You can use `tag_rw` to avoid having to do this in code.

Thanks very much for bringing this to my attention!

kyle-github, 2021-02-23:
The current definitions right now for all PCCC-based PLCs are:

```c
    .int16_order = {0,1},
    .int32_order = {0,1,2,3},
    .float32_order = {2,3,0,1}, /* yes, it is that weird. */
```

Note that _only_ the 32-bit floating point definition is odd on PLC-5 PLCs.   The 32-bit integers are the L data file and do not exist on PLC-5 IIRC.

kyle-github, 2021-02-23:
If it turns out that I have the wrong mapping for MicroLogix, I will make a new release with the correct mapping.

xinthose, 2021-02-23:
@kyle-github OK.  Thank you for your work Kyle!  I can test with your new release.

kyle-github, 2021-02-23:
I am a bit confused.   You will test with the new attributes or it worked with some value other than `2301`?  Sorry.  No caffeine yet this morning...

xinthose, 2021-02-23:
@kyle-github if you see the issue, I will wait for your next release, test with that, and get back to you

kyle-github, 2021-02-23:
I cannot change anything without some feedback from someone who has a MicroLogix.   That is why I was hoping for some help above :-)  I will reach out on the forum to see if someone with a MicroLogix can test.

xinthose, 2021-02-23:
@kyle-github OK.  So you want me to try v2.3.3 with this on the end of the tag: `&float32_byte_order=0123`?

xinthose, 2021-02-23:
A MicroLogix costs about $200 on eBay.  I would be willing to give your PayPal some cash if you want.

GitHubDragonFly, 2021-02-23:
Are you using Modbus to access your MicroLogix 1400 or AB protocol?

xinthose, 2021-02-23:
@GitHubDragonFly AB protocol

GitHubDragonFly, 2021-02-23:
Kyle normally has those byte_order specifiers for Modbus.

I just tried version 2.1.22 and it works without byte_order.

This is broken starting with version 2.2.0.

GitHubDragonFly, 2021-02-23:
This works fine in Windows with version 2.2.1, so give it a try:

tag_rw -t real32 -p "protocol=ab-eip&gateway=192.168.1.10&cpu=micrologix&elem_size=4&elem_count=1&name=F8:100&float32_byte_order=0123" -w "12.57"

or just correct your string to include &float32_byte_order=0123 at the end.

xinthose, 2021-02-23:
@GitHubDragonFly  I see.  I don't actually need the newest features, but I like to update my application's libraries occasionally.  I will remember this and wait to see what Kyle does.

kyle-github, 2021-02-23:
Thanks @GitHubDragonFly for testing.   This is doubly frustrating because I asked for testers with MicroLogix to check this out when I was doing the ModBus additions since I suddenly had to support big-endian data and I got a bug report on floats with PLC-5 systems.  At the time, I got a couple of positive test reports with MicroLogix.   Sigh.

I added all the special `XYZ_byte_order` overrides at that same time because there is no consistent mapping for values from ModBus PLCs.   I did not think I would use them for Allen-Bradley!   I wonder how many other byte order values are wrong?

I will roll out another release with MicroLogix and SLC (they use the same PCCC protocol commands, unlike PLC-5) set to using byte order `0123`.

I will look into another hardware resource.   The company that I used to be able to borrow stuff from went out of business due to Boeing's business troubles.

closed 2021-02-24 by kyle-github

kyle-github, 2021-02-24:
@xinthose can you please test with 2.3.4 to see if it solves your problem?   Do either of you have SLCs?  I looked on eBay and other used PLC sites and SLCs are pretty expensive.  I thought they would be cheap because they have been more or less dropped by AB, but no.

xinthose, 2021-02-24:
@kyle-github Hello.  I just tested with my test program and v2.3.4, it worked without any issues.  These are the numbers I wrote and read back without problems:
```
fvec.push_back(123456);
fvec.push_back(1);
fvec.push_back(-123.456);
fvec.push_back(7777.77);
fvec.push_back(123.456);
```

kyle-github, 2021-02-24:
Thanks!   Can you check in RSLogix that the values you pushed are what you see in the GUI?   Or, you can write with the old version of the library and read with the new and vice versa.   One thing that tripped me up in initial testing was that I was using the same version of the library to read and write.  The _library_ was happy about the format, but RSLogix showed me the error of my ways :-/

Also, you do not need to have `path` or `share_session` elements in your tag strings.   Sessions are shared by default and paths are ignored (except for DH+ and bridging) for PCCC-based PLCs.

Thanks again for testing.

xinthose, 2021-02-24:
@kyle-github I see.  Yes, I was live with the PLC on RSLogix 500 and could see the values.

kyle-github, 2021-02-24:
Great!   Thanks.   That proves that the format was right.

GitHubDragonFly, 2021-02-25:
Unless this is intended for MicroLogix, you might need to check your int32 (and possibly int64).

In my test, it doesn't return 2 consecutive values correctly (the first one is correct but the second one is the default min, a possible reason why tag_rw only returns the first value).

This was tested with 2.2.1 and I assume not much has changed for integers in 2.3.4 (which I can eventually test).

GitHubDragonFly, 2021-02-25:
It does actually return the second number if I double the elem_count, so with elem_size=4 it has to be elem_count=4 in order to get to sint32 values with tag_rw.

commit b6ea3a56 (2021-02-24): Fix for MicroLogix float byte order.  Fix #247. (#248)

* Split out byte order for SLC/MicroLogix tags as floats are not the same as in PLC-5.