libplctag/libplctag #252: Ability to read type information from ab eip_cip tags

davitz, 2021-03-10:
ab eip_cip tags provide type information when you read from them. Currently this information is read into ab_tag_t.encoded_type_info, but isn't exposed in any of the higher level APIs. When working with these types of PLCs, being able to easily access this information would eliminate the need to know the type of the tag before reading it, which is pretty convenient.

Perhaps we could use this information to automatically fill the elem_type property for these types of tags after a read, or expose it in some other way.

davitz, 2021-03-10:
> Not an authoritative answer: is using the tag listing `@tags` feature suitable? It returns a tag type code which can be used to generate tag objects at runtime.

This is a valid option, it just seems a bit circuitous to be separately fetching the list and searching for my tag name to find its type when the tag itself is already sending back type information on every read that I could be using.

kyle-github, 2021-03-10:
Hi, thanks for bringing this up.

As @timyhac said, you can pull all the tag information with `@tags` and soon `@udt`.

The information you get back from the controller is partial.   You get type information _only_ for basic types like DINT.   You do not get such information for arrays or UDTs.  You get variable information depending on whether you access a single bit within a SINT or a field within a UDT or a element of an array.   Arrays and UDTs are particularly annoying because at least UDTs have a unique 12-bit ID that _could_ have been returned.   But Rockwell chose not to.

It also is not consistent across types of PLCs.   For instance on Omron, when you pull a tag that is an array, you get the whole array _even if you request just one element_.   

So, unfortunately, it is not at all generally possible.

davitz, 2021-03-10:
Even if the information can't perfectly predict the type in every situation, wouldn't it still be useful to at least make the type information that is delivered accessible to the application that's using the library so that if it is expecting atomic types it can use that information to determine _which_ atomic type it has received?

kyle-github, 2021-03-11:
I have been thinking about setting up an integer attribute on tags where it is possible to know the type.  This would include many (not all) types such as DINT, INT, STRING etc. where we do get data.  What has stopped me so far are the following:

- Only some types have usable information returned from the PLC.   
- In some cases it appears to be _model_ specific.   It is not clear that the same information can be trusted on Micro800 for instance.   Or Omron.
- The information is available in other ways for some PLCs, such as PLC-5 with the data file type.
- Modbus gets nothing at all beyond registers and coils.
- S7 looks like it will not get much either.

This means that it would be code very specific to the specific type and model of the PLC.   

One alternative is to simply pass through anything I get to the application without any interpretation.  That puts a lot of burden on any wrappers or applications to write duplicate code.

davitz, 2021-03-11:
As a start, passing through the raw data would allow anyone with specific knowledge about the models they're working with to make use of that information, which would be a good start.

From there I suppose we could expose a second value that gives an interpreted type identification where possible or some indication that the value is "unknown" where it's not possible or where we don't yet know / haven't yet implemented the interpretation.

That way users who are connecting to model types that play nicely can easily read what type to expect, but others who use more troublesome models will see that the type was not able to be determined automatically and will have to continue reading it the old way. And if someone knows enough to be able to interpret raw type information that the library wasn't able to, it's at least available for them to do so.

timyhac, 2021-03-11:
I'm really interested to find out what the specific use case is on this is, there could be some interesting ramifiations for the wrappers.

I don't understand how you would know the tag name but not its type.

davitz, 2021-03-12:
My specific case is that my organization maintains a system that gathers data from AB Logix PLCs at our customer sites in order to provide analytics.

Currently our software is using the InGear NetLogix drivers. For reasons I won't elaborate on, we want to move off of those drivers. In my research I found that this library meets almost all our needs. The one problem is that our old drivers never needed us to specify type information for numeric types as it used the CIP type information to automatically return the correct types to us. As a result, the existing configurations that we have for hundreds of tags that we already connect to do not specify type, instead they just used the type information that the driver provided to us.

Because of this, in order for us to port over to use this library as-is we would need to go through each of the tags that our system already connects to, learn their types and add those types as a new configuration option first so that when we make the change, the software will have the type information necessary to connect via libplctag.

To me it makes much more sense to modify the library to have this feature than it does to revisit all of those tags. For that reason I have already made the enhancement on my own fork to the extent that it would work for our purposes. I opened this issue to begin the process of potentially merging the feature back into the main project so that we can eventually move back to the official build rather than sticking to my separate version and so that others in a similar position can make use of it.

Based on our discussion I can tell that I'll need to make further changes to my version of the feature before it could be merged, I'd just like to get agreement that it would be a good addition and what form it should take so I can make the necessary changes.

kyle-github, 2021-03-15:
Thanks for the background @davitz, this makes sense.

How do you handle tags that are not DINT etc.   I.e. UDT tags?   In those cases, AB does _not_ send back data that is unique.  If you have two UDTs with the same fields, you will get the same hash back.

davitz, 2021-03-15:
We don't currently work with any UDT tags.

The most complex type that we read currently is string which we do need to specify in our configuration. The rest of the types we work with including boolean and various numeric types are all detected when we read the tag meaning we don't have to specify if it's an integer or a floating point or a boolean in our configuration.

It looks like the library we're currently using just detects the tag type as "object" in the case of a UDT and requires you to give it a struct to convert the value in that case.

kyle-github, 2021-03-15:
If I remember correctly, strings get a fixed "UDT" ID.  A lot of system types are like that.   I will try across my PLCs and see what I get.   I have an old L55, a L61 and a L81.  The old L55 has firmware that is pretty old.   Version 16 or something?  I can't remember.

Questions:

- Other than tag type, is there other metadata that you need?  I.e. without doing a full tag listing.   For instance, all that is really required to read a single element from a tag is the tag name (on Control/CompactLogix).  From the response back you can get the element size (since you got one element).   Unfortunately that does not work on Omron since the entire tag comes back (which could be large as tag can be arrays).
- What about array vs. not array?   We do not get the array element count, but we do get some information if a tag is an array.

davitz, 2021-03-15:
For my existing PLC connections all that I need is the element type.

When connect to array tags we always specify the specific indices that we need so we use tag names like "Name[2]" which gives us the individual value that we need. That being said, if CIP gives us the type, then comparing the size of the data to the size of that type should tell us that it's an array, and also its length.

We do not connect to any Omron at this point so I'm not familiar with its behavior. If I'm understanding you correctly, are you saying that Omron will always reply with the entire array even if you give it a tag name with an index? If that's correct then the options would probably be (A) treat the result the same as if we requested an entire array from a different model, or (B) parse the name ourselves to see if the caller was looking for a specific index and get that value for them.

kyle-github, 2021-03-16:
Yeah, Omrom PLCs always return the entire tag.  You must pass a count (elem_count) of 1, always.  And that gets the entire tag.  I do not have one, so I am not sure what happens if you access a single element (myTag[42]).

ronytigo, 2023-01-09:
So is there a way to get a tag type?

joed74, 2023-06-30:
> Is getting the tag element type still of interest?

Yes, please. A function like 

`LIB_EXPORT int plc_tag_get_raw_typeinfo(int32_t id, uint8_t *buffer, int buffer_length);`

would be nice. 

If the underlying system has no typeinfo at all the function could return `PLCTAG_ERR_UNSUPPORTED`
If the returned tag has no typeinfo the function could return `PLCTAG_ERR_NO_DATA`

kyle-github, 2024-06-10:
This is fixed in 2.6.0. That has not released yet, but soon.   The fix is to add another API function for getting byte array attributes.  Also, there is a new naming convention used for getting the length of a byte array attribute "byte_array_attr_name**.length**".

The new functionality is available on the prerelease branch now.

The new API function is:
```c
int plc_tag_get_byte_array_attribute(int32_t tag, const char *attrib_name, uint8_t *buffer, int buffer_length);
```

closed 2024-06-10 by kyle-github

timyhac, 2024-06-10:
Wow!