libplctag/libplctag #253: Throughput & safety of using multiple threads using plc_tag_get_...

Laro88, 2021-03-11:
Is the plc library safe to use in a multi threaded environment provided that there never is two threads accessing the same tag?

Right now I measure ~8ms round trip time pr read, however since there are thousands of tags things become a bit slow. So I would like to use threads to do the refreshes in parallel - is this ok?

kyle-github, 2021-03-11:
Yes, the library is safe to use from threads.  There are several example programs using threads to demonstrate the use.

All the main API functions are mutex-protected.

However, the best way to get performance is to use async mode.  Look at the async example programs.  Threads work as well but many platforms do not do all that well with large numbers of threads.   Look at the stress test example program for one of the core tests I run.

The library itself will multiplex requests onto one TCP connection per PLC and compact multiple requests into one packet to the PLC.  At least if the PLC is a Control- or CompactLogix.

I hope this helps!

kyle-github, 2021-03-17:
Please reopen if you still have questions!

closed 2021-03-17 by kyle-github