libplctag/libplctag #254: ImportError: dynamic module does not define module export function (PyInit_libplctag)

hussainkadori, 2021-03-12:
Hi,
I have an import error with the python wrapper:

```
from plctag import libplctag
ImportError: dynamic module does not define module export function (PyInit_libplctag)
```
I am using python 3.9.2

GitHubDragonFly, 2021-03-12:
It should be:   from libplctag import *

Is "plctag" something that you created?

You should check this project to make sure that you know how to reference things:

   https://github.com/GitHubDragonFly/PythonTest

hussainkadori, 2021-03-12:
@GitHubDragonFly I test your gui and I have same error:
```
    from libplctag import *
ImportError: dynamic module does not define module export function (PyInit_libplctag)
```

hussainkadori, 2021-03-12:
I solved the problem when I build the library on fedora there is two files **libplctag.so** and **libplctag.so.2.3** , I was loading the libplctag.so instead of libplctag.so.2.3. 

Thanks

kyle-github, 2021-03-17:
Please reopen if you still find a problem!

closed 2021-03-17 by kyle-github