libplctag/libplctag #255: Raspberry Pi 2 - Compiling plctag library and Running python GUI in Raspbian

GitHubDragonFly, 2021-03-18:
This is just an informational post for those who would like to try it.

The attached zip file has 3 files inside, one of which is a text file showing commands and some outputs in chronological order.

@kyle-github should check some warnings and extra brackets that are in the code, which didn't prevent the library from compiling successfully.

This was all done directly in Raspbian and some users might need to install python and/or python3 if not already there.

The libplctag.py file was modified to only look for and load the compiled plctag library, directly from the /usr/local/lib/ folder.

[RPi plctag.zip](https://github.com/libplctag/libplctag/files/6161488/RPi.plctag.zip)

timyhac, 2021-03-18:
Hi @GitHubDragonFly nice job!

Do you know if either of the ARM builds work on that device?
https://github.com/libplctag/libplctag/releases/tag/v2.3.5

I did some testing with Raspbian on a Raspberry Pi 3b, but I believe the CPUs are different.

kyle-github, 2021-03-18:
Thanks!  I will look at this this weekend.   I have a RPi 400 now but I do not have it fully set up to use an external USB drive (I find RPis too fragile when using the SD card).   The 1 and 2 use a different processor than the 3, 4 and 400.   I took what I found online to create the existing ARM binary builds.   I should get a RPi 2 to make sure it all works directly.

GitHubDragonFly, 2021-03-18:
Both arm6hf and arm7l experimental libraries seem to work fine.

I did update the text file to reflect using either of those libraries as well.

GitHubDragonFly, 2021-03-19:
@kyle-github related to one of the warnings produced in the output, which can be seen in the text file from my initial post, does this discussion help you any or is the warning produced due to the processor itself:

  https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=861026
  https://github.com/cockpit-project/cockpit/pull/6374/files

kyle-github, 2021-03-20:
long long error: Ah, looks like the field is not 64-bit on a 32-bit RPi.  I will try to get my RPi 400 fully set up this weekend and fix this.  Could be a bit interesting as I am not sure what the portable value is going to be here.  

The strncpy error is one that I see sometimes, but I will try to fix it.  I do not see it on my system.  I wonder why?   It could be a false positive, but usually these should be checked.  I am doing some questionable casting there (size_t to int) and that itself is not a great idea (breaks if your strings are >2gb).

Thanks for pointing these out!

kyle-github, 2021-03-20:
I opened issue #257 to track the compiler warnings that remain.   Yuck.   I love working around compiler bugs.

I was able to test the ARM6 and ARM7 builds on my RPi 400 running in 32-bit mode.   Worked fine!   I need to figure out how to get it to boot 64-bit ARM Debian or Raspian as well and then I can test the 64-bit ARM build.

kyle-github, 2021-03-20:
Now that I have a working RPi I can start getting the release for that platform much more solid.

GitHubDragonFly, 2021-03-21:
@kyle-github you can close this whenever you want.

kyle-github, 2021-03-21:
I mean to, but then realized that this still has your Python updates in it.   I need to look those over.

GitHubDragonFly, 2021-03-22:
Those python modifications are not really an update but specifically designed for the Pi.

The search path on the Pi is the issue for the original libplctag.py code.
It was reaching the point of trying to load the library but would error by stating that folder or file couldn't be found.

Some suggestions on the Internet asked to append the path before python starts but that wasn't really an option I wanted to go with.

The modified code should work for both 32 and 64 bit libraries for as long as they are found in the /usr/local/lib folder.

GitHubDragonFly, 2021-04-14:
Besides for the Raspbian Buster OS, it appears that Kali Linux has a rather high potential on the Pi since it appears to be well organized and updated. This would be true for 32-bit on Raspberry Pi 2 and hopefully should be as good for 64-bit on Pi 3 or 4.

kyle-github, 2022-11-11:
Need to retest all this when I can find my RPi (had the kitchen rebuilt and now the house is in complete disorder!).