libplctag/libplctag #258: Multi tag scheduled read on MODBUS gives PLCTAG_STATUS_PENDING after tag creation

atifali, 2021-04-11:
So I am setting up about 27 odd MODBUS sync reads (500 ms) over a simulated MODBUS PLC. In my program, once I setup the tags using `plc_tag_create()`, before performing the first read I check tag statuses using the following logic:

```
/* wait for all tags to be ready */
while(check_tags() == PLCTAG_STATUS_PENDING){
    pdebug(DEBUG_INFO, "Waiting for tags to be ready...");
    util_sleep_ms(1);
}
```
It seems like the tags are just stuck with the `PLCTAG_STATUS_PENDING` status even though I can see packets being sent and received in the debug log.

I also have a feeling that this is something related to issue #188. Any ideas as to what could be wrong here?

GitHubDragonFly, 2021-04-15:
If you have a Windows based computer then you can test the Modbus functionality of the libplctag by using this project:

https://github.com/GitHubDragonFly/AB-Modbus-Master

It is currently configured with libplctag v2.1.22 but has instructions on how to change the library files.

atifali, 2021-04-15:
@GitHubDragonFly I will check this out for sure.

However, I do have a MODBUS simulator that my current setup is based around and was just wondering why the tags stayed in the pending status once I had more than a certain amount of tags being sync read. 

From the logs it looks like the first tag created never actually leaves the pending state.

I do not have the issue when lets say I am only sync reading 10 tags.

GitHubDragonFly, 2021-04-16:
I cannot be smart about that.

My program allows up to 12 tags and here is the picture showing automatic reading of all 12 tags with the simulator.

[image]

kyle-github, 2021-04-16:
Sorry for the delays real world life is getting in the way right now, and thanks @GitHubDragonFly for the tips.

This does sound like the same problem as #188.   What version are you using?   If it is a later version (i.e. this year), can you please capture a log dump?   In order to do that you need to set the log level to 4 and capture the output.  This can be done in the tag string by appending "&debug=4" or by calling the plc_tag_set_debug_level(PLCTAG_DEBUG_DETAIL) function.

kyle-github, 2021-06-16:
Any update on the logs on this?

kyle-github, 2021-09-19:
I have a possible fix for #279 that might help here as well.   Clearly I need to spend some more time with Modbus to make it more solid.

kyle-github, 2021-09-24:
Please let me know if this works any better.  If it is still a problem, please reopen this issue.   Thanks!

closed 2021-09-24 by kyle-github