libplctag/libplctag #259: iotedge module runtime error  Access to the path '/app/libplctag.so' is denied.

nainesh72, 2021-04-13:
Hi
I create iotedge module on vs 2019 for linux arm, I user nuget package libplctag, libplctag.NativeImport i  work fine on iotedge simualtion but when i deploy on my rasbary pi (os rasbarian ) it give me runrime erro 
my vs 2019 is on my desktop it has win10 os
IoT Hub module client initialized.
System.UnauthorizedAccessException: Access to the path '/app/libplctag.so' is denied.
 ---> System.IO.IOException: Permission denied
   --- End of inner exception stack trace ---
   at Interop.ThrowExceptionForIoErrno(ErrorInfo errorInfo, String path, Boolean isDirectory, Func`2 errorRewriter)
   at Microsoft.Win32.SafeHandles.SafeFileHandle.Open(String path, OpenFlags flags, Int32 mode)
   at System.IO.FileStream.OpenHandle(FileMode mode, FileShare share, FileOptions options)
   at System.IO.FileStream..ctor(String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize, FileOptions options)
   at System.IO.FileStream..ctor(String path, FileMode mode, FileAccess access, FileShare share)
   at System.IO.File.InternalWriteAllBytes(String path, Byte[] bytes)
   at System.IO.File.WriteAllBytes(String path, Byte[] bytes)
   at libplctag.NativeImport.LibraryExtractor.ExtractAppropriateLibraryToDirectory(String outputDirectory)
   at libplctag.NativeImport.LibraryExtractor.Init(Boolean forceExtract)
   at libplctag.NativeImport.plctag.ExtractLibraryIfRequired()
   at libplctag.NativeImport.plctag.plc_tag_create(String lpString, Int32 timeout)
   at libplctag.NativeTag.plc_tag_create(String lpString, Int32 timeout)
   at libplctag.NativeTagWrapper.Initialize()
   at libplctag.Tag.Initialize()
   at libplctag.Tag`2.Read()
 in app folder list is
cd ..
moduleuser@4844abf76be7:/app$ ls -l
total 6964
-rwxr--r-- 1 root       root        175184 Oct  8  2018 DotNetty.Buffers.dll
-rwxr--r-- 1 root       root         34896 Oct  8  2018 DotNetty.Codecs.Mqtt.dll
-rwxr--r-- 1 root       root        139856 Oct  8  2018 DotNetty.Codecs.dll
-rwxr--r-- 1 root       root        153168 Oct  8  2018 DotNetty.Common.dll
-rwxr--r-- 1 root       root         51280 Oct  8  2018 DotNetty.Handlers.dll
-rwxr--r-- 1 root       root        180816 Oct  8  2018 DotNetty.Transport.dll
-rwxr--r-- 1 root       root        392568 Sep 19  2020 Microsoft.Azure.Amqp.dll
-rwxr--r-- 1 root       root        698256 Mar 22 21:24 Microsoft.Azure.Devices.Client.dll
-rwxr--r-- 1 root       root         37760 Jan 21 20:34 Microsoft.Azure.Devices.Shared.dll
-rwxr--r-- 1 root       root         20360 Oct 19 18:48 Microsoft.Extensions.Configuration.Abstractions.dll
-rwxr--r-- 1 root       root         28040 Oct 19 18:48 Microsoft.Extensions.Configuration.Binder.dll
-rwxr--r-- 1 root       root         19848 Oct 19 18:49 Microsoft.Extensions.Configuration.EnvironmentVariables.dll
-rwxr--r-- 1 root       root         25480 Oct 19 18:50 Microsoft.Extensions.Configuration.FileExtensions.dll
-rwxr--r-- 1 root       root         24960 Oct 19 18:51 Microsoft.Extensions.Configuration.Json.dll
-rwxr--r-- 1 root       root         30600 Oct 19 18:48 Microsoft.Extensions.Configuration.dll
-rwxr--r-- 1 root       root         35320 Nov 14  2016 Microsoft.Extensions.DependencyInjection.Abstractions.dll
-rwxr--r-- 1 root       root         16776 Oct 19 18:48 Microsoft.Extensions.FileProviders.Abstractions.dll
-rwxr--r-- 1 root       root         35720 Oct 19 18:49 Microsoft.Extensions.FileProviders.Physical.dll
-rwxr--r-- 1 root       root         39304 Oct 19 18:48 Microsoft.Extensions.FileSystemGlobbing.dll
-rwxr--r-- 1 root       root         44032 Feb 17  2017 Microsoft.Extensions.Logging.Abstractions.dll
-rwxr--r-- 1 root       root         18432 Feb 17  2017 Microsoft.Extensions.Logging.dll
-rwxr--r-- 1 root       root         39296 Oct 19 18:48 Microsoft.Extensions.Primitives.dll
-rwxr--r-- 1 root       root        777800 Sep 15  2018 Microsoft.WindowsAzure.Storage.dll
-rwxr--r-- 1 root       root        693680 Nov  9  2019 Newtonsoft.Json.dll
-rwxr--r-- 1 root       root        380848 Nov 20  2017 System.Configuration.ConfigurationManager.dll
-rwxr--r-- 1 root       root         24880 Jul 19  2017 System.Security.Cryptography.ProtectedData.dll
-rwxr--r-- 1 root       root        351624 Oct 19 18:47 System.Text.Json.dll
drwxrwxrwx 2 moduleuser moduleuser    4096 Apr 13 19:14 echo
-rwxr-xr-x 1 root       root         90712 Apr 13 19:51 eip3
-rw-r--r-- 1 root       root         92141 Apr 13 19:51 eip3.deps.json
-rw-r--r-- 1 root       root         11264 Apr 13 19:51 eip3.dll
-rw-r--r-- 1 root       root         12800 Apr 13 19:51 eip3.pdb
-rw-r--r-- 1 root       root           146 Apr 13 19:51 eip3.runtimeconfig.json
-rwxr--r-- 1 root       root       2328576 Mar 20 22:54 libplctag.NativeImport.dll
-rwxr--r-- 1 root       root         41472 Mar  7 05:21 libplctag.dll
drwxr-xr-x 3 root       root          4096 Apr 13 19:51 runtime
drwxr-xr-x 3 root       root          4096 Apr 13 19:51 runtimes
moduleuser@4844abf76be7:/app$

please guide us
nainesh

timyhac, 2021-04-13:
The problem is that the user this application is running under does not have permission to [extract libplctag.so to disk](https://github.com/libplctag/libplctag.NET#libplctagnativeimport).

Either you've hit [this issue](https://github.com/libplctag/libplctag.NET/issues/137), or the user needs additional write permissions to that directory. You can also [supply the dll/so yourself](https://github.com/libplctag/libplctag/releases) as part of the deployment.

nainesh72, 2021-04-13:
Hi
I all ready copy this file in module app folder but still same problem

[image]

timyhac, 2021-04-13:
OK - please read the documentation on libplctag.NativeImport here: https://github.com/libplctag/libplctag.NET#libplctagnativeimport

You need to set ForceExtractLibrary = false

nainesh72, 2021-04-14:
Hi
i not understan why module need linux base file where my module are run in container not on base os, it is iotedge module and my othre window base dll work fine

timyhac, 2021-04-14:
Sorry - could you be more specific please. Does your issue relate to libplctag specifically, or is about which platform you are targeting?

nainesh72, 2021-04-15:
Hi
Thanks for Your Support, Now My problem is solved, I explain what i do and where is problem
1 I create container(docker) application in my window machine,  but on window pc i use linux container, on when i run this application from my window pc with use of linux container it work fine but when i deploy this container on base linux os and linux container it need .so file so i change in my docker file,  I write copy command to copy .so file in my container then i change permission in my docker file like this
[image]

then i deploy my container on linux container on linux on then it work fine

nainesh72, 2021-04-15:
Thanks for support

closed 2021-04-15 by nainesh72