libplctag/libplctag #260: c++ plc_tag_write and plc_tag_read functions are slow! 50-60ms.

AdonJP, 2021-04-14:
I am using C++ wrapper for this library and everything works good and is fast, except for the functions:
plc_tag_write
and
plc_tag_read

Right now the operations are taking about 40-60 ms.

I am able to ping the PLC < 1ms so am surprised that the operation is taking this long.

Please advise.

Adon

kyle-github, 2021-04-14:
Hi, thanks for bringing this up.   I'll need to get some questions answered before I can help:

- What kind of PLC are you using?
- What version of the library are you using?
- What OS are you using?
- Can you share the line of code where plc_tag_read/write are called?

Some PLCs can take that long to respond.  Particularly older PLC-5, SLC or MicroLogix are very slow.  If you are using a ControlLogix you can see very slow response times if you have an ENBT and it is heavily overloaded.

AdonJP, 2021-04-15:
Hi, Thanks for the prompt response.

- The PLC is a we are communicating with is a Controllogix 5000 (1756-L72 ControlLogix 5572 controller)
- I cannot find the version of the library listed anywhere explicitly, but I pulled the code off git within the past few months so I imagine 2.3. How can I confirm this?
- Windows 10 x64
```
- rc = plc_tag_write(tag_, TIMEOUT); where tag_ is a std::vector<uint16_t> size : 1 element, TIMEOUT = 5000
- rc = plc_tag_read(tag_, TIMEOUT);
```
My code was taken near exactly from the cpp wrapper example located here "\libplctag\src\wrappers\cpp". As expected, the other parts from the example IE. function 
`plc_tag_get_uint16(tag_, i * elem_size);`
only take about 1ms to complete.

Thanks for the information regarding the ENBT I will look into this - I think we are using one of these. Is there an AB replacement you know of we can replace the ENBT with?

kyle-github, 2021-04-25:
Sorry for the delay in response.   Lots of things happening at work and with family right now.

If you want to replace the ENBT and cannot afford to move to the L8x series CPU modules, I suggest using EN2T modules.   They are a bit better (still not great).   Any network that is not part of the CPU module will be limited by the bandwidth and latency of the backplane in the chassis.

That said, I know of many systems using ENBT that do not show this kind of latency at all.   I have ENBT test racks and I get less than 20ms latency most of the time, over WiFi.   Here is a sample command line:

`./tag_rw2 --type=sint32 '--tag=protocol=ab-eip&gateway=10.206.1.39&path=1,0&plc=controllogix&name=TestDINTArray' --debug=4`

This responds in about 15ms on my laptop using a rack with an ENBT and a L55 (very old!) CPU.   Using a similar tag string and aiming at a L81 CPU (which has its own 1Gbps Ethernet) I get 11ms.

Both of these test are from my laptop over WiFi.   The ping times are on the order of 10ms, so most of the time is that overhead.

If I go to my desktop which is connected to my PLC racks via a 1Gbps switch:

```text
kyle@homesrv2:~/Documents/Projects/libplctag/libplctag/build$ ping 10.206.1.39
PING 10.206.1.39 (10.206.1.39) 56(84) bytes of data.
64 bytes from 10.206.1.39: icmp_seq=1 ttl=64 time=1.44 ms
64 bytes from 10.206.1.39: icmp_seq=2 ttl=64 time=0.654 ms
64 bytes from 10.206.1.39: icmp_seq=3 ttl=64 time=0.683 ms
64 bytes from 10.206.1.39: icmp_seq=4 ttl=64 time=0.751 ms
64 bytes from 10.206.1.39: icmp_seq=5 ttl=64 time=0.653 ms
64 bytes from 10.206.1.39: icmp_seq=6 ttl=64 time=0.658 ms
^C
--- 10.206.1.39 ping statistics ---
6 packets transmitted, 6 received, 0% packet loss, time 5092ms
rtt min/avg/max/mdev = 0.653/0.806/1.441/0.285 ms
kyle@homesrv2:~/Documents/Projects/libplctag/libplctag/build$ ping 10.206.1.40
PING 10.206.1.40 (10.206.1.40) 56(84) bytes of data.
64 bytes from 10.206.1.40: icmp_seq=1 ttl=64 time=0.615 ms
64 bytes from 10.206.1.40: icmp_seq=2 ttl=64 time=0.274 ms
64 bytes from 10.206.1.40: icmp_seq=3 ttl=64 time=0.257 ms
64 bytes from 10.206.1.40: icmp_seq=4 ttl=64 time=0.248 ms
64 bytes from 10.206.1.40: icmp_seq=5 ttl=64 time=0.290 ms
64 bytes from 10.206.1.40: icmp_seq=6 ttl=64 time=0.282 ms
^C
--- 10.206.1.40 ping statistics ---
6 packets transmitted, 6 received, 0% packet loss, time 5115ms
rtt min/avg/max/mdev = 0.248/0.327/0.615/0.129 ms
```
You can see that the latency is much lower!

If I run the same tests there I see 5ms to the ENBT/L55 and 3ms to the L81.

That was all a very long winded way of saying, "your network makes a big difference."

My ENBT modules are not loaded at all.  Nothing but my tests run against those racks.

All that said, there definitely seems to be a common thread with latency, recent library versions, and Windows.   I will do some testing to see if I can replicate this.  Unfortunately, I do not have any Windows machines, only Windows VMs.

AdonJP, 2021-04-29:
I much appreciate the detailed response Kyle. You can close this ticket.

It is indeed an ENBT module but when I was testing this there was no other traffic on either networks and the PC program was running an isolated near copy of the example. I wouldn't be surprised if the ~50ms delay was related to the OS (Windows specifically).

If you do find any information on improving this response time on Windows, I would love to hear about it. 

Cheers

kyle-github, 2021-06-16:
There is a thread on this in the forum.   I am going to leave this open as this is a real problem that needs to be fixed for Windows users.  Unfortunately, it is not clear just how to solve it yet without large changes in code.

xinthose, 2021-06-22:
I do not have speed issues on Linux using C++ and this library.

kyle-github, 2021-09-03:
I am able to replicate this on my Windows VM.  Definitely a problem here.   It only showed up after I applied about a year's worth of updates.

xinthose, 2021-09-03:
@kyle-github interesting

kyle-github, 2021-09-06:
I have been able to fix this, for some definition of fix.   Windows now has different behavior with regards to the base timer period in the OS.   Microsoft changed this somewhat recently.  

On my system, I see about 100ms to read 30 tags with the existing code.   When I set the timer period to 4ms, it takes about 25ms to read 30 tags.   

The trick here is that a lot of programs set the timer period.  That period is a global setting so any other program that does this will impact any program using libplctag.   Windows will now (later versions of Windows 10) take the smallest timer period of all processes that set the value.

I have a prerelease test out to make sure that the fix works for everyone without introducing any reversions.

Note that this is only a problem on Windows.

AdonJP, 2021-09-06:
Very excited about this! Looking forward to the update I can download.
Thanks Kyle

kyle-github, 2021-09-07:
You can pick up the core C DLLs from here: https://github.com/libplctag/libplctag/releases/tag/v2.3.8.

I have not made it a full release yet.  I want more people to test it if possible.

One of the challenges with this is that some other programs change the timer period because it is a global resource.   For instance at least until recently MS Visual Studio definitely did.  It may not now.  

I had to test by compiling everything on my Windows VM, then making sure that no other programs other than the command line were running.   Once I did that, I could see the bad latency before and the better latency after the fix.

kpreisser, 2021-09-16:
Hi,
thank you for pointing out this issue, I also was not aware that Windows would use a low resolution (15 ms on my machine) for timer APIs by default. It can actually be surprising that e.g. in .NET programs, APIs like `Thread.Sleep(int millisecondsTimeout)`, `SemaphoreSlim.Wait(int millisecondsTimeout)` etc. operate on a 15 ms raster on Windows (10 Version 2004+) unless you set a lower timer interval with `timeBeginPeriod`, which doesn't occur on other OSes like Linux.

However, my recommendation would be to not automatically set the timer period as soon as `libplctag` is used, as this affects various timer APIs of the whole process (as described above), and it could be unexpected for application developers that the timer resolution of various platform APIs depends on whether `libplctag` is used or not. Instead, maybe this could be added to Usage/How-To guides (to recommand to call `timeBeginPeriod` yourself), or maybe included in some kind of `init` function where you can pass a parameter in order to enable this functionality.
What do you think?

Thanks!

kyle-github, 2021-09-18:
@kpreisser Agreed that this is not a great solution.   However, the problem impacts performance so badly that I wanted to get something out to help users while I try to find a better solution :frowning_face:.  For that reason I made this happen by default on all Windows machines.   

It appears that using the various WaitForXXXX() functions on Windows may help but it is not clear.   MSDN is not particularly helpful and what I find on the Internet is not conclusive.   That said, I think that I will need to try to eliminate all uses of sleep() in the library in order to have a chance at working around the problem without using timeBeginPeriod().

Note that other programs impact the timer interval.   I was using an older version of MS Visual Studio and it used to set the timer interval so I never noticed the longer interval until I updated both it and Windows!

kyle-github, 2021-09-18:
Closed by release 2.3.8.

closed 2021-09-18 by kyle-github