libplctag/libplctag #261: Read the scope of the tag list, to filter the reading by tag type

Pedronet5, 2021-05-03:
Good afternoon,

I'm trying to read all the tags in my program and get the type so I can filter the reading with the correct mapping of the variable type, I'm using the TagInfoPlcMapper to get all the tags and then get the type by the hexadecimal, however when I do this I can't access my program

var tags = new Tag<TagInfoPlcMapper, TagInfo[]>()
            {
                Gateway = "10.40.60.22",
                Path = "1,0",
                PlcType = PlcType.ControlLogix,
                Protocol = Protocol.ab_eip,
                Name = "@tags" //here I can only inform @tags
            };

            tags.Initialize();
            var status = tags.GetStatus();

            if (status.ToString().ToUpper() == "OK")
            {
                tags.Read();
                var xpto = tags.Value.Select(x => x.Type);
            }

            for (int i = 0; i < tags.Value.Length; i++)
            {
                Console.WriteLine("tag: " + tags.Value[i].Name.ToString());
            }

In the rockwell manual i found this:
list controller tags. name=@tags
list each program's tags (one call per). name=Program:NiftyProg@tags
get UDT details, one at a time. name=@udt/xyz. I might need to change the separator to /.
get the special values that indicate when tags got changed. name=@wtfRockwell.

is there any way to obtain the scope of the programs, perhaps informing a specific name in the name property, example @ tags / Program: Controle_Ambientes?

kyle-github, 2021-05-04:
Hi, it looks like you are using the C# wrapper. 

The `name=@udt/1234` syntax is not yet supported in the main line of the library.  It is only available in the `event` branch right now.

If you need to get the tag types, you can only get the base types from the `@tags` pseudo tag.   You need the extra functionality in the new branch, but that is not supported in the C# wrapper yet.

I am not sure why Rockwell made it so hard to get the types of UDT tags. 

I do not understand your question about the scope of programs.   Are you trying to get the code of a program?

Pedronet5, 2021-05-04:
Hello Kyle, I'm from Brazil, I really liked your initiative to facilitate communication between PLC and C#, I was using DDE communication using RsLinx, but I had a lot of problems with communication. The application I'm working on is for an industrial system that makes a lot of requests and RsLinx does not support it has overhead and restarts itself leaving bad communication with my windows services.

The question I needed to solve would be reading the tags, to use the Wapper I need to know what type of tag I have, I have a table with the registration of all tags, but some types are wrong and when the application reads this tag the returned value is wrong Example: 0.23564 the value is in float but my type is DINT the returned value is 1512085188 and this locks the IHM due to the wrong sending to the PLC.

I saw it is not possible to carry out the reading as you said, do you know another way to get the tags with the types? it could be in excel or something like that, i would put together a script and normalize my entries in the database.

Thank you very much for creating the library, it made reading and writing on the PLC a lot easier, very fast and without crashes, congratulations to everyone involved in the project.

GitHubDragonFly, 2021-05-05:
You could check the code of the C# wrapper where tags are being extracted and if there is a part where the tag type is being checked then you can potentially use it to create a text file of tags and their types (plain txt or csv for use in Excel).

Also, take a look at this project as well, specifically "Get Tags" region, since it extracts the tag type and it also has a sort of identity table on top of the file:

https://github.com/GitHubDragonFly/AB-Modbus-Master/blob/master/AB-Modbus-Master/Form1.vb

It is coded in VB Net which is not hard to convert to C#.

If you are handy with the code then you can also try using [pycomm3](https://github.com/ottowayi/pycomm3) python library to get tags (which would include UDT members as well). The project has a GUI example which does the tag listing. This library can be used either in a python project alone or check this video that shows how to integrate python with C#:

https://www.youtube.com/watch?v=BQX7BFsogjs

Find time to read everything that's available for all of the above mentioned.

kyle-github, 2021-09-24:
Full UDT definition listing is now supported in both the core library and the .Net library.   Please reach out to the .Net library authors at the [libplctag.NET issues](https://github.com/libplctag/libplctag.NET/issues) page.

closed 2021-09-24 by kyle-github