libplctag/libplctag #262: Intermittent Timeout Exception when calling Tag.Initialize()

WetTheSystem, 2021-05-13:
libtagplc v1.0.3

libplctag.LibPlcTagException
  HResult=0x80131500
  Message=ErrorTimeout
  Source=libplctag
  StackTrace:
   at libplctag.Tag.Initialize()
   at PLCTagInterfaceHandler.Service1.CheckUnackedAlarms(Object source, ElapsedEventArgs e) in C:\Users\tjohnson8\source\repos\PLCTagInterfaceHandler\Service1.cs:line 284
   at System.Timers.Timer.MyTimerCallback(Object state)

Some context: This code is installed on 2 servers and running concurrently. I occasionally get a timeout that throws an unhandled exception, but only ever on one server. 

Here's the snippet that creates the exception:

`var unackedArray = TagGenerator.TagGenArray("AlarmCheck", 44, "bool", ###.###.###.###");
unackedArray.Initialize();`

Here's a code snippet from class TagGenerator that creates this bool array:

`var boolTag = new Tag<BoolPlcMapper, bool[]>()
                    {
                        Name = tag_name,
                        Gateway = gateway,
                        Path = path,
                        PlcType = PlcType.ControlLogix,
                        Protocol = Protocol.ab_eip,
                        ArrayDimensions = new int[] { array_length, array_length },
                        Timeout = TimeSpan.FromMilliseconds(1000),
                    };

                    return boolTag;`

Can you give some suggestions on where this timeout originates from so I can troubleshoot it, or if you have seen it in the past provide some insight? Is it because I'm grabbing such a large array of bools (a length value of 44 results in a bool array of size 1952 which may be some issue itself. I derived this value by just checking my result to get an array size as close as possible to 2000)

kyle-github, 2021-05-15:
Hi, this is really about the [libplctag.NET](https://github.com/libplctag/libplctag.NET) project.   Can you please raise this issue [there](https://github.com/libplctag/libplctag.NET/issues)?  There are many strange edge cases with boolean arrays!

kyle-github, 2021-05-26:
As noted above, please raise the issue on the [libplctag.NET](https://github.com/libplctag/libplctag.NET) project.  Thanks1

closed 2021-05-26 by kyle-github