libplctag/libplctag #263: Can Mitsubishi PLC module be connected?

byeondwl, 2021-05-19:
Recently I tried to use libplgtag to connect to Mitsubishi’s QJ71EIP71 communication module. How should I write TAG_PATH? Please provide me with any good suggestions, thank you ！

kyle-github, 2021-06-03:
This library only supports CIP (Rockwell, Omron) and Modbus.   What protocol does the QJ71EIP71 support?

byeondwl, 2021-06-07:
> 
> 
> This library only supports CIP (Rockwell, Omron) and Modbus. What protocol does the QJ71EIP71 support?

The QJ71EIP71 module allows MELSEC-Q series modules to connect to the EtherNet/IP network;
Connectivity to the EtherNet/IP network:

1.  Communications using tags (Tag communication);
2.  Cyclic Transmission (Class1 Tag communication);
3. Communications using read/write requests (Class3 Tag communication);
4.  Asynchronous communication without establishing a connection (UCMM Tag communication).