libplctag/libplctag #269: Status of AB Implicit IO messages

Justin-Davis, 2021-06-24:
Did this feature ever get picked up? I see it referenced in documentation here or there, but can't find any implementation details.

kyle-github, 2021-07-03:
Hi, no it has not been picked up.   There is little documentation I have found on it.  It may be possible to reverse engineer.  One large challenge is the current code base.   It is very much written with the existing flow in mind.

Justin-Davis, 2021-07-06:
@kyle-github understood! thank you for the reply. It would be a nice feature to have, but with a lack of documentation I can see that being very difficult to implement.

kyle-github, 2021-07-16:
I am pretty much stalled on the library for a little while.  I joined Amazon AWS and there is some overlap with some internal projects there.  I am working with Amazon's legal team to make sure I have all the exceptions and permissions so that there is absolutely no question about IP (intellectual property) and time.  I've been on vacation for a bit too so that slowed things even more.

That all said, getting implicit IO supported would be a really nice feature and has been requested for a long time.

kchiloff, 2023-02-25:
Hi Kyle, 

 I have recently created a sparkplug bridge using libPLCtag, so got seriously interested in this "feature".  Can you comment on what's needed to get this implemented?!  Can we maybe help somehow? 

Regards,
Kamen

kchiloff, 2024-03-08:
Hi @kyle-github, 

How is your AWS adventure going? You might have missed my question above, but we really need help on this one!  Can we get in touch to discuss in the coming weeks?  You can count on my help, where needed.

Regards,
Kamen

kyle-github, 2024-03-08:
Hi, there hasn't been any movement on that feature yet.

I left AWS a little while ago for health reasons (among others) so I don't
have any direct contact with any of the AWS teams any more.

Implicit (type 1 connections) are quite a bit different than type 3
(explicit) connections.   They are not documented in any public docs I've
found on Rockwell's site.  There are quite a few unknowns.  The overall
flow is something like this:

   1. PC/PLC opens an EIP connection (seems to be TCP??, if so when does it
   get dropped?) to the device it wants to read.
   2. The PC/PLC negotiates several things with an expanded ForwardOpen
   message (not documented that I've found).  This includes:
      1. The RPI of data from the device.
      2. The RPI of ACKs from the PC/PLC to the device.
      3. The type of source.  As far as I can tell a produced tag looks a
      little different from an IO device.
      4. Whether the connection is multicast or unicast from the device.
      If multicast, then the multicast IP address is handed from the device to
      the PC/PLC.
      5. Whether the ACKs are sent via TCP or unicast UDP (it is unclear
      that this is directly negotiated)
      6. The size of the data both directions (I think, it is also not
      clear), the limit is about 500 bytes
      7. The number of allowed dropped packets (in both directions, I think)
      8. Probably more I do not know about.
   3. The PC/PLC then connects to the multicast address (if there is one).
   4. The PC/PLC listens for multicast packets containing a stream of data.
   5. The PC/PLC sends periodic ACKs to keep the connection alive.

Most of implicit IO is based on UDP rather than TCP.   The library
currently is strictly TCP.  Not a big deal but a fair amount more code and
testing.

The biggest challenge is the internal control flow.  The entire library is
architected around a specific model of interaction that is different for
implicit IO.  With explicit IO, there is an action that triggers a request
to the PLC so the library knows when to wait for a response.  With implicit
IO that is not true.  The response just stream in from time to time.

I did a quick analysis of the scope of the changes a couple of years ago
and realized that I would have to touch about 90% of the CIP code and at
least 25% of the generic code to make this work.   Or I would have to write
a completely separate protocol stack for implicit IO.   I am strongly
leaning toward the latter as it does not have the same probability of
breaking existing code.

Note that having UDP support is also useful for doing device/PLC
discovery.

At this point my priority to get back to maintaining the library (I have
not done a release in a long time!) and investigate ways to simplify it.
I seriously considered learning Rust and writing the library in that, but
the LLVM compiler infrastructure that Rust relies on does not support
anywhere near the number of platforms/CPUs that GCC supports, so the
semi-embedded users of the library might not be able to use the code.

I hope this explains the situation.  If you have additional information
about how implicit IO works, I would love to hear about it.

Best,
Kyle

kchiloff, 2024-03-19:
Hi Kyle,

Thanks for your swift and elaborate reply!!  And excuses for the late reply - found your message in spam.
I think the biggest need for this kind of functionality -at least the one of my project-, is to get rid of the polling WoW and move to event based way of communication.  
Having said that I’m not sure if this changes anything in your story below, or makes it a bit more simple since it does not involve IO.  That because  I envision to use to MSG instructions to push messages, and I do not know if MSG instruction communication is UDP or TCP based. If latter is the case, I hope this can simplify things in terms of implementation/integration in current library.  Any thoughts on that one?  

Looking forward to your reply!!

Regards,
Kamen 
P.S. Otherwise I totally understand your point in building something next to the current implementation instead of massively changing the current code base.