libplctag/libplctag #271: Windows lock/unlock issue possible.

OlegUA, 2021-07-05:
Hi All
I'm a long time using this library and no issues were here before, thank you for the nice job. But now I've faced up with an issue when plc_tag_lock is hanging up due to race conditions when I'm trying to read/write the same tag from different threads. Unfortunately, I can't direct check it by supplied examples due to ones are UNIX related and can't be directly compiled in Windows. I'm using it from c# via standard DLL wrapper. 
So, the question - is it an issue? Is those functions pair could be used in Windows or not? Or do I have to just use synchronizing on the upper (c#) level and don't touch those functions?

timyhac, 2021-07-19:
I think the recommendation is to use synchronization primitives (i.e. `lock`) available in c#.

> ... if you are using a wrapper like Java, you should use the native synchronization facilities of that language!
https://github.com/libplctag/libplctag/wiki/API#controlling-multi-threaded-access-to-a-tag

Whether there is an issue I'm not able to say - are you able to set up a minimal example of an issue occurring in the library?

OlegUA, 2021-07-21:
> I think the recommendation is to use synchronization primitives (i.e. `lock`) available in c#.
> 
Yes, I already did it. 

> Whether there is an issue I'm not able to say - are you able to set up a minimal example of an issue occurring in the library?
I don't know how to correctly insert the piece of code here. I've tried everything but it was inserted in such a way, sorry.

`
public bool lockTagDLL(int timeout = 0) {
            int rc = 1;
            if(timeout > 0) {
                Stopwatch stopWatch = new Stopwatch();
                stopWatch.Start();
                do {
                    rc = plctag.plc_tag_lock(tagHandle);
                    if(stopWatch.ElapsedMilliseconds > timeout)
                        throw new LibPlcTagException($"Unable to lock {Name}, timeout {timeout} exceeded");
                    Thread.Sleep(50);
                } while(rc > 0);
                stopWatch.Stop();
            } else {
                rc = plctag.plc_tag_lock(tagHandle);
            }
            if(rc < 0)
                throw new LibPlcTagException($"Unable to lock {Name}, status {rc} ({plctag.plc_tag_decode_error(rc)})");
            return rc == (int)STATUS_CODES.PLCTAG_STATUS_OK;
        }

        public bool unlockTagDLL() {
            return plctag.plc_tag_unlock(tagHandle) == 0;
        }

`
All calls to DLL are direct-mapped via DLL Import. And I'm using the 64-bit target.

timyhac, 2021-07-21:
Sorry, it's still unclear: what is not working?

OlegUA, 2021-07-23:
The plctag.plc_tag_lock(tagHandle) is hanging up the current thread forever when I'm trying to use those functions. I've found a workaround, but maybe this information is useful for the improvement of this great library.

kyle-github, 2021-07-23:
The code above does not look right to me, but I have not used the C# wrapper.  The core library provides the lock/unlock functions, but those do not have timeouts.   If you call one, your thread hangs until the mutex can be taken.  Once the mutex is taken by one thread, all other threads will hang if they try to lock the mutex.

Java has a hidden mutex associated with every object and that is what I use there.   I assume that C# has a solid set of synchronization primitives and you should use those.

I will make the examples work on Windows as soon as possible to make sure that there is no hidden flaw in the library.

timyhac, 2021-07-23:
I think that the equivalent of Java's `synchronized` is [`lock`](https://docs.microsoft.com/en-us/dotnet/csharp/language-reference/keywords/lock-statement) - both of which are keywords in their language.

I don't understand why you need to use `plc_tag_lock` instead of c#'s lock keyword. Maybe [Monitor](https://docs.microsoft.com/en-us/dotnet/api/system.threading.monitor?view=net-5.0) is what you're looking for in that it returns instantly irrespective of whether the mutex lock was acquired or not, and you can continue in both cases.

I would never do this in my application, but to prove that the `plc_tag_lock` method is working, see the following example:

### Example code
```csharp
public static void LockExample()
{
    myTagHandle = plctag.plc_tag_create("protocol=ab_eip&gateway=192.168.0.10&path=1,0&plc=LGX&elem_size=4&elem_count=1&name=zFloat[4]", 2000);
    var status = (STATUS_CODES)plctag.plc_tag_status(myTagHandle);
    Console.WriteLine($"{GetCurrentTime()} Tag created {status}");

    var thread1 = new Thread(new ThreadStart(TagLockThread));
    var thread2 = new Thread(new ThreadStart(TagLockThread));

    thread1.Start();
    thread2.Start();

    thread1.Join();
    thread2.Join();

    Console.WriteLine($"{GetCurrentTime()} Finished");
}

static int myTagHandle;
static void TagLockThread()
{
    var threadId = Thread.CurrentThread.ManagedThreadId;
    Console.WriteLine($"{GetCurrentTime()} Attempting to lock in ThreadId={threadId}");
    var status = (STATUS_CODES)plctag.plc_tag_lock(myTagHandle);
    Console.WriteLine($"{GetCurrentTime()} Lock acquired {status} ThreadId={threadId}");

    // While this thread is active, the other thread can't aquire the lock
    Thread.Sleep(10000);
}

static string GetCurrentTime()
{
    return DateTime.Now.ToString("hh:mm:ss.fff");
}
```

### Output
Note that the second thread had to wait for the other thread to finish before `plc_tag_lock(..)` could return.
```
08:59:51.150 Tag created PLCTAG_STATUS_OK
08:59:51.342 Attempting to lock in ThreadId=8
08:59:51.342 Attempting to lock in ThreadId=7
08:59:51.345 Lock acquired PLCTAG_STATUS_OK ThreadId=8
09:00:01.363 Lock acquired PLCTAG_STATUS_OK ThreadId=7
09:00:11.377 Finished
```

kyle-github, 2021-07-23:
Thanks, @timyhac.   Where are you unlocking?  The pattern that is in the C examples is:

```c
int thread_function(void)
{
    int rc = PLCTAG_STATUS_OK;
    /* ... */

    rc = plc_tag_lock(tag_id);
    /* check return code */

    /* do work with the tag. */

    rc = plc_tag_unlock(tag_id);
```

timyhac, 2021-07-24:
I guess when the thread returns, it releases the mutex.

Inserting a call to `plc_tag_unlock` reveals an issue - maybe this is what @OlegUA was seeing - it hangs on this call.

### Code
```csharp
static void TagLockThread()
{
    var threadId = Thread.CurrentThread.ManagedThreadId;

    Console.WriteLine($"{GetCurrentTime()} Attempting to acquire lock in ThreadId={threadId}");
    var status = (STATUS_CODES)plctag.plc_tag_lock(myTagHandle);
    Console.WriteLine($"{GetCurrentTime()} Lock aquired {status} ThreadId={threadId}");

    // While this thread is active, the other thread can't aquire the lock
    Thread.Sleep(10000);

    Console.WriteLine($"{GetCurrentTime()} Attempting to release lock in ThreadId={threadId}");
    var unlockStatus = (STATUS_CODES)plctag.plc_tag_unlock(myTagHandle);
    Console.WriteLine($"{GetCurrentTime()} Lock released {unlockStatus} ThreadId={threadId}");
}
```

### Output
```
09:49:36.508 Tag created PLCTAG_STATUS_OK
09:49:36.546 Attempting to acquire lock in ThreadId=5
09:49:36.546 Attempting to acquire lock in ThreadId=6
09:49:36.546 Lock aquired PLCTAG_STATUS_OK ThreadId=6
09:49:46.551 Attempting to release lock in ThreadId=6
```

If you only run one thread, it works:

### Code
```csharp
public static void LockExample()
{
    myTagHandle = plctag.plc_tag_create("protocol=ab_eip&gateway=192.168.0.10&path=1,0&plc=LGX&elem_size=4&elem_count=1&name=zFloat[4]", 2000);
    var status = (STATUS_CODES)plctag.plc_tag_status(myTagHandle);
    Console.WriteLine($"{GetCurrentTime()} Tag created {status}");

    var thread1 = new Thread(new ThreadStart(TagLockThread));
    //var thread2 = new Thread(new ThreadStart(TagLockThread));

    thread1.Start();
    //thread2.Start();

    thread1.Join();
    //thread2.Join();

    Console.WriteLine($"{GetCurrentTime()} Finished");
}
```

### Output
```
10:17:37.832 Tag created PLCTAG_STATUS_OK
10:17:37.858 Attempting to acquire lock in ThreadId=5
10:17:37.858 Lock aquired PLCTAG_STATUS_OK ThreadId=5
10:17:47.872 Attempting to release lock in ThreadId=5
10:17:47.872 Lock released PLCTAG_STATUS_OK ThreadId=5
10:17:47.878 Finished
```

kyle-github, 2021-07-24:
Thanks for the output log.  That's odd.   Releasing a lock should be fine.   This helps.  I will debug this weekend.

timyhac, 2021-07-24:
Hi @kyle-github, here is a trace log for the code example that uses `plc_tag_unlock`: [trace.log](https://github.com/libplctag/libplctag/files/6871836/trace.log).

Once it gets to about line 1400, there appears to be a kind of loop. I cut the session short after a few seconds - it would keep logging the same messages repeatedly.

OlegUA, 2021-07-24:
> I guess when the thread returns, it releases the mutex.
> 
> Inserting a call to `plc_tag_unlock` reveals an issue - maybe this is what @OlegUA was seeing - it hangs on this call.

But I have a case when one of the threads never returns (only when the service is stopped). 

In C# I'm using semaphores to implement inter-task synchronization:
`        public bool lockTag(int timeout) {
            return semaphore.Wait(timeout);
        }

        public void unlockTag() {
            semaphore.Release();
        }
`

kyle-github, 2021-08-30:
This should be fixed now with the latest 2.3.7 release.   Can you please try with the latest C DLLs?

timyhac, 2021-08-30:
@kyle-github - yes this is fixed. libplctag.NativeImport wrapper has been updated with v.2.3.7 dlls and fixes this issue.

I've uploaded the port of multithread.c to [here](https://github.com/libplctag/libplctag.NET/blob/master/src/Examples/CSharp%20NativeImport/Multithread.cs).

I think this issue can be closed.

Here is the output:
```
Creating 10 threads.
Locking in 6
Locking in 7
Locking in 8
Locking in 1
Locking in 3
Locking in 4
Locking in 5
Locking in 0
Locking in 2
Locking in 9
Reading in 6
Unlocking in 6
Thread 6 got result 1 with return code PLCTAG_STATUS_OK in 00:00:00.1484066
Reading in 0
Locking in 6
Unlocking in 0
Thread 0 got result 1 with return code PLCTAG_STATUS_OK in 00:00:00.2911839
Reading in 2
Locking in 0
Unlocking in 2
Thread 2 got result 1 with return code PLCTAG_STATUS_OK in 00:00:00.4172447
Reading in 3
Locking in 2
Unlocking in 3
Thread 3 got result 1 with return code PLCTAG_STATUS_OK in 00:00:00.5603842
Reading in 6
Locking in 3
Unlocking in 6
Thread 6 got result 1 with return code PLCTAG_STATUS_OK in 00:00:00.5700817
Reading in 3
Locking in 6
Unlocking in 3
Thread 3 got result 1 with return code PLCTAG_STATUS_OK in 00:00:00.3159987
Reading in 6
Locking in 3
Unlocking in 6
Thread 6 got result 1 with return code PLCTAG_STATUS_OK in 00:00:00.2841729
Locking in 6
Reading in 2
Unlocking in 2
Thread 2 got result 1 with return code PLCTAG_STATUS_OK in 00:00:00.7442418
Reading in 6
Locking in 2
Unlocking in 6
Thread 6 got result 1 with return code PLCTAG_STATUS_OK in 00:00:00.2681773
Reading in 3
Locking in 6
Unlocking in 3
Thread 3 got result 1 with return code PLCTAG_STATUS_OK in 00:00:00.5339771
Reading in 1
Locking in 3
Unlocking in 1
Thread 1 got result 1 with return code PLCTAG_STATUS_OK in 00:00:01.5853938
Reading in 5
Locking in 1
Unlocking in 5
Thread 5 got result 1 with return code PLCTAG_STATUS_OK in 00:00:01.7261869
Reading in 3
Locking in 5
Unlocking in 3
Thread 3 got result 1 with return code PLCTAG_STATUS_OK in 00:00:00.4091355
Reading in 9
Locking in 3
Unlocking in 9
Thread 9 got result 1 with return code PLCTAG_STATUS_OK in 00:00:01.9957631
Reading in 5
Locking in 9
Unlocking in 5
Thread 5 got result 1 with return code PLCTAG_STATUS_OK in 00:00:00.3868525
Reading in 7
Locking in 5
Unlocking in 7
Thread 7 got result 1 with return code PLCTAG_STATUS_OK in 00:00:02.2537926
Reading in 0
Locking in 7
Unlocking in 0
Thread 0 got result 1 with return code PLCTAG_STATUS_OK in 00:00:02.0712580
Reading in 6
Locking in 0
Unlocking in 6
Thread 6 got result 1 with return code PLCTAG_STATUS_OK in 00:00:01.1713615
Reading in 2
Locking in 6
Unlocking in 2
Thread 2 got result 1 with return code PLCTAG_STATUS_OK in 00:00:01.4204024
Reading in 5
Locking in 2
Unlocking in 5
Thread 5 got result 1 with return code PLCTAG_STATUS_OK in 00:00:00.6090301
Locking in 5
Reading in 7
Unlocking in 7
Thread 7 got result 1 with return code PLCTAG_STATUS_OK in 00:00:00.5924138
Reading in 2
Locking in 7
Unlocking in 2
Thread 2 got result 1 with return code PLCTAG_STATUS_OK in 00:00:00.3578966
Reading in 6
Locking in 2
```

kyle-github, 2021-09-02:
Huh.  At least in the C code, you see "thread X got result Y with return code..." where Y is an incrementing value.

The test is to read and increment and write back a single tag in a PLC across multiple threads.  By use of the plc_tag_lock and plc_tag_unlock functions you guard the active portion such that only one thread can read, increment, then write the tag value at a time.

In the log above, it looks like you are getting the same value over and over.

timyhac, 2021-09-02:
@kyle-github - ah yes, the attribute string was referencing an INT (16bit) - here is the output with a DINT tag.

```
Creating 10 threads.
Locking in 8
Locking in 0
Locking in 9
Locking in 5
Locking in 6
Locking in 1
Locking in 3
Locking in 4
Locking in 2
Locking in 7
Reading in 8
Unlocking in 8
Thread 8 got result 0 with return code PLCTAG_STATUS_OK in 00:00:06.7156197
Reading in 0
Locking in 8
Unlocking in 0
Thread 0 got result 1 with return code PLCTAG_STATUS_OK in 00:00:06.8566091
Reading in 2
Locking in 0
Unlocking in 2
Thread 2 got result 2 with return code PLCTAG_STATUS_OK in 00:00:06.9803056
Reading in 0
Locking in 2
Unlocking in 0
Thread 0 got result 3 with return code PLCTAG_STATUS_OK in 00:00:00.2524977
Reading in 3
Locking in 0
Unlocking in 3
Thread 3 got result 4 with return code PLCTAG_STATUS_OK in 00:00:07.2367662
Reading in 5
Locking in 3
Unlocking in 5
Thread 5 got result 5 with return code PLCTAG_STATUS_OK in 00:00:07.3457213
Locking in 5
Reading in 1
Unlocking in 1
Thread 1 got result 6 with return code PLCTAG_STATUS_OK in 00:00:07.4723271
Reading in 5
Locking in 1
Unlocking in 5
Thread 5 got result 7 with return code PLCTAG_STATUS_OK in 00:00:00.2333642
Reading in 4
Locking in 5
Unlocking in 4
Thread 4 got result 8 with return code PLCTAG_STATUS_OK in 00:00:07.7193005
Reading in 3
Locking in 4
Unlocking in 3
Thread 3 got result 9 with return code PLCTAG_STATUS_OK in 00:00:00.6057719
Reading in 5
Locking in 3
Unlocking in 5
Thread 5 got result 10 with return code PLCTAG_STATUS_OK in 00:00:00.3751647
Reading in 3
Locking in 5
Unlocking in 3
Thread 3 got result 11 with return code PLCTAG_STATUS_OK in 00:00:00.2520425
Reading in 0
Locking in 3
Unlocking in 0
Thread 0 got result 12 with return code PLCTAG_STATUS_OK in 00:00:01.1283896
Reading in 7
Locking in 0
Unlocking in 7
Thread 7 got result 13 with return code PLCTAG_STATUS_OK in 00:00:08.3909545
Reading in 8
Locking in 7
Unlocking in 8
Thread 8 got result 14 with return code PLCTAG_STATUS_OK in 00:00:01.7708407
Reading in 1
Locking in 8
Unlocking in 1
Thread 1 got result 15 with return code PLCTAG_STATUS_OK in 00:00:01.1228925
Reading in 8
Locking in 1
Unlocking in 8
Thread 8 got result 16 with return code PLCTAG_STATUS_OK in 00:00:00.2340997
Reading in 0
Locking in 8
Unlocking in 0
Thread 0 got result 17 with return code PLCTAG_STATUS_OK in 00:00:00.5930074
Reading in 9
Locking in 0
Unlocking in 9
Thread 9 got result 18 with return code PLCTAG_STATUS_OK in 00:00:09.0088430
Reading in 4
Locking in 9
Unlocking in 4
Thread 4 got result 19 with return code PLCTAG_STATUS_OK in 00:00:01.4102996
Reading in 0
Locking in 4
Unlocking in 0
Thread 0 got result 20 with return code PLCTAG_STATUS_OK in 00:00:00.3788035
Reading in 3
Locking in 0
Unlocking in 3
Thread 3 got result 21 with return code PLCTAG_STATUS_OK in 00:00:01.2563413
Reading in 2
Locking in 3
Unlocking in 2
Thread 2 got result 22 with return code PLCTAG_STATUS_OK in 00:00:02.5387317
Reading in 0
Locking in 2
Unlocking in 0
Thread 0 got result 23 with return code PLCTAG_STATUS_OK in 00:00:00.3780125
Reading in 1
Locking in 0
Unlocking in 1
Thread 1 got result 24 with return code PLCTAG_STATUS_OK in 00:00:01.1512666
Reading in 5
Locking in 1
Unlocking in 5
Thread 5 got result 25 with return code PLCTAG_STATUS_OK in 00:00:01.9135641
Reading in 0
Locking in 5
Unlocking in 0
Thread 0 got result 26 with return code PLCTAG_STATUS_OK in 00:00:00.3757222
Reading in 9
Locking in 0
Unlocking in 9
Thread 9 got result 27 with return code PLCTAG_STATUS_OK in 00:00:01.1634947
Reading in 2
Locking in 9
Unlocking in 2
Thread 2 got result 28 with return code PLCTAG_STATUS_OK in 00:00:00.7547573
Reading in 4
Locking in 2
Unlocking in 4
Thread 4 got result 29 with return code PLCTAG_STATUS_OK in 00:00:01.2915946
Reading in 0
Locking in 4
Unlocking in 0
Thread 0 got result 30 with return code PLCTAG_STATUS_OK in 00:00:00.5231422
Reading in 4
Locking in 0
Unlocking in 4
Thread 4 got result 31 with return code PLCTAG_STATUS_OK in 00:00:00.2702642
Reading in 0
Locking in 4
Unlocking in 0
Thread 0 got result 32 with return code PLCTAG_STATUS_OK in 00:00:00.2690463
Reading in 5
Locking in 0
Unlocking in 5
Thread 5 got result 33 with return code PLCTAG_STATUS_OK in 00:00:01.0753208
Locking in 5
Reading in 3
Unlocking in 3
Thread 3 got result 34 with return code PLCTAG_STATUS_OK in 00:00:01.7310009
Reading in 0
Locking in 3
Unlocking in 0
Thread 0 got result 35 with return code PLCTAG_STATUS_OK in 00:00:00.3766020
Reading in 9
Locking in 0
Unlocking in 9
```

timyhac, 2021-09-02:
Yes @kyle-github  - I get around 50-60ms per network operation mostly - I can help with testing if needed.

I guess we can close this ticket now.

closed 2021-09-03 by kyle-github