libplctag/libplctag #272: Node.js wrapper

Ernestynian, 2021-07-08:
My current node.js project requires communication over ethernet/ip with Allen Bradley plcs.

AFAIK existing ethernet/ip libraries for node.js only support atomic types and don't provide bundling multiple requests into one packet as libplctag does in async mode. Low latency communication is important for my application and it's impossible to make it work with multiple (on average) milliseconds per tag read/write operation.

My question(s):

1. Did anyone before me attempt writing a node.js wrapper for this library?
2. If you are knowledgeable about writing native node.js addons, do you have some tips for me or warnings of possible roadblocks?

I did consider modifying existing libraries, but I'm not proficient enough in coding network communication and I feel like it would require major refactoring anyway.

I've also thought of wrapping around one of the existing wrappers for libplctag, but that would potentially involve including another big dependency, which I would like to avoid.

xinthose, 2021-07-08:
I have not done this, but did look into it.  My suggestion is node-ethernet-ip: https://github.com/cmseaton42/node-ethernet-ip/issues/87

Ernestynian, 2021-07-08:
@xinthose thanks for the suggestion. I did test it out not long ago and the performance wasn't satisfactory, although as I'm giving it a second glance now, seems like I've missed a pretty important feature, namely tag groups. This could potentially work in a similar manner as libplctag does in async mode, though the implementation seems pretty basic as it is missing packet size negotiation, and grouping has to be done manually.

PR code with UDT support does solve another one of my issues with this library.

I will test it again tomorrow, but in the meanwhile please consider this thread open to further discussion.

kyle-github, 2021-08-25:
Any luck using node-ethernet-ip?

Ernestynian, 2021-08-25:
My coworker tested it and found some critical issues with tag groups. As it was some time ago, I don't remember what it was exactly.

That led me to writing a wrapper of this lib for node.js, but it's not feature complete and the parent project has gotten paused before it was tested in a real-world application.

As this thread didn't get any responses in over a month, I am going to close it. Feel free to open it again.  If you are reading this, thank you for your time.

closed 2021-08-25 by Ernestynian