libplctag/libplctag #273: Access to the path *\bin\x86\Debug\AppX\plctag.dll' is denied.

JakePorter05, 2021-07-08:
I am currently trying to get this system to work with a Xamarin.Forms app. It compiles and everything is good, but when I run the app and hit ReadTag.Read(); after creating the tag it comes up with: Access to the path *\bin\x86\Debug\AppX\plctag.dll' is denied.
I looked at the path and the plctag.dll was missing I copied the dll in manually but no change. 
Is this possible or should I just go back to a WPF application to get this to run? (I currently do have a WPF app working)
This part of the App will only be used in UWP so it doesn't have to run in IOS and Droid which I saw wasn't compatible yet. 
The app was made with the latest templates in the latest version of VS 2019 for windows.

JakePorter05, 2021-07-08:
Also getting android working would be amazing.

timyhac, 2021-07-09:
Hi @JakePorter05 - you've hit https://github.com/libplctag/libplctag.NET/issues/137

Please read the README info about "ForceExtractLibrary" [here](https://github.com/libplctag/libplctag.NET#libplctagnativeimport).

jakeporter21, 2021-07-16:
I went in and read the previous post and also looked at the readme. I have turned off the force extract library with plctag.ForceExtractLibrary = false;. Now I get: An attempt was made to load a program with an incorrect format. (Exception from HRESULT: 0x8007000B). What is not clear in what I have read is how do I compile the native lib for UWP I pulled the GitHub and have the code. It compiles just fine, but in the plctag\out\build\x64-Debug\bin_dist folder I find the plctag.dll, but I don't see libplctag.dll or the libplctag.nativeimport.dll. Any more direction would be great thank you.

timyhac, 2021-07-16:
The issue is probably a 32bit vs 64bit mismatch. Make sure the binary you're providing is compiled for the same mode you're running the app.

jakeporter21, 2021-07-16:
Sweetness, it totally worked. I will load a .md in here as a comment with the steps I took to fix it for future noobs.

jakeporter21, 2021-07-20:
## Getting UWP working. 

I set up a basic Xamarin app with a single page. The page was the current template(2021) with a button to click on and an entry to enter an IP address. The Code behind was just the basic use case of this library.

```
// Instantiate the tag with the appropriate mapper and datatype
var myTag = new Tag<DintPlcMapper, int>()
{
    Name = "PROGRAM:SomeProgram.SomeDINT",
    Gateway = "10.10.10.10",
    Path = "1,0",
    PlcType = PlcType.ControlLogix,
    Protocol = Protocol.ab_eip,
    Timeout = TimeSpan.FromSeconds(5)
};

// Read the value from the PLC
myTag.Read();

// Output to Console
Console.WriteLine(myTag.Value);
```

Simple. It didn't work. 

I got access denied. Which also when looking at the bin folder means it just wasn't there. 

Required: You need the plctag.dll, libplctag.dll, and libplctag.NativeImport.dll.
Which the compiler gives you all but the plctag.dll. So here are the steps to get it working.

* Add this line to the constructor of the class that calls the dll
    *  plctag.ForceExtractLibrary = false;
        * Which is in the documentation in the readme
* Then get the native compiled libs
    * Download this project from GitHub
    * Open the project it will auto build
    * Modify the CMake settings to what you need   <- this is what I didn't do
        * Just close and open the project and it should have a CMake overview page. Which has an "Open The CMake Settings Editor" option under Configure CMake
        * This has all of the compile options debug/release and 86/64/86-64
        * Make your changes hit save
    * Highlite solution
    * Build all under Build menu
* Look for the plctag.dll
    * plctag\out\build\x86-Debug\bin_dist 
    * Should be somewhere here ^ depending on what name you have the CMake 
* Copy into your bin folder or your solution and run. 

## Voila Your Done.

I hope this helps any other noobs out there.

timyhac, 2021-07-20:
Thanks Jake, you can also download plctag.dll from the releases page:
https://github.com/libplctag/libplctag/releases

kyle-github, 2021-07-21:
Er, why do we need all those DLLs?  I am by no means a C#/.Net programmer but in Java, there's just one DLL per platform when doing something similar.

@timyhac could this set of steps go into the notes/wiki for the .Net project?

timyhac, 2021-07-22:
Hi guys - I've added some notes to the README here: https://github.com/libplctag/libplctag.NET#developing-for-systems-with-immutable-application-directories

kyle-github, 2021-07-22:
Thanks, @timyhac!

timyhac, 2021-07-25:
İ guess this ticket can be closed, right @JakePorter05 ?

jakeporter21, 2021-07-26:
Yes, it can be closed.

closed 2021-07-27 by kyle-github