libplctag/libplctag #274: Omron-njnx Error when I try to write sint or usint.

joao147, 2021-07-14:
I m using libplctag.NET, I opened this issue there, but bug was not there. A colab said that error can be on core project or with my plc.
This is thee error that is returned using libplctag.NET api.
[image]

Same error using native wrapper.
[image]

kyle-github, 2021-07-16:
Hi thanks for bringing this up.  Can you run the test with debugging turned in to level 4?  Getting that output will help understand what is happening here.

timyhac, 2021-07-29:
@joao147 add &debug=4 into the attribute string to do this.

kyle-github, 2021-08-25:
Hi, have you had any time to try to get a debugging dump?

kyle-github, 2021-09-24:
Please reopen this when you are able to get the debug logs.

closed 2021-09-24 by kyle-github