libplctag/libplctag #275: Request to provide a PLCTAG_EVENT_CREATED event

timyhac, 2021-07-19:
I'm looking at an event-driven asynchronous API for the .NET wrapper. The current implementation just polls the tag status every 10ms until the status is not Pending.
Read and write operations provide an event when they are finished (e.g. `PLCTAG_EVENT_READ_COMPLETED`), which can instead be used to resolve the asynchronous call.

Tag creation is also optionally asynchronous, but there is no event that notifies a consumer when the operation is complete. The only way to know is to continually poll the tag status as it does currently.

This is a request is to provide a callback for event tag creation.

kyle-github, 2021-07-20:
Let me look into this.   Getting an actual "tag is ready for use" is surprisingly difficult due to the back end being entirely async.  And handling the state sharing between the tag and the "plc" object is a little tricky.   

I do think that this is doable.  It will require a minor version bump because it impacts a public API.

I am still currently blocked by legal negotiations with my employer.   Everyone is on vacation which is delaying things :-/

timyhac, 2021-07-20:
All good, no rush from my end. İ created this ticket with the full expectation that this was the case.

kyle-github, 2021-07-21:
There might be a workaround here.  The back end triggers a read before handing back the tag handle.  This is to make sure that if the user does a write immediately after the tag is created the values in the tag are from the PLC and not all zeros.  I may be able to note when that read completes and use that to synthesize a "tag ready" event.

timyhac, 2021-08-14:
I've just deployed a new alpha version of libplctag.NET that uses that initial Read as the CREATED event. Seems to work OK. This means that the Async methods all use true async now (event-driven) as opposed to polling.

kyle-github, 2021-08-25:
Hi Tim, thanks for being so patient.   I really appreciate it.   I think having a CREATED event is a good idea here.  I'll take a peek at the code over the next couple of days to see if there is a good place to set this up.   The library is in layers and the top layer is the one that generates the events and nothing really bubbles up that says "creation is done" because so much in the lower level is async.  I can probably figure something out to make this work.

Also, if you could drop me an email about what would make it easier to support true async in C#, I would appreciate it.   As I work on better Windows support, I want to make sure that I do not accidentally make CLR a second class citizen!

kyle-github, 2021-09-09:
I was looking at this again and this is not as clear cut as it seemed.   When you create a tag, a bunch of stuff gets kicked off in the background in another thread.   The connection to the PLC is done completely in parallel to whatever you are doing with the tag handle in the client program.   When this happens, there is actually a background read that is done as well.   What this means is that the library may be ready for you to use the tag but the back end has not finished connecting to the PLC yet.   So you could trigger a read and then later it will fail because the connection is not ready.

So the obvious way to do this won't really help that.  I need to find a way for the client thread domain to be able to see state in the background thread.   Safely.

timyhac, 2021-09-09:
The way I've implemented the InitializeAsync() in .NET is that the first
READ_COMPLETED event it sees it actually interprets as the completion of an
async plc_tag_create(...).

I don't have a good handle on how the core is structured, but when I first
looked at this it would be something around this (
https://github.com/libplctag/libplctag/blob/release/src/lib/lib.c#L366) -
add a flag somewhere that records if this if the first Read or a subsequent
read and switch which event is thrown based on that. Not sure.

kyle-github, 2021-09-09:
I can set it up so that if it is an initial read, then I fire both a READ and CREATE event.  However, there are some tag types that do not do an initial read (raw tags, tag listing, UDT tags...).  So that even is not thrown.  But you really want it to be so that you know when to trigger a read.

Once I figure out what to do about the Windows perf problem to eliminate the sleep() calls I think I will have a better structure for this.

timyhac, 2021-09-11:
Awesome, that would be great - so this would mean that raw CIP and those other tag types you mentioned would also be supported by `InitializeAsync()` - great result!

kyle-github, 2022-02-02:
So still have not got a great structure for this.   But what I will do the following: If any other event is outstanding and we have not fired a CREATE event (PLCTAG_EVENT_CREATED with value 7) yet, I'll fire one.  For now it looks like this will work with normal tags only.   Special tags like raw tags and listing tags do not have reads triggered during creation.   So there isn't anything I can do there...   

I think that full support is going to have to wait until the state of the PLC connection is more closely linked with the state of the tag.  

In the mean time, version 2.5.0 is going to come out soon with support for an extended callback and this partial support for PLCTAG_EVENT_CREATED.

timyhac, 2022-02-02:
Fantastic!

kyle-github, 2022-02-03:
Wait until you try it before you get excited.   There is still issue #338.   This appears to be a race condition due to the fact that the routine that handles the callbacks when events are raised gets the tag status as of the time the routine is called and not as of the time the event was raised.  

I may try to fix that first before I drop 2.5.0 as it is annoying to pass what is essentially the wrong status.

I tried a more sophisticated method of handling events but it was too complicated and had overflow behavior that was not good.   I think I have a simpler solution that might work well enough.

kyle-github, 2022-02-05:
Can you give the prerelease branch a test run?   I think I may still be generating some events more than once but it looks like the PLCTAG_EVENT_CREATED events are getting generated correctly.  

This is all a bit of a dog's breakfast.   Once I start moving the structure over to the way that Modbus is done, it will clean up a lot of code.

kyle-github, 2022-02-09:
I think I have more of this cleaned up now.   Can you give the prerelease branch a try?

timyhac, 2022-02-09:
I haven't done extensive testing on this, I don't have a device to work with. Based on what I can do with the simulator I can say that this is working OK.

kyle-github, 2022-02-09:
Thanks for testing this.  I am still reviewing the code to make sure that all instances of internal access for event/callback handling are safe.

kyle-github, 2022-02-09:
BTW, not that there are two API changes here.   The first is a new type of callback via `plc_tag_register_callback_ex()` and a new tag handle constructor `plc_tag_create_ex()`.  The second one allows you to set up a callback at the time you create the tag handle.  Otherwise, you would miss the CREATED event in some cases.

timyhac, 2022-02-09:
Ah I missed that thanks

kyle-github, 2022-02-14:
Sorry for the delay.  I found out today that I'm not handling the PLCTAG_EVENT_DELETED events right.   They are getting dropped sometimes.   I need to fix that before release.

commit 1b329d1e (2022-02-14): Release for 2.5.0 (#354). Fix #275.  Fix #338. Fix #344. Fix #352.

* Don't call the tag's abort function if it doesn't have one.

* Remove refactored common request status checking code.

* Add common request status checking code.
* Fix duplicate slash in path.

* Bump version for request use-after-free fix.

* callback with user data (#345)

* add callback with user data

* add test_callback2

* rename plc_tag_register_callback2 to plc_tag_register_callback_ex

* rename test_callback2 to test_callback_ex

* refer to each other

* start fix ups of version.

* Break apart the ifdef checks as s [... cut]
closed 2022-02-14 by kyle-github