libplctag/libplctag #276: Python Wrapper .DLL Not Found

Vexx112, 2021-07-29:
Is the Python wrapper still supported for this library? I see in the README.TXT that it “should work”, but upon compiling, Thonny says that libplctag is trying to reference a .dll file that does not exist:

[image]

GitHubDragonFly, 2021-08-09:
You should read the note that is just above the code you posted.

If the file could not be found then nobody put it there.

kyle-github, 2021-08-25:
As @GitHubDragonFly noted, you need to install the C DLL first.   Once it is installed where Python will find it, then the Python code should work.

If you need a pure Python solution, I strong suggest looking at the pylogix or pycomm3 projects on GitHub.  They are really nice and well maintained.   The authors of those projects and I regularly exchange ideas and they are great to work with!

kyle-github, 2021-09-21:
I have tested this on Linux and it finds the DLL just fine in ubuntu_x64. 

Just copy the C DLL into the correct directory and it will be found.  It looks like this is on Windows.  If it is 64-bit x86 Windows, then use the windows_x64 directory and copy plctag.dll into it.

Please reopen if this does not work for you!

closed 2021-09-21 by kyle-github