libplctag/libplctag #286: make CPP wrapper cannot find -lplctag

aa359128, 2021-09-23:
Hi, I'm trying to use cpp code wrapper exemple but I'm not able to call the make command without some errors : 

First I got an error with the file `libplctag.h` missing

`plctag.hpp:8:10: fatal error: libplctag.h: No such file or directory
    8 | #include <libplctag.h>
      |          ^~~~~~~~~~~~~
`

So I found the file `libplctag.h `  in the projet, placed it in include folder and changed `#include <libplctag.h>` by `#include "libplctag.h"` in file `plctag.hpp` .

Now, I'm getting this error :
 
`ccache g++ -Wall -O3 -std=c++11 -DBOOST_LOG_DYN_LINK -m64 -o main obj/main.o obj/plctag.o obj/logging.o -L/usr/local/lib -L/usr/lib -L64_lib -L/usr/lib/x86_64-linux-gnu -L/lib/x86_64-linux-gnu -lplctag -lboost_system -lboost_thread -lboost_timer -DBOOST_LOG_DYN_LINK -lboost_log_setup -lboost_log -lpthread
/usr/bin/ld: cannot find -lplctag
collect2: error: ld returned 1 exit status
make: *** [Makefile:63: main] Error 1
`

I'm currently calling the make command in libplctag/src/wrappers/cpp folder. I read notes in main.cpp file and call all the 3 commands before calling the make command. 

What am I doing wrong ? 

Thanks in advance

kyle-github, 2021-09-24:
You need to install the library so that the .h file is in a location that the compiler will search such as /usr/local/include or /usr/include.  Similarly, you need to install the library .so into a location that the loader will find when the program runs.   For instance /usr/local/lib.

aa359128, 2021-09-24:
Thank you it's working ! Is there a possibility to make it work on windows ?