libplctag/libplctag #288: Automatic disconnect/reconnect

asterisk81, 2021-10-13:
Hi Kyle,
When the connection has automatically disconnected, and a request for a read is received, is the library supposed to reconnect transparently? What would you expect to be returned by the read request that provokes the need to reconnect? I am seeing return codes indicating pending, timeout, and bad connection.

Hi Aravind,

Right now, there is a way to make the disconnect/reconnect time shorter, but not longer.   
It sounds like you need a few changes:

- disable automatic connect and disconnect.
- report the connection status on a tag.
- change the idle timeout.

Is that correct?

I will have to experiment a bit to see what the PLC allows (at least my PLCs) for the idle time.   It is part of the overall set up of a connection for sessions that use connected mode.   There is an underlying limit (from what I have read) but I do not know if it is possible to change that limit in the PLC.   I believe that it is higher than the current 5 seconds.   However, there is an upper limit and I am not sure what it is.  I do not think it is all that long.  Perhaps less than one minute?

Can you describe more about what you are doing that needs these changes?

_Originally posted by @kyle-github in https://github.com/libplctag/libplctag/issues/111#issuecomment-533761896_

kyle-github, 2021-10-13:
The automatic disconnect kicks in after 5 seconds of no activity.  After a fair amount of testing on several ControlLogix PLCs (both using ENBT modules and with the L81's direct connection), the PLC seems to enforce something slightly longer than this.  I have tried different settings in the connection negotiation and nothing really seems to "stick".  It looks like about 5 seconds is the value that works.  6 seconds occasionally fails and 10 seconds definitely is too long.  

I experimented with the automatic reconnection quite a bit and found that the PLC can be annoying if you let it do the disconnect.  Connections that are idle for more than about 5-6 seconds will go into a failed state (I assume, that is at least the behavior I see from the outside) but will not clean up internal resources in the PLC for a while.   As long as the library closes the connection first and does so fairly cleanly, the PLC will clean up the internal resources.   The library goes to some effort to send a Forward Close message for instance.   That seems to be the critical one.

A read that happens after a disconnect will take a longer to return.   Could be 50-100ms, at least on my network and systems.  It should be about the same as the very first read on a tag since it is setting up the connection again.  There are three stages to this: 1) open the TCP connection to the PLC, 2) start a new EIP session, and 3) start a new CIP connection.  Each one has at least one handshake.

If you use the async mode with the timeout to plc_tag_read() set to zero, then you should see a PLCTAG_STATUS_PENDING status come back from plc_tag_status() and you will need to poll the status until the read returns.

What timeout are you using in your code (either via synchronous mode or in your own wait time)?

If you are seeing problems with the reconnect process, possibly your PLC is closing the connection even sooner than after 5 seconds of idle time.

Can you set the debug level (plc_tag_set_debug_level(4)) to PLCTAG_DEBUG_DETAIL and capture the log output?  If I can see the internal state of the library and the packets it sends to the PLC I may get a better idea of what is happening.

kyle-github, 2021-10-20:
I did not realize that the Mesta wrapper was so out of date!   Yeah, you should try Tim and Jody's .Net wrapper.  They have put a lot of work into it and it is really solid.

Using non-zero timeouts lets you queue up multiple tag operations and then wait for them all at once.  This is the approach used in the async and async_stress examples and has the highest performance.

Can we close this issue?

kyle-github, 2021-10-21:
@timyhac thanks for the update.  I did not know that they were taking updates.  That's really cool that there is a clear sharing there now!