libplctag/libplctag #297: Handling Embedded Gateways

Jamesjiveyjr, 2021-11-03:
Hello,

Let me preface this with: I highly appreciate all of the work that has been on this project. Extremely valuable as a diagnostic tool. We have  several hundred processors, every type and configuration you can think of, and would be more than happy to help with testing. 

Due to the high volume of processors and old technology  ( Lot of the equipment was built with PLC 5 controllers and deemed to expensive to upgrade all the processors at the same time. Company went the route of bootstrapping L8s over the PLC 5s to provide safety ) we end up with lots of embedded gateways.

EX
Trying to connect to embedded L81ES

xxx.xxx.xxx.142 / Backplane / Slot 07 ENBT / A (Ethernet) / 192.xxx.xxx.101 / Backplane / Slot 0 == L81ES

I have not been able to resolve this address. On a positive note the library handled our insane amount of DH+ bridging like a champ.

Thanks,

JJ

kyle-github, 2021-11-03:
I have not tested bridging in a while, but it used to work :-)

I think you want a tag attribute string like this:

"protocol=ab-eip&gateway=xxx.xxx.xxx.142&path=1,7,18,192.xxx.xxx.101,1,0&plc=ControlLogix&name=MyNiftyTag"

The magic here is in the "18,192.xxx.xxx.101" part.   18 = Port A.  19 = Port B (if such a thing exists).

kyle-github, 2021-11-03:
Seems to still work. I probably should add that to the tests!

```text
$> ./tag_rw2 --type=sint32 '--tag=protocol=ab_eip&gateway=10.206.1.39&path=1,6,18,10.206.1.40,1,4&plc=ControlLogix&name=TestBigArray[0]'
Library version 2.4.3.
Processing argument 1 "--type=sint32".
Processing argument 2 "--tag=protocol=ab_eip&gateway=10.206.1.39&path=1,6,18,10.206.1.40,1,4&plc=lgx&name=TestBigArray[0]".
data[0]=28 (1c)
```

kyle-github, 2021-11-03:
The ports make more sense in hex: 18r10 = 12r16 where the port is 2.  I was confused above.  Port 1 is the backplane.  Port 2 is the first output port (A).  Port 3 is the second output port (B).  So in hex that is port 12 (for port A), or 13 (for B).

JJIJ88, 2021-11-03:
Ah, That makes entirely more sense. And it works on my test bench. Still can't talk to the ones in the field but error code has changed from bad gateway to timeout. Pretty sure that's on our side ( Our company finds managing bandwidth to be a waste of time )

kyle-github, 2021-11-04:
If you can grab debugging output (set debug level to 4) that may help understand why it is timing out.  If you are on Windows there is definitely some annoying behavior when you attempt to connect to a PLC that is not there.

JJIJ88, 2021-11-05:
Sorry, I've been in the midst of installing safety processors to get SIL levels we need. I reverted back to a different project ( When I had wrapped the library I didn't extend the logging capabilities. ). But I still see the exact situation. 

Level 4 didn't provide anything help. Just times out. Switched to spew and its rather interesting. It does find the PLC. Looks like there's some port conflict  and throws PLCTAG_ERR_REMOTE_ERR but I never see that propagate through.

The error starts on line 620. After that it goes through a few socket states where it ultimately just loops sessions with every session returning SESSION_WAIT_RETY state until the timeout timer.

[SpewSpewSpew.txt](https://github.com/libplctag/libplctag/files/7479629/SpewSpewSpew.txt)

JJIJ88, 2021-11-05:
Attached is a screenshot from linx classic. Sorry for the all the blurring but our company treats IP address like social security numbers. 

[image]

JJIJ88, 2021-11-05:
We have a ton of equipment. Most of the equipment was built around PLC 5s so we add 5000s overtop and make convoluted pathways. 

I see nothing invalid with your description of the path. ENBT is indeed on slot 7. Target is a guard Logix but unless something has changed that's just a controllogix CPU with some safety stuff sprinkled on top.

JJIJ88, 2021-11-06:
I'm confident that it is something on our end.  I haven't quite found it though. I can't talk to that processor not matter which network I come in through ( Net 1 goes through the ENBT/A as in the example above.  Net 2 comes in through the EN2T 192.xxx.xxx.100 ). Ruled out the card types and revision levels testing them on non-bridged application. Funny thing is we have  other equipment that is bridged in roughly the same matter ( same hardware and revision levels ) and I can R/W perfectly fine with them.  Studio 5000 obviously sees everything and easily connects. I'll keep playing around with it when I have time and report back if I find anything.

kyle-github, 2021-11-06:
You might want to see if you can set up a similar PLC but using two plain ControlLogix CPU modules instead of the Safety ones.   I remember that someone said that two synced CPUs would lose sync if they wrote to one of the CPUs.   Maybe that has something to with it.

Just spitballing, but is it possible that there is a duplicate IP address on the network?  I have seen some really weird stuff when that happens.

xinthose, 2021-11-06:
Are you making custom SCADA software for use at your company?  Even if you get this to work, would it be reliable?

JJIJ88, 2021-11-06:
@kyle-github 
I have a couple controls on my desk I could probably replicate with. I would be hesitant to blame cip sync though. Interlocks will throw a fit at the slightest delay in our system ( not by design. Not my design )  Not currently a chance. This is a on one of cleaner systems. No device conflicts anywhere, everything freely talks to each other. 

@xinthose 
Negative. Nor does the library have the capability of being a SCADA software. It allows me to write small programs that fills the void in troubleshooting, simulating, and data collection. Plus always being confined to religious type casting makes me insane.

JJIJ88, 2021-11-06:
@kyle-github  I played around with different setups and to be honest I don't think it has anything to do with bridging. My bad. I can't talk to any of the controllers that are directly mapped via IP onto the network ( As they are in the screenshot above. ) Didn't really hit me until I started playing around with the controller directly on the network. Studio5k connects directly to the controller without going through the backplane ( Library throws bad param when using empty path for ControlLogix. Trying to get to the backplane through the controller doesn't work also ). If I put an ENBT on the rack and connect that to network I can freely communicate with the processor.

JJIJ88, 2021-11-07:
@kyle-github I haven't figured it out quite yet. It's only this on this subnet I have issues. I can communicate to other l81es with along with their partners. Tested both cases ( ENBT to backplane and Directly through the processor ). Tested both cases through a bridge. I just can't see anything on this subnet, including regular control logix. Port not available. While I don't have a solution currently I feel better knowing its most likely something we have configured here (Presumably something that doesn't affect linx.. I can connect to everything via RSlogix through both bridges to this net).  I will certainly circle back to this when I have more time and report any findings in case somebody find its helpful.

kyle-github, 2021-11-07:
Thanks for the update!  This is quite odd.  I think that RSLinx uses UDT to do discovery.   

I think you said that there is more than one *Logix that can bridge to the particular subnet and that all of them show the same issue.   Was that reading correct?

I am honestly stumped here.   The whole "port not available" thing seems to imply that you are getting close to the target PLC and maybe even talking to it.  I'll keep searching to see if I can find some post somewhere that explains the circumstances where that error would be returned.

JJIJ88, 2021-11-08:
Yep, your reading of that is correct. I added a few screenshots below emphasize the point. 

I'm convinced it is talking to the PLC but I haven't been able to infer what is happening yet.  I only get the "port not available" for valid address. I also found that "ErrorNotFound" will prorogate as a "timeout" sometimes if I'm going through a bridge. I just make sure I'm targeting a valid tag name to avoid that confusion. 
[image]
[image]
[image]

JJIJ88, 2021-11-11:
@kyle-github 
I was able to get this working. Watching communication within Wireshark I see CIP forward open request was being sent to non-existent ports. There was a padded 0 in the path. 

[image]

I edited cip.c (cip_encode_path) for testing purposes:

` if(is_dhp && (plc_type == AB_PLC_PLC5 || plc_type == AB_PLC_SLC || plc_type == AB_PLC_MLGX)) {
        /* DH+ bridging always needs a connection. */
        *needs_connection = 1;

        /* add the special PCCC/DH+ routing on the end. */
        tmp_conn_path[conn_path_index + 0] = 0x20;
        tmp_conn_path[conn_path_index + 1] = 0xA6;
        tmp_conn_path[conn_path_index + 2] = 0x24;
        tmp_conn_path[conn_path_index + 3] = dhp_port;
        tmp_conn_path[conn_path_index + 4] = 0x2C;
        tmp_conn_path[conn_path_index + 5] = 0x01;
        conn_path_index += 6;

        *dhp_dest = (uint16_t)dhp_dest_node;
    } else if(!is_dhp) {
        if(*needs_connection) {
            pdebug(DEBUG_DETAIL, "PLC needs connection, adding path to the router object.");

            /*
             * we do a generic path to the router
             * object in the PLC.  But only if the PLC is
             * one that needs a connection.  For instance a
             * Micro850 needs to work in connected mode.
             */
            conn_path_index -= 1; // Test *******
            tmp_conn_path[conn_path_index + 0] = 0x20;
            tmp_conn_path[conn_path_index + 1] = 0x02;
            tmp_conn_path[conn_path_index + 2] = 0x24;
            tmp_conn_path[conn_path_index + 3] = 0x01;
            conn_path_index += 4;
        }

        *dhp_dest = 0;
    }`

This results in a successful forward open using message router. 

[image]

I still have to test cases around this and to see if this is always the case ( Which I doubt). I've been kind of swamped lately so I haven't had too much time to play around with everything

kyle-github, 2021-11-11:
Wow!  This is a very good find.  I am probably adding the padding intentionally because the whole path needs to be a multiple of 16-bit words in length.

I will need to do some experiments to see if I can make it fail with just padding at the end.

I am traveling for with this week and will not be able to get to this untill the weekend.

kyle-github, 2021-11-13:
I'm back home.  I should have noticed there is an entirely different bug here:

```c
if ((*addr_seg_len) && (uint8_t)0x01)
```

The problem is "&&".  This should be "&" for a binary mask.  I will fix that immediately.

There may still be an issue with the zero-padding of the IP address string.

kyle-github, 2021-11-13:
I think this is the actual bug.  I checked with odd length and even length addresses and there was always a zero padding byte before.  Unfortunately, all my network addresses are odd in length so they should have a zero padding byte.

This will be fixed in the fix_303 branch which I will push either tonight or tomorrow morning.

commit 72807d69 (2021-11-13): Fix #303.  Signal cond var when there are still requests in the queue. Fix #297. (#304)

* Fix #303.  Signal cond var when there are still requests in the queue.

* Add more debugging output when more requests are queued.

* Output detailed log.

* Fix incorrect && that should be &.  Fixes #297.
closed 2021-11-13 by kyle-github