libplctag/libplctag #298: Exception 0xc0000005

crapher, 2021-11-03:
Hi Kyle, 

I was testing the last version of the library (2.4.2) and everything worked excellent until I started with the acid tests when I received an Access Violation Exception.

The error appears to be random (I don't think it is, it is just I could not generate it every time) but I could have some logs when it happens.
I attach some of them to see if there is something that can help to find the problem.

My system configuration is:
- Windows 10 21H1 x64
- libplctag 2.4.2 [x86]
- PLC AB Logix5563

What I am doing to generate the error is:
- Disconnect the PLC (I have a USB to Ethernet converter, with disconnect the PLC I mean, I disconnect the USB cable)
- Execute the application that uses the PLC
- Wait a couple of seconds and connect the PLC again.

I am going to try to generate the error again to give you a way to generate it also on your end.

[Log1.txt](https://github.com/libplctag/libplctag/files/7470148/Log1.txt)
[Log2.txt](https://github.com/libplctag/libplctag/files/7470150/Log2.txt)

Thanks!
Regards
Diego

kyle-github, 2021-11-03:
Thanks for the excellent bug report!   The exception 0xc0000005 is raised when the machine tries to read an invalid pointer.

There was a change in the way that non-blocking operations were handled between 2.4.0 and 2.4.1.  I0xc0000005 strongly suspect that is why you see this problem now but not then.

Based on the logs, it looks like there is a dangling pointer left somewhere.   Interestingly, my normal tests are not triggering this on Linux under Valgrind.   All but the platform level code is the same across all supported OSes, so if there is a logical flaw in the non-platform code, it should behave the same on all platforms.

I will be trying to repro this (thanks for the steps but my PCs are all on WiFi!).  I am going to try to recreate the issue locally.

One thing I noticed right away is that there is a very long time (20+ seconds) when the main session/PLC thread is hung waiting for connect() to return.  I am working on that in a side branch right now as it was brought up by another user.  See issue #295.

I found a tool on Windows that seems to provide many of the same benefits as Valgrind.  I will use that to figure out what is going on.

kyle-github, 2021-11-04:
I have done some more experimentation here and have not been able to repro this yet. From the log files it appears that there is some sort of use-after-free bug happening.   However, it is unclear exactly where because of issue #295.   That is what is causing the very long delays in the log.   You should not see longer than the set timeout (default in tag_rw2 is 5 seconds).  

I will keep trying until I get this.

crapher, 2021-11-04:
I am trying to generate a toy example to reproduce the problem.  
It is kind of difficult because it does not happen always, it happens sometimes which makes it harder to reproduce the scenario.  I think as you mentioned, it is a use-after-free bug caused by a timing problem in the reconnection.

crapher, 2021-11-04:
Kyle, I could reproduce the error with full log (log level 5).  

[Exception_full_log.zip](https://github.com/libplctag/libplctag/files/7475466/Exception_full_log.zip)
[Exception_full_log_1.zip](https://github.com/libplctag/libplctag/files/7475485/Exception_full_log_1.zip)

There are some ref counts that caught my attention. (I think they are consequence of the problem)
- Exception_full_log: Line 161874 -> Ref count is 10093209
- Exception_full_log_1: Line 161542 -> Ref count is 18568985

Also, there are some Invalid ref count errors on both logs
- Invalid ref count (0) from call at tag_tickler_func line 301!  Unable to take strong reference.

In this log (Log level 4)
[Exception_MUTEX_DESTROY.zip](https://github.com/libplctag/libplctag/files/7475610/Exception_MUTEX_DESTROY.zip)

There is another thing that caught my attention (maybe it is a coincidence) but 2 mutexes were potentially destroyed twice (I am not 100% sure because they belong to different tags but it sounds weird).
- mutex 012B8928 (Line 548 / Line 783)
- mutex 012B892C (Line 553 / Line 785)

I'll try to continue looking to narrow the problem.

kyle-github, 2021-11-05:
Thanks for this invaluable log information!

I took the log files and split them out by thread (grep is your friend and this is why the log files have the information they have in them).  I found something that looks like the smoking gun.

Thread 3 seems to show a failure on tag creation.  It times out.   However, during the process, I can see where it allocates a new tag ID (14) and then adds it to the internal hash table which is used for looking up tags.  The entry in the hash table is a pointer to the internal tag structure.   Then it waits for the connection to the PLC to complete.  Which times out.   It then calls up the clean up routines and everything gets destroyed.   So far so good.

But it does not remove the tag entry from the hash table.  So the hash table is still sitting there with an entry for key "14" and a pointer to the (now destroyed) old tag memory.   

There is another thread (thread 6 in the MUTEX DESTROY log) that runs through all the tags in the hash table and calls a function on each one to check whether it needs to do anything or not.   

Notice the tag IDs:

```text
2021-11-04 09:08:37.523 thread(6) tag(0) DETAIL tag_tickler_func:498 Tag tickler thread timed out waiting for something to do.
2021-11-04 09:08:37.523 thread(6) tag(13) DETAIL tag_tickler_func:314 Tickling tag 13.
2021-11-04 09:08:37.523 thread(6) tag(14) DETAIL tag_tickler_func:314 Tickling tag 14.
2021-11-04 09:08:37.523 thread(6) tag(12) DETAIL tag_tickler_func:314 Tickling tag 12.
2021-11-04 09:08:37.523 thread(6) tag(15) DETAIL tag_tickler_func:314 Tickling tag 15.
2021-11-04 09:08:37.523 thread(6) tag(11) DETAIL tag_tickler_func:314 Tickling tag 11.
```

So far so good.  We see it run across all 5 tags that were in the process of being created.  Then it waits a bit and starts over.

```text
2021-11-04 09:08:37.523 thread(6) tag(0) DETAIL cond_wait_impl:1126 Starting.
2021-11-04 09:08:37.523 thread(6) tag(0) DETAIL cond_wait_impl:1128 Starting. Called from tag_tickler_func:496.
2021-11-04 09:08:37.556 thread(6) tag(0) DETAIL cond_wait_impl:1151 Condition var wait returned.
2021-11-04 09:08:37.556 thread(6) tag(0) DETAIL cond_wait_impl:1174 Condition var signaled for call at tag_tickler_func:496.
2021-11-04 09:08:37.556 thread(6) tag(0) DETAIL cond_wait_impl:1185 Done for call at tag_tickler_func:496.
```

The wait is done. 33ms seems like a strange amount of time, but ... Windows.  Now tickling the tags...

```text
2021-11-04 09:08:37.556 thread(6) tag(0) DETAIL tag_tickler_func:314 Tickling tag 0.
2021-11-04 09:08:37.556 thread(6) tag(0) WARN default_tickler:778 This should be overridden by a PLC-specific function!
2021-11-04 09:08:37.556 thread(6) tag(0) DETAIL tag_tickler_func:314 Tickling tag 0.
2021-11-04 09:08:37.556 thread(6) tag(0) WARN mutex_try_lock_impl:794 null mutex pointer.
2021-11-04 09:08:37.556 thread(6) tag(16) DETAIL tag_tickler_func:314 Tickling tag 16.
2021-11-04 09:08:37.556 thread(6) tag(16) DETAIL tag_tickler_func:314 Tickling tag 16.
2021-11-04 09:08:37.556 thread(6) tag(11) DETAIL tag_tickler_func:314 Tickling tag 11.
2021-11-04 09:08:37.556 thread(6) tag(11) WARN mutex_try_lock_impl:794 null mutex pointer.
```

Uh oh.  That doesn't look right.  At all.   So we've got the tag ID is zero which it should never be.   And we've got tag 16, which we never had before.    And we have a null pointer to a mutex.   In the clean up routines I normally set all pointers to null to catch just this kind of problem.  So by this time, we've had another piece of memory allocated over the top of at least one of the freed tags.   Most implementations of malloc() will reuse the same memory it just freed if it is the same size (which it would be here.

The code fix to ensure that the tag is removed from the hash table before it is destroyed is simple and I am building a release based on that now.

So in order to trigger this, we need to have a failure in plc_tag_create, which means you must use blocking mode with a timeout.  Then you have to create a tag after the failure. 

Thank you so very, very much for all the help with this!

closed 2021-11-05 by kyle-github

kyle-github, 2021-11-06:
Please let me know if version 2.4.3 prevents the problem from happening again.   I am still unable to repro it here.  I have some ideas I am going to try this weekend.

crapher, 2021-11-06:
Yesterday, after you released version 2.4.3, I used the same code to run, first version 2.4.2 and then version 2.4.3 (just changing the DLL).
After 10 tests of each one, version 2.4.2 crashed 8 times and version 2.4.3 never crashed.  I think the change you did, fixed the problem.
On Monday, I will do more tests with the real PLC.

Not related to this bug, and just to let you know (I think you already know it), checking the speed between version 2.4.0 and 2.4.2 the improvement was around 50%, and between version 2.1.4 (the one I was using) and 2.4.0 was around 500%.  That is a HUGE improvement.  On Monday I will check version 2.4.3 (I don't think I am going to see any difference with version 2.4.2)

Thank you very much for the work and effort you put into the library.

kyle-github, 2021-11-06:
Thanks for testing again!

The performance is troubling.   I have followed everything I could find about Windows performance but I think I may need to adjust the timer period back down.   As you noted the performance difference with the short timer period is much, much higher than without it.  My only test Windows system is in a VM and I think I am not getting useful performance data from it.

I have a question about the performance you are seeing.   Let's call the performance of version 2.1.4 1.0.   So version 2.4.0 has a performance of 5.0 (500% improvement).   Was 2.4.2 50% larger than 2.4.0 (so 7.5 or 750% of 2.1.4) or 50% larger than 2.1.4?

I certainly hope that there isn't a performance difference between 2.4.2 and 2.4.3!   It is only in an error path that there were any changes (it was about 9 lines of code).

crapher, 2021-11-06:
I have a laptop with Windows 10 21H1 (It is not a VM) and the quick performance test I did showed me that version 2.4.2 is around 10 times faster than 2.1.4 and version 2.4.0 is around 5 times faster than 2.1.4.

I will check better on Monday and I will give you better information about this.  If you have any script, program or something, you want me to run to check the performance of the library, let me know and I'll run it.
I have a tester developed in C# used to test a library that uses the plctag.dll.  On that test, I read and write tags with different data types, arrays of different data types, and UDTs (string and fake strings built with a DINT and an array of SINT)

About version 2.4.3, I agree with you, it was a small change and it should not affect in any way the performance of the library.

crapher, 2021-11-08:
Kyle, here are the logs of the tester I am using to test my library.
I am using the same code (without recompilation) to generate the logs.  I just changed the plctag.dll.

[log.2.1.4.txt](https://github.com/libplctag/libplctag/files/7497397/log.2.1.4.txt)
[log.2.4.0.txt](https://github.com/libplctag/libplctag/files/7497395/log.2.4.0.txt)
[log.2.4.3.txt](https://github.com/libplctag/libplctag/files/7497396/log.2.4.3.txt)

These are the tags I created in the PLC to do the test
```
******************************************************************************
*  NAME                                    DATATYPE                          *
******************************************************************************
*  BOOL                                    BOOL                              *
*  SINT                                    SINT                              *
*  INT                                     INT                               *
*  DINT                                    DINT                              *
*  LINT                                    LINT                              *
*  REAL                                    REAL                              *
*  STRING                                  STRING                            *
*  LONG_STRING                             UDT (LEN: DINT | DATA: SINT[255]) *
*  BOOLARRAY                               BOOL[64]                          *
*  SINTARRAY                               SINT[10]                          *
*  INTARRAY                                INT[10]                           *
*  DINTARRAY                               DINT[10]                          *
*  LINTARRAY                               LINT[10]                          *
*  REALARRAY                               REAL[10]                          *
******************************************************************************
```
The header of each group means the way I am reading/writing the tags.  
ex. BOOLEAN - READ means that I am reading all the tags in the group as Booleans, and CHECK OK means I am receiving the expected value. (don't worry about that, it is just to test my library)

Check the elapsed time on each line in each file and you will see the performance difference between versions 2.1.4, 2.4.0, and 2.4.3.

kyle-github, 2021-11-08:
Thanks!

Interesting values.  I will see if I can separate them out into a spreadsheet.   I need to implement a test program like yours that will let me track the latency etc!

Do these tests open the tag, read it once and then do the performance test?   Creating a tag has a lot of overhead for the first tag in a PLC.  In the background, the library sets up a complex connection with the PLC and that takes a lot of time and handshaking.

kyle-github, 2021-11-08:
I use the stress_test, async_stress, and thread_stress programs.  Those all either hide the start up cost or run long enough to average it out.

crapher, 2021-11-10:
Hi Kyle,
I ran the async_stress test with 2 tags (a DINT tag and a DINT array of 10 positions tag)

The following logs have the results:

Version 2.1.4
- [test.2.1.4.async_stress.log](https://github.com/libplctag/libplctag/files/7514883/test.2.1.4.async_stress.log)

Version 2.4.0
[test.2.4.0.async_stress.log](https://github.com/libplctag/libplctag/files/7514884/test.2.4.0.async_stress.log)

Version 2.4.3
[test.2.4.3.async_stress.log](https://github.com/libplctag/libplctag/files/7514885/test.2.4.3.async_stress.log)

Version 2.4.3 (with async_stress.exe version 2.4.0 - to have similar output)
[test.2.4.3.async_stress.async.2.4.0.log](https://github.com/libplctag/libplctag/files/7514886/test.2.4.3.async_stress.async.2.4.0.log)

I don't see the difference I mentioned in my previous message between versions 2.4.0 and 2.4.3 according to that test.

kyle-github, 2021-11-11:
Thanks for running those tests!

I am traveling for with and won't be able to give these proper attention until the weekend.

commit a8e009d5 (2021-11-05): Fix  #298. Memory corruption when tag creation fails followed by reconnection. (#299)