libplctag/libplctag #303: Excesive timeouts on threads

mad93, 2021-11-08:
Hello,

I'm trying to read several variables of an Omron NJ5 with CIP protocol and the performance is so slow.

Using Wireshark I see that the PLC takes 4ms to return a response with the data requested, but between each request is taking 100 ms aprox before doing another request.

Enabling debugging on the library I've seen these lines:

> 2021-11-08 08:24:57.536 thread(2) tag(0) DETAIL cond_wait_impl:1032 Starting. Called from tag_tickler_func:496.
> 2021-11-08 08:24:57.536 thread(2) tag(0) DETAIL cond_wait_impl:1060 Waiting for 100ms.
> 2021-11-08 08:24:57.628 thread(3) tag(0) DETAIL cond_wait_impl:1067 Timeout response from condition var wait.
> 2021-11-08 08:24:57.628 thread(3) tag(0) DETAIL cond_wait_impl:1092 Condition wait terminated due to error or timeout for call at session_handler:1238.
> 2021-11-08 08:24:57.628 thread(3) tag(0) DETAIL cond_wait_impl:1100 Done for call at session_handler:1238.

Following these logs I've ended in src/platform/posix/platform.c at function cond_wait_impl. This function receives a variable called timeout_ms that I'm unable to see where is defined or the function calling this one, so I forced this timeout to 1. 

Doing so, reading 15 arrays of 50 DINT values goes from 1,5s to 300ms, which is a huge performance improvement.

Also tried with 10ms, so I can reduce the  load in the CPU, and takes about 370ms

Can you help me to understand from where this timeout_ms comes, so I can solve this problem without 'hacking' the library?

Thank you

kyle-github, 2021-11-08:
Hi, thanks for bringing this up.  I am getting conflicting reports on performance with newer versions of the library.  Which version are you using and what OS are you using?

This looks more like a logical bug or problem than a timeout problem.  That condition variable should not be hit at all if the code is waiting for a response.  That condition variable wait is only for the case that the internal thread has nothing to do and is not waiting on a response.

Can you capture all the logs and add them here?   Debug level 4 would be very helpful.  Thanks!

kyle-github, 2021-11-08:
See the commentary in issue #298 for examples of where the timing changed for the better with 2.4.3.   I am not saying you are not seeing what you are seeing.   But this is odd.  I wonder if it has something to do with the target PLC being an Omron?  Seems unlikely, but I am looking through the code and it is not obvious why there would be such a difference for one PLC than another using the same code.

BTW, thread 2 is a tag tickler thread.  It should be waiting a fair amount.  Thread 3 is probably (timing can change the ID) the thread for the PLC session and should not be waiting if the thread is waiting for a response from the PLC.

mad93, 2021-11-08:
Thank you Kyle,

Maybe I'm using a weird mix, using Ubuntu on wsl2 on a Windows 10 Pro machine. Production machine would be running with a docker instance, so I don't know if there will be the same issue, not yet there on development.

Added log as a file, since is quite large.

[output.txt](https://github.com/libplctag/libplctag/files/7498193/output.txt)

I'll look on the issue #298

kyle-github, 2021-11-09:
Oh, interesting!  I have not tried to run WSL before.   I use almost the opposite: Windows in a VM on Linux.  It is a lot less performant than running the tests in a VM.  I understand that WSL is just Linux in a HyperV container.   That should be fairly performant.

Can you try running the same tests directly on Windows?   The library will work and build with the free version of MS Visual Studio.  If that works fine then I suspect that it is something about WSL that impacts the network functions.

Thanks for the log file.  I think there might be a bug here.  There is some logic flow that is not signaling the PLC session thread's conditional variable when it should be.

kyle-github, 2021-11-09:
Are you able to try the new branch "fix_303"?   If you can, can you please retest.   I found a spot where use of PLCs that do not pack requests (Omron does not) will end up in some circumstances waiting for the whole conditional variable timeout period (100ms).

I will be traveling for work for the rest of the week but I will check email and GitHub.

mad93, 2021-11-09:
Hello Kyle and thank you for your time,

The fix_303 didn't improve the performance, here the log [output.txt](https://github.com/libplctag/libplctag/files/7502410/output.txt).

I'll try on Windows later to see if it improves without using WSL.

kyle-github, 2021-11-09:
Thanks for testing!   Unfortunately, I am not sure that the correct version was tested.  I do not see the debugging output I added at the sites where I changed code.   Can you double check that you are calling the right library version?

I pushed another update to fix_303.  This has more output like this:

```text
2021-11-09 07:25:58.909 thread(3) tag(0) DETAIL session_handler:1105 There are 6 requests pending before cleanup and sending.
```

and

```text
2021-11-09 07:25:58.909 thread(3) tag(0) DETAIL session_handler:1138 There are 6 requests still pending after abort purge and sending.
```

If you do not see those lines, then your test program might be finding the old library version.  If you use shared libraries, it is possible that the system-level library is being found before your local version.

Thanks very much for being patient with this process!

I am a little concerned about GitHub right now.  I had to reload multiple times to see the commit show up in the branch.   I wonder if GitHub did not have the commit that changed the checks?

Please try fix_303 again.   If you can look at the code, look at line 1135 in src/protocols/ab/session.c.   It should read:

```c
            critical_block(session->mutex) {
                int num_reqs = vector_length(session->requests);
                if(num_reqs > 0) {
                    pdebug(DEBUG_DETAIL, "There are %d requests still pending after abort purge and sending.", num_reqs);
                    cond_signal(session->wait_cond);
                }
            }
```

Can you please try again?

mad93, 2021-11-10:
Hi Kyle,

I cloned a new library to make the test and seems the changes weren't there. Now I've made a pull and got the changes on branch.

I think you nailed it! With debug got less than a second and without it about 340 ms!

I see that now is not freezing all the threads 100ms as before:

> 2021-11-10 08:03:16.531 thread(2) tag(48) DETAIL tag_tickler_func:314 Tickling tag 48.
> 2021-11-10 08:03:16.532 thread(2) tag(0) DETAIL cond_wait_impl:1032 Starting. Called from tag_tickler_func:496.
> 2021-11-10 08:03:16.532 thread(2) tag(0) DETAIL cond_wait_impl:1060 Waiting for 98ms.
> 2021-11-10 08:03:16.532 thread(1) tag(0) DETAIL session_add_request_unsafe:899 Starting.
> 2021-11-10 08:03:16.532 thread(1) tag(0) DETAIL session_add_request_unsafe:918 Total requests in the queue: 36

Here you have the debug output, 
[output.txt](https://github.com/libplctag/libplctag/files/7510547/output.txt)

kyle-github, 2021-11-13:
Thanks for checking again!

I will release this today.   

The Omron series is slightly different from many other CIP PLCs.   Do you know what commands it supports?   I suppress use of command 0x0A (multiple request packets) and do not allow tag listing.   

Can you do some tests for what is supported?  By turning off support for multiple request packets, you should be able to simulate an Omron but trick the library into thinking that it is talking to a ControlLogix:

```text
./tag_rw2 --type=uint8 '--tag=protocol=ab-eip&gateway=10.206.1.40&path=1,4&plc=ControlLogix&name=@tags&allow_packing=0' --debug=4
```

You would need to make the path correct for the Omron.  If this comes back with data and not an error, then the Omron does support tag listing.

closed 2021-11-13 by kyle-github

mad93, 2021-11-15:
Hello Kyle, 

I don't know if i should answer here or in a  new issue. Anyway, seems that omron NJ5 doesn't support tag listing, using fix_303 branch.  With wireshark I see that the PLC is answering with "Service not supported".

Here is the ouptut to the suggested comman
[output.log](https://github.com/libplctag/libplctag/files/7536507/output.log)
d

kyle-github, 2021-11-16:
Thanks for testing!  I will go through the log and see if I can find anything obvious.

kyle-github, 2021-11-16:
Well, there was at least one interesting thing.   The Forward Open negotiated a packet size of 1996 bytes.   That is a very strange size that I have never seen before!   It would be interesting to see if that works when in "normal" Omron mode.  If it is not trying to set the packet size that high, then I need to change some code.

Unfortunately, it is responding to the tag listing command with an unsupported error.   I will do some research to see if there is anything that indicates a different command should be used.

mad93, 2021-11-16:
In Omron CIP Communication Server User Manual I see this:

> There is a 1994 bytes maximum length limitation for CIP class 3 connection
> (Large_Forward_Open) messages, so the maximum number of items which can be read
> by one message depends on read message type. The OmronCIP Server currently
> supports two types of read messages:
> 1) Multiple Services Packet (Service Code 0A) messages, used for reading of single
> Global Variables, except array elements. The maximum number of items which can be
> read by one Multiple Services Packet message is limited to 32 items (it is limited by length
> of request command sent to Controller).
> 2) Read Service for Variables (Service Code 4C) messages, used for reading of arrays
> – in case item name indicates this item is array element, the whole array (starting from
> beginning) is read from Controller. Therefore it is important to use the array elements with
> small indexes, as the maximum number of items which can be read by one Read Service
> for Variables is limited by Controller response message. For example, the maximum
> number of DWORD (double word) type array elements which can be read from Controller
> is 495 double words.
> Note: If possible, it is highly recommended to use arrays – that can greatly increase the
> OmronCIP Server performance in case it is necessary to read large amount of data.

Also here is the ForwardOpen call when trying to list tags setting omron as the plc (tries to get like a variable):

> 2021-11-16 15:16:56.233 thread(3) tag(0) INFO receive_forward_open_response:2199 Starting
> 2021-11-16 15:16:56.233 thread(3) tag(0) INFO recv_eip_response:1883 Starting.
> 2021-11-16 15:16:56.237 thread(3) tag(0) DETAIL socket_read:1444 Socket can read data.
> 2021-11-16 15:16:56.237 thread(3) tag(0) DETAIL socket_read:1444 Socket can read data.
> 2021-11-16 15:16:56.237 thread(3) tag(0) INFO recv_eip_response:1954 request received all needed data (70 bytes of 70).
> 2021-11-16 15:16:56.237 thread(3) tag(0) INFO recv_eip_response:1956  6f 00 2e 00 47 01 fc be 00 00
> 2021-11-16 15:16:56.237 thread(3) tag(0) INFO recv_eip_response:1956  00 00 ba 50 f3 11 00 00 00 00
> 2021-11-16 15:16:56.237 thread(3) tag(0) INFO recv_eip_response:1956  00 00 00 00 00 00 00 00 01 00
> 2021-11-16 15:16:56.237 thread(3) tag(0) INFO recv_eip_response:1956  02 00 00 00 00 00 b2 00 1e 00
> 2021-11-16 15:16:56.237 thread(3) tag(0) INFO recv_eip_response:1956  db 00 00 00 c1 22 3b 46 65 38
> 2021-11-16 15:16:56.237 thread(3) tag(0) INFO recv_eip_response:1956  55 07 d2 66 3d f3 45 43 50 21
> 2021-11-16 15:16:56.237 thread(3) tag(0) INFO recv_eip_response:1956  40 42 0f 00 40 42 0f 00 00 00
> 2021-11-16 15:16:56.237 thread(3) tag(0) INFO recv_eip_response:1963 Done.
> 2021-11-16 15:16:56.237 thread(3) tag(0) INFO receive_forward_open_response:2261 ForwardOpen succeeded with our connection ID 7553865 and the PLC connection ID 463b22c1 with packet size 508.
> 2021-11-16 15:16:56.237 thread(3) tag(0) INFO receive_forward_open_response:2266 Done.

kyle-github, 2021-11-16:
Thanks!  There is a lot of useful information in your note here.  Could you open a new issue for "Omron performance"?

Is the document you quoted publicly available?  If it is, could you attach it to the new issue?

I did not add the original Omron support and I do not have one, so I need to rely on users like you for information!  From what you posted it is clear that performance on these PLCs could be a lot better.

commit 72807d69 (2021-11-13): Fix #303.  Signal cond var when there are still requests in the queue. Fix #297. (#304)

* Fix #303.  Signal cond var when there are still requests in the queue.

* Add more debugging output when more requests are queued.

* Output detailed log.

* Fix incorrect && that should be &.  Fixes #297.