libplctag/libplctag #305: Reading/Writing different tags with dedicated threads at the same time.

robintomar1, 2021-11-11:
Hi all, 
 
I am working with an Omron Plc, my application requires reading and writing from PLC at frequency greater than 100hz.
Average ping time to Plc is about 4ms.
I used two tags one for read and other for write. These tags are connected to dedicated read and write registers in the PLC.
Average read time (read tag and get data from library to user code) takes about 5-8 ms, same for the write operation.
So my idea was, since i have different memory locations in plc for read and write, I can do these operations in parallel as they wont affect each other and get a frequency above 100HZ. 

But the calls to read_tag() write_tag() functions seems to blocking each other even when they are made from different threads and for different tags.
I am not sure if what i am thinking is right. Any suggestion would be highly appreciated.
thanks.

kyle-github, 2021-11-11:
The library will attempt to reuse the same TCP connection to the PLC for its tags.  That means that only one request can be in flight at a time.

I used to have code in place that would let you stop this sharing but I think I removed it due to lack of use.

I will check when I get back from travelling this weekend.

robintomar1, 2021-11-11:
Thanks for the quick response @kyle-github kyle.
Would be great if you could get that.
Safe travels !!

kyle-github, 2021-11-25:
I see that I forgot about this.   The code appears to still be there.   Add "&share_session=0" to your tag attribute string.  That will cause every tag (with that attribute) to use a different TCP connection.  If you add that to all your tags, then each one will have a separate session to the PLC.

Note that I have no idea what the PLC does internally.   Perhaps it batches up the requests and processes them serially?

kyle-github, 2021-11-25:
Please reopen if this is not working for you.

closed 2021-11-25 by kyle-github

robintomar1, 2021-11-29:
Hey @kyle-github , thanks for the input, it worked for me.