libplctag/libplctag #307: Reading strings in omron NJ5

mad93, 2021-11-16:
Hi Kyle,

I'm trying to read strings from the PLC and I'm having problems. Seems that str_count_word_bytes is set to 4 (default on compactlogic?). I've found that must be set to 2 to work properly (2 bytes by char I understand).

I've added on src/protocols/ab/ab_common.c after line 254, on omron customitzations I've added:
`attr_set_int(attribs, "str_count_word_bytes", 2);`

Thank you.

kyle-github, 2021-11-17:
Thanks for the information.   Is there more documentation about data types on Omron PLCs?

Control and CompactLogix both use strings that are 88 bytes with 82 bytes of string data and a 4-byte count word and 2 bytes of padding.  PLC-5, SLC and MicroLogix use 84 bytes for strings with 82 bytes of string data and a 2-byte count word and no padding.  It sounds like the Omron PLC might be closer to that.

Strings can be very strange in PLCs.   The MicroLogix has byte swapped characters (in big-endian word order).

Can you please read in a single string and see how many total bytes there are?   You can use the tag_rw2 tool and output the data with type uint8.   That will show the total bytes.  

In order to set up string support correctly I need to know:

- how long the count word is.   Sounds like it is 2 bytes.
- how long the string data is.   On some PLCs like the Micro800 series it is 255 characters.
- if there are any padding bytes and where they are.
- if there is any null character termination (as in C strings).
- whether or not there is any byte swapping of character bytes.

Some PLCs have different kinds of strings with different data.   Allen-Bradley Control/CompactLogix use at least three different string types.  The main one for tags which is 88 bytes, another one for tag listing which uses a 2-byte count word but is variable length, and a final one for UDT field names which is a C-style string with no count word and nul byte termination.

kyle-github, 2021-11-17:
Do you know if it is possible to list the tags in an Omron PLC?   The programming tools must be able to do it somehow.

mad93, 2021-11-17:
Hi Kyle,

About strings:

- Null terminated Unicode string
- String lengths range from 1 to 1985
characters. This equates to variables defined in
Sysmac Studio as STRING[2] and STRING
[1986] respectively. The extra character
accounts for the null terminator.

I've extracted this information from the document linked on #309 

You need anything else? There is information missing?

mad93, 2021-11-17:
About PLC tag listing, as you can see in [this document](https://assets.omron.eu/downloads/manual/en/v3/v240_application_design_guide_for_cip_communications_technical_manual_en.pdf) the sofware SYSMAC Gateway is capable of doing so (page 26). I have read on forums a guy that suggest looking with WireShark how the application is doing so.  

It's weird that this guy [doen't want to explain](https://forums.mrplc.com/index.php?/topic/33174-wireshark-capture-of-pc-omron-nj-via-cip-protocol/&do=findComment&comment=169135) the instruction used to obtain the tag listing, maybe because as is not being documented anywhere is not oficial and may be inconsitent between PLC models?

Sadly I don't have this software and requires purchase a license.

kyle-github, 2021-11-17:
Huh.  So there is a 2-byte count word and then a nul-terminated set of data bytes?   That's a strange mix.   I think I can represent that with my current string configuration.  It will take some experiments!

Thanks for the information about the mrplc.com forums.  I have sometimes found some information there, but often only enough to know that the information exists but not what I was looking for :-/

kyle-github, 2021-11-26:
I just released a prerelease of support for Omron strings.   They use a 2-byte count word and nul-byte termination from your description above.   

If you go to the GitHub site, you can download binaries [here](https://github.com/libplctag/libplctag/releases/tag/v2.4.6-pre).  Can you please give this a test?

There are also some performance enhancements such as making the negotiated packet size 1996 bytes instead of 508 and supporting multi-request packets.

kyle-github, 2021-12-13:
@mad93, this is now fully released.  If you still have access to an Omron, could you please give this a try?   Thanks!

mad93, 2021-12-13:
Hi Kyle,

Just tested the new library version and is not working properly.

Looking with wireshark complains about multiple service packet request being too large:

> 39	7.795042	192.168.250.224	192.168.250.100	CIP	592	Multiple Service Packet: Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c), Service (0x4c)
> 41	7.802071	192.168.250.100	192.168.250.224	CIP	116	Routing failure, response packet too large: Restore

When I tried the limits of data transmission, the NJ5 PLC gave me an error when trying to get an array of more than 125 DINT, and iwn this multiple service packet I see a lot more of data requested. In the request are grouped 26 variables I'm requesting. Almost all of these variables are arrays of REAL, DINT or BOOL.

Seems to understand the Multiple Service Packet but is way too large.

kyle-github, 2021-12-13:
I am not sure what to make of the 125 DINT limit.   That is 500 bytes.   There is no protocol limit for that.

mad93, 2021-12-14:
I can't list the tags  on the PLC. This is the returned error:

> Class (0x6b) - Service (0x55)
> Service not supported: Class (0x6b) - Service (0x55)

kyle-github, 2022-01-11:
Hi @mad93, can you try the "list_tags_logix" program in the latest release?

Also, there were some string-related fixes in 2.4.10.   I think you should be able to use the `str_*` tag attributes in the tag string to set up the right definition.   Can you try those?

kyle-github, 2022-11-16:
Please reopen if you get a chance to test this.  Thanks for all the data about Omron PLCs!

closed 2022-11-16 by kyle-github

zN3utr4l, 2024-03-12:
I found a library written in python (https://github.com/aphyt/aphytcomm) that might help with Omron.

It also has tag discover, as well as read/write (works on strings)