libplctag/libplctag #310: protocolo de comunicacion IP

InofensivoJimmy, 2021-11-22:
Podría obtener el protocolo de comunicación IP de esta librería?

kyle-github, 2021-11-23:
If you want documentation of the CIP/EIP protocol, please search for the document "1756-PM020G-EN-P".   That describes much of the protocol you need.  Rockwell has the document published on their site.

closed 2021-11-23 by kyle-github

InofensivoJimmy, 2021-12-06:
Helloi could you send me or give me examples of libplctag4j?

("", "protocol=ab-eip&gateway=10.206.1.38&plc=PLC5&elem_size=2&elem_count=1&name=N7:0",TIMEOUT)

protocol=ab-eip "aqui se indica el tipo de plc (allen bradley)

gateway=10.206.1.38 " ip del equipo"

plc=PLC5 " ? "

elem_size=2&elem_count=1 " PATH"

name=N7:0 " nombre del Tag (se puede sustituir por el nombre real del tag?)

InofensivoJimmy, 2021-12-07:
Pretendo comunicarme con la librería libplctag4j, con Rslogix emulate 500.
me gustaria hacer una prueba con tag booleano llamado TAG_PRUEBA. como
debería completar el siguiente string?
 tag = new Tag(
"protocol=ab-eip&gateway=10.206.1.39&path=1,5&plc=ControlLogix&elem_count=1&name=TestStringArray",
TIMEOUT);

kyle-github, 2021-12-07:
Please reach out to the [forum](https://groups.google.com/g/libplctag/).   There was a recent thread there where the result was that RS Emulator 5000 does not work.  It does not provide a network interface.

See the [thread](https://groups.google.com/g/libplctag/c/cxfC6H2PWwA/m/Ix1uhb_rAgAJ).