libplctag/libplctag #319: comunicacion con rslinx emulate500

InofensivoJimmy, 2021-12-07:
Quiero comunicarme con un tag con el nombre TAG_PRUEBA de tipo bool. Utilizo el elmulador rslinx emulate500 para hacer la prueba.
Using library version 2-3-4
"protocol=ab-eip&gateway=192.168.18.16&path=1,2&plc=ControlLogix&elem_size=4&elem_count=1&name=TAG_PRUEBA",TIMEOUT)
no tengo claro elem_size, ni siquiera si es necesario ponerlo. ver

lpc921, 2021-12-09:
Emulator doesn't provide the CIP interface. You need a real PLC or SoftLogix.

kyle-github, 2021-12-10:
Thanks @lpc921!

Please reopen if there are further questions.

closed 2021-12-10 by kyle-github

TheFern2, 2022-07-10:
> Emulator doesn't provide the CIP interface. You need a real PLC or SoftLogix.

Actually, this isn't entirely true. Emulator does provide a cip interface but only on localhost. With that being said I haven't tested with libplctag.

Have a look here https://github.com/dmroeder/pylogix/issues/23