libplctag/libplctag #322: Access Violation error

tneon1, 2021-12-13:
i am updating a project that was working fine with Libplctag version 1.0.4, but once I updated to the current version I started getting intermittent access violation errors . When these violation my try/catch would not catch the error. It appears that a second class I was using which was set to non async was causing the issue. Not sure why the catch was not triggering though. I have attached a copy of the class library I am using to access libplctag, I changed the checkctatus function to be async and the issue has seemed to go away. this class is called from a timed event in another project. My question is what are the best practices when using libplctag and I am not sure why the catch did not trigger.
 
Thank You
[Libplctag issue.docx](https://github.com/libplctag/libplctag/files/7705312/Libplctag.issue.docx)

kyle-github, 2021-12-15:
What wrapper are you using?  It looks like C#.  The C programming language does not have try/catch, so you definitely are not using the C DLL directly.

Did you look for similar issues on the [C# wrapper site](https://github.com/libplctag/libplctag.NET)?  Jody and Tim can probably help you more than I can.

timyhac, 2021-12-15:
In terms of why you aren't seeing an exception this seems to be a question about how async/await and Tasks work, as opposed to libplctag. Here is some reading:
* https://docs.microsoft.com/en-us/dotnet/csharp/programming-guide/concepts/async/#asynchronous-exceptions
* https://docs.microsoft.com/en-us/dotnet/standard/parallel-programming/exception-handling-task-parallel-library
* https://stackoverflow.com/a/15203282

In terms of the Access Violation Error, are you able to provide a minimal example of what is causing the error

tneon1, 2021-12-15:
Sorry,

I am using C# libplctag wrapper.
My project is a c# project as well

I have narrowed it down to a fix for bools not reading. The tag appears to sometimes be null. I added the class within my projects name space. It only happens on the bool tag write.

 public class CustomBoolPlcMapper : IPlcMapper<bool>
    {

        public int? ElementSize => 1;

        public PlcType PlcType { get; set; }
        public int[] ArrayDimensions { get; set; }

        public int? GetElementCount() => 1;

        bool IPlcMapper<bool>.Decode(Tag tag) => tag.GetUInt8(0) == 1;

        void IPlcMapper<bool>.Encode(Tag tag, bool value) => tag.SetUInt8(0, value == true ? (byte)1 : (byte)0);
    }

tneon1, 2021-12-15:
I will post on the wrapper site 
Thanks

kyle-github, 2021-12-15:
Closing this in favor of [#226](https://github.com/libplctag/libplctag.NET/issues/226) on the .Net wrapper site.

closed 2021-12-15 by kyle-github