libplctag/libplctag #324: PLC5 dhrio bridge broken in v2.4.6

ngx45, 2021-12-15:
Hi Kyle,

PLC5 dhrio bridge broken in version 2.4.6

libplctag_2.4.6_windows_x86\Release>tag_rw.exe  -t sint16  -p "protocol=ab_eip&gateway=192.168.1.103&path=1,3,B:13:2&cpu=PLC5&elem_size=2&elem_count=1&name=n7:0"
Library version 2.4.6.
Arg[0]=tag_rw.exe
Arg[1]=-t
Arg[2]=sint16
Arg[3]=-p
Arg[4]=protocol=ab_eip&gateway=192.168.1.103&path=1,3,B:13:2&cpu=PLC5&elem_size=2&elem_count=1&name=n7:0
ERROR PLCTAG_ERR_TIMEOUT: error creating tag!

Version 2.4.5 OK
libPlcTag\2.4.5\libplctag_2.4.5_windows_x86\Release>tag_rw.exe  -t sint16  -p "protocol=ab_eip&gateway=192.168.1.103&path=1,3,B:13:2&cpu=PLC5&elem_size=2&elem_count=1&name=n7:0"
Library version 2.4.5.
Arg[0]=tag_rw.exe
Arg[1]=-t
Arg[2]=sint16
Arg[3]=-p
Arg[4]=protocol=ab_eip&gateway=192.168.1.103&path=1,3,B:13:2&cpu=PLC5&elem_size=2&elem_count=1&name=n7:0
data[0]=2 (2)
Done

Regards,
ngx

kyle-github, 2021-12-15:
You have found a bug!   It looks like I manage to break this while changing the logic for how packet sizes are negotiated for Omron.   I will add this to the test suite to avoid future breakage and roll out a release as soon as possible.

- [ ] set flag to use old ForwardOpen for PCCC PLCs.
- [ ] add DH+ bridging to tests.

kyle-github, 2021-12-16:
Fixed in the latest release v2.4.7.

closed 2021-12-16 by kyle-github