libplctag/libplctag #327: 1769-L35E PROBLEM

rtorresmatheus, 2021-12-22:
I would like to know if anyone has tried to acquire tags with the list_tags example application. I'm trying to perform with PLC 1769-L35E and timeout error appears.

```
root@ASAOPS010:/home/cesar/CESAR/libplctag/build/bin_dist# ./list_tags 192.168.25.251 1,0
Starting with library version 2.4.0.
2021-12-22 10:28:38.440 thread(1) tag(0) WARN receive_forward_open_response:2164 Forward Open command failed, response code: Unsupported service. (8)
2021-12-22 10:28:38.440 thread(1) tag(0) WARN receive_forward_open_response:2167 Received CIP command unsupported error from the PLC!
2021-12-22 10:28:43.402 thread(2) tag(11) WARN plc_tag_read:1283 Read operation timed out.
Error PLCTAG_ERR_TIMEOUT trying to send CIP request!
Usage: list_tags <PLC IP> <PLC path>
Example: list_tags 10.1.2.3 1,0
2021-12-22 10:28:43.402 thread(1) tag(0) WARN process_requests:1365 Error receiving packet response PLCTAG_ERR_ABORT!
2021-12-22 10:28:43.402 thread(1) tag(0) WARN session_handler:1080 Error while processing requests PLCTAG_ERR_ABORT!
2021-12-22 10:28:43.455 thread(2) tag(11) WARN recv_forward_close_resp:2294 Unexpected EIP packet type received: 112!
2021-12-22 10:28:43.455 thread(2) tag(11) WARN perform_forward_close:1926 Forward close response not received, PLCTAG_ERR_BAD_DATA!
```

kyle-github, 2021-12-22:
Thanks for raising this issue, @rtorresmatheus, can you modify the source code in list_tags.c to turn on more debugging.  Near the top of the `main` function there will be a line like this:

```c
plc_tag_set_debug_level(PLCTAG_DEBUG_ERROR);
```

It should be around line 138 in the source code.   Modify that so that it reads:

```c
plc_tag_set_debug_level(PLCTAG_DEBUG_DETAIL);
```

Please run that and capture the output and post it to this issue.

rtorresmatheus, 2021-12-22:
Sorry, the log I got was this one, the log is too big, i had to add that file.
[log_list_tag.txt](https://github.com/libplctag/libplctag/files/7764634/log_list_tag.txt)

kyle-github, 2021-12-22:
Thanks for the log file.   That explains what is happening.   It is and it is not an error.   It is a configuration problem.   I hard coded the timeout for tag reading to be 5 seconds.   You apparently have many, many tags in that CompactLogix system.   For whatever reason, CompactLogix systems all seem to negotiate a small packet size (about 500 bytes), so when you have a lot of tags (it looks like you may have at least a thousand and possibly many more), it takes a long, long time to read all of them.   Thanks, Rockwell!

The tag listing works just fine up until the point that the 5 second timeout is hit.

If you are able to recompile the code, please change the constant `TIMEOUT_MS` at about line 46 in the list_tags.c program and recompile.  I would suggest at least 10 seconds (the timeout is in milliseconds, so 10000) and probably 20 seconds would be safer.

I will take this as a bug in the list_tags.c program to make it more like tag_rw2 where you can add a timeout on the command line.

rtorresmatheus, 2021-12-22:
Thank a lot! It worked!

kyle-github, 2021-12-22:
Great!  I will see what to do with the list_tags.c code.

I now think that maybe I should just get rid of the program and fold the functionality into tag_rw2.

kyle-github, 2021-12-23:
Can we close this?

rtorresmatheus, 2021-12-23:
Yes

closed 2021-12-23 by kyle-github