libplctag/libplctag #328: String write causes memory corruption

kyle-github, 2021-12-22:
Pulled from [#225](https://github.com/libplctag/libplctag.NET/issues/225) in libplctag.NET.

@revereb said:
> We found the problem to be in the native C dll which does not allocate a buffer large enough during a tag write and subsequently causes a buffer overflow. C# detects the corrupted memory when the garbage collector cleans up the tag handles and immediately closes the application for security reasons. Without even trying your suggestion I can guarantee that it will be replicated with logging turned on, especially since said logging appears to be only in the C# world.
>
> That being said, you are more that welcome to try it yourself. I have posted a PLC tag file (UDT_Handshaking.zip) that can be used to upload the tags on a PLC as well as a test project (PLCBoolIssue.zip) which uses the Native Import in order for you to replicate the issue. The fact that I am using the Native Import should tell you that I don't have access to the logging that you are referring to. However, you are welcome to convert the project to use the .NET version and enable the logging. At this point I am sure the issue will still occur.

radrevere, 2021-12-22:
For some reason outputting stdout in C# does not include stdout for loaded dll's so I captured the last bit in the console window after the crash occurred.  I believe it still holds the information you need.  Let me know if you need more and I can increase the buffer size of the console window.
[dbgout.txt](https://github.com/libplctag/libplctag/files/7763687/dbgout.txt)
.

kyle-github, 2021-12-22:
Thanks, but unfortunately this raises more questions than it answers.   I can see that there is one tag open with ID 11.   I can also see that for some reason the library is not connected to the PLC at the time of the snippet.   Just at the very end, a new tag is being created (ID 12, looks like the tag name is "Comms_FromNorthPallet.Build[1].LPN"?) and then ... boom.

Can you try using the command line tools that come with the C DLL?   There is a download [here](https://github.com/libplctag/libplctag/releases/).  The tag_rw2 program should be able to read the same tag (and write it) and should be able to give us a full set of logs.   We can then compare these to the ones from the C# wrapper to find out what differs.

Alternatively, if you can capture the full log set, maybe I can see what is happening.

kyle-github, 2021-12-22:
I've been going through this more and the library is definitely not connected to the PLC at the point of the log snippet.   In fact it is in a state where it is waiting to reconnect, so something made it fail the connection before the snippet.  Is that expected @radrevere?

radrevere, 2021-12-22:
I am currently testing from home over VPN and I did notice that from here it doesn't always connect to the PLC. Once it does connect and attempt the write then ... boom ... as you said.  The name of the tag is Comms_FromNorthPallet.Build[1].LPN and is part of a larger group of tags.  Instead of running tag_rw2 I wrote my own simple app that loops on the write attempt in C++ because of the unstable connection. 
again, attaching to the stdout of an external dll proves to be problematic in Windows so I again captured what I could of the output.  It ends as Visual Studio hits a hard stop with:
Unhandled exception at 0x00007FFE26BEF199 (ntdll.dll) in TestPLCCrash.exe: 0xC0000374: A heap has been corrupted (parameters: 0x00007FFE26C577F0).
[dbgout.txt](https://github.com/libplctag/libplctag/files/7764936/dbgout.txt)
here is the whole of my test app:
`#include <iostream>
#include <thread>
#include "libplctag.h"

#define TAG_PATH "protocol=ab_eip&gateway=192.168.123.151&path=1,0&cpu=LGX&name=Comms_FromNorthPallet.Build[1].LPN&debug=4"
#define DATA_TIMEOUT 5000
#define BUFFER_LEN 512

char buffer[512] = { 0 };

bool WriteTag()
{
    int32_t tag = 0;

    tag = plc_tag_create(TAG_PATH, DATA_TIMEOUT);
    if (tag < 0)
    {
        std::cout << "unable to create tag" << plc_tag_decode_error(tag);
        return false;
    }

    int rc = plc_tag_status(tag);
    if (rc != PLCTAG_STATUS_OK)
    {
        std::cout << "Bad tag status: " << plc_tag_decode_error(rc);
        plc_tag_destroy(tag);
        return false;
    }
    int strlen = plc_tag_get_string_length(tag, 0);
    if (strlen > BUFFER_LEN)
    {
        std::cout << "String it too large for input buffer: " << strlen;
        plc_tag_destroy(tag);
        return false;
    }
    rc = plc_tag_get_string(tag, 0, buffer, strlen + 1);
    if (rc != PLCTAG_STATUS_OK)
    {
        std::cout << "Bad tag status: " << plc_tag_decode_error(rc);
        plc_tag_destroy(tag);
        return false;
    }
    std::cout << "Tag Contents: " << buffer;
    memset(buffer, 0, BUFFER_LEN);
    strcpy_s(buffer, "HELLO");

    rc = plc_tag_set_string(tag, 0, buffer);
    if (rc != PLCTAG_STATUS_OK)
    {
        std::cout << "Bad tag status: " << plc_tag_decode_error(rc);
        plc_tag_destroy(tag);
        return false;
    }
    std::cout << "Success";
    return true;
}

int main()
{
    while (WriteTag() == true)
    {
        std::this_thread::sleep_for(std::chrono::milliseconds(5000));
    }
}`

kyle-github, 2021-12-24:
Thanks for creating a good minimal repro program.   If you add a call to `plc_tag_set_debug_level(PLCTAG_DEBUG_DETAIL)` at the top of `main()`, that will cause the library to output a lot of data and if you can grab that it would help.   You can redirect the output to a file if you run the program on the command line.  

```text
C:\aDir\my_program.exe > log.txt 2>&1
```

Also, change the code like this:

```c++
#include "libplctag.h"

#define TAG_PATH "protocol=ab_eip&gateway=192.168.123.151&path=1,0&cpu=LGX&name=Comms_FromNorthPallet.Build[1].LPN&debug=4"
#define DATA_TIMEOUT 5000
#define BUFFER_LEN 512

char buffer[512] = { 0 };

bool WriteTag()
{
    int32_t tag = 0;

    tag = plc_tag_create(TAG_PATH, DATA_TIMEOUT);
    if (tag < 0)
    {
        std::cout << "unable to create tag" << plc_tag_decode_error(tag);
        return false;
    }

    int rc = plc_tag_status(tag);
    if (rc != PLCTAG_STATUS_OK)
    {
        std::cout << "Bad tag status: " << plc_tag_decode_error(rc);
        plc_tag_destroy(tag);
        return false;
    }

    int strlen = plc_tag_get_string_length(tag, 0);
    if (strlen > BUFFER_LEN)
    {
        std::cout << "String it too large for input buffer: " << strlen;
        plc_tag_destroy(tag);
        return false;
    }

    /*** WARNING, this is dangerous, if strlen (which is a standard C function) is equal to BUFFER_LEN, then the following could overwrite the buffer end. */
    rc = plc_tag_get_string(tag, 0, buffer, strlen + 1);
    if (rc != PLCTAG_STATUS_OK)
    {
        std::cout << "Bad tag status: " << plc_tag_decode_error(rc);
        plc_tag_destroy(tag);
        return false;
    }
    std::cout << "Tag Contents: " << buffer;

    /* ADD THIS */
    std::cout << "Tag buffer size: " << plc_tag_get_size(tag) << std::endl;

    /* AND THIS */
/****** Comment this all out for now 
    memset(buffer, 0, BUFFER_LEN);
    strcpy_s(buffer, "HELLO");

    rc = plc_tag_set_string(tag, 0, buffer);
    if (rc != PLCTAG_STATUS_OK)
    {
        std::cout << "Bad tag status: " << plc_tag_decode_error(rc);
        plc_tag_destroy(tag);
        return false;
    }
********/

     std::cout << "Success";
     return true;

}

int main()
{
    while (WriteTag() == true)
    {
        std::this_thread::sleep_for(std::chrono::milliseconds(5000));
    }
}
```

That should just print out the string and the total length of the internal buffer for the tag.   If that number is 44, then try the original code, but with the following tag string:

```text
protocol=ab_eip&gateway=192.168.123.151&path=1,0&cpu=LGX&name=Comms_FromNorthPallet.Build[1].LPN&debug=4&str_is_byte_swapped=0&str_is_counted=1&str_is_fixed_length=1&str_is_zero_terminated=0&str_count_word_bytes=4&str_max_capacity=40&str_pad_bytes=0&str_total_length=44
```

revereb, 2021-12-24:
Thank you for catching the bug, I changed the strlen comparison the be:
`if (strlen >= BUFFER_LEN)`
which will take care of the null termination concern (I initially didn't care because I know the PLC buffer size is 12 shorts).  I also added a line to destroy the tag before exiting the function to ensure there isn't a memory leak ;)
That being said, the output you are looking for says "Tag Buffer size: 16".
I am certain this is a problem based on the PLC buffer being specified as 12 sint characters so the buffer size should be at least 26 bytes in order to account for the null termination.

FYI, strlen is currently 7 and so 16 bytes is appropriate in accounting for the sizeof(sint[7]) + 2 for the null termination.
This suggests to me that perhaps the entire sint[12] array is somehow involved in the tag write?

Updated cpp file with the write not commented:
[TestPLCCrash.zip](https://github.com/libplctag/libplctag/files/7774471/TestPLCCrash.zip)
(can't upload cpp files so I zipped it)
Let me know if you would like the entire vs2019 project (I know you would prefer a cross-platform version).

revereb, 2021-12-24:
Almost forgot:
[log.txt](https://github.com/libplctag/libplctag/files/7774577/log.txt)
After discovering the buffer size I realized I forgot to send you an output log.  (Thank you for the tip on outputting to file).  I added the `plc_tag_set_debug_level(PLCTAG_DEBUG_DETAIL)` to the top of main() before running this (version I sent doesn't have that).
I noticed this time that the write actually fails to write.  It appears to have succeeded multiple times before the crash, however if you look the PLC tag is always read as the string 'Nothing' even though on line 52 I set the write buffer to be 'HELLO'.  So not only do we have a buffer overrun but the tag is also never written.

Here is another example of multiple writes that claim to have succeeded:
[log2.txt](https://github.com/libplctag/libplctag/files/7774620/log2.txt)
You will notice that every time the read happens, the buffer contains the string 'Nothing'.  Search for the string "Tag Contents: Nothing" and you will see that it appears before every write attempt.

kyle-github, 2021-12-24:
Thanks for the feedback.  I think we are getting closer.

If the tag buffer size is 16 bytes, then that is a 4-byte count word followed by 12 bytes of string data (usually *Logix strings are _not_ null terminated, but here, who knows?).  The library is null terminating strings when it hands them back to you to make them valid C-style strings.   The underlying data may not be null terminated.

When a string is written, the entire buffer is sent at once.  This is like sending an entire UDT.  Strings are treated like UDTs inside the PLC even though the normal STRING type is considered a special system type.

Initially, the library would think that the string is composed of three parts for a standard string:
1. a 4-byte count word, LEN.
2. 82 bytes of string data, DATA.
3. 2 bytes of padding to make the whole thing a multiple of 4 bytes in length (Logix pads to DINT sizes).

The extended tag string I sent can change all of these values.  Is it 40 bytes for the character data or 12 as seems to be the case here?  If it is 12, change the sample tag string I sent to the following:

```text
protocol=ab_eip&gateway=192.168.123.151&path=1,0&cpu=LGX&name=Comms_FromNorthPallet.Build[1].LPN&debug=4&str_is_byte_swapped=0&str_is_counted=1&str_is_fixed_length=1&str_is_zero_terminated=0&str_count_word_bytes=4&str_max_capacity=12&str_pad_bytes=0&str_total_length=16
```

This defines strings to be fixed length, counted with a 4-byte count word, and have 12 bytes of character data, with no padding.

BTW, waiting 5 seconds between attempts to read/write the tag will cause the library to close the connection as idle.   Try waiting less time.   The PLC will drop the connection on its side and if it does, it is harder to restart the connection, so I made the library do it first where we can do it very cleanly.

kyle-github, 2021-12-24:
Thanks for the logs.   I see the repeated attempts (successful) to read the data from the tag.   Did you comment out the write as I suggested earlier?   It appears that you did because I do not see any attempted writes, just reads.

Also, as a performance tip, you can create the tag and hold onto its handle for as long as you like.   Days, even weeks, and the library will just make it work internally.  The way you are doing the test program sets up and tears down the connection to the PLC each time which is a very heavy operation (at least on the PLC side).

Another note: you were not kidding about the network being bad!   Wow.   At least one of the connection set ups took more than 300ms!  And I thought my home WiFi network was a good stress test!

The raw data returned looks very much like a 4-byte count word and 12 bytes of data.   Try uncommenting the write code and use the tag string I sent above.   It looks like it would work.  Here's the packet:

```text
2021-12-24 07:27:29.084 thread(5) tag(0) INFO recv_eip_response:1998  70 00 2e 00 2b a0 03 40 00 00
2021-12-24 07:27:29.084 thread(5) tag(0) INFO recv_eip_response:1998  00 00 00 00 00 00 00 00 00 00
2021-12-24 07:27:29.084 thread(5) tag(0) INFO recv_eip_response:1998  00 00 00 00 00 00 00 00 00 00
2021-12-24 07:27:29.084 thread(5) tag(0) INFO recv_eip_response:1998  02 00 a1 00 04 00 df 0c 00 00
2021-12-24 07:27:29.084 thread(5) tag(0) INFO recv_eip_response:1998  b1 00 1a 00 01 00 d2 00 00 00
2021-12-24 07:27:29.084 thread(5) tag(0) INFO recv_eip_response:1998  a0 02 97 4e 07 00 00 00 4e 6f
2021-12-24 07:27:29.084 thread(5) tag(0) INFO recv_eip_response:1998  74 68 69 6e 67 00 00 00 00 00
```

Breaking this down into pieces:

```text
2021-12-24 07:27:29.084 thread(5) tag(0) INFO recv_eip_response:1998  70 00 2e 00 2b a0 03 40 00 00
2021-12-24 07:27:29.084 thread(5) tag(0) INFO recv_eip_response:1998  00 00 00 00 00 00 00 00 00 00
2021-12-24 07:27:29.084 thread(5) tag(0) INFO recv_eip_response:1998  00 00 00 00 00 00 00 00 00 00
2021-12-24 07:27:29.084 thread(5) tag(0) INFO recv_eip_response:1998  02 00 a1 00 04 00 df 0c 00 00
2021-12-24 07:27:29.084 thread(5) tag(0) INFO recv_eip_response:1998  b1 00 1a 00 01 00 
```

All that is header for EIP encapsulation of CIP over Ethernet, plus the CIP common packet format header.  The real core of the response follows:

```text
2021-12-24 07:27:29.084 thread(5) tag(0) INFO recv_eip_response:1998  xx xx xx xx xx xx d2 00 00 00
2021-12-24 07:27:29.084 thread(5) tag(0) INFO recv_eip_response:1998  a0 02 97 4e 07 00 00 00 4e 6f
2021-12-24 07:27:29.084 thread(5) tag(0) INFO recv_eip_response:1998  74 68 69 6e 67 00 00 00 00 00
```

The part `d2 00 00 00` is the successful CIP command header.   The next part, `a0 02 97 4e` is the information about the type of the data.  In this case it means it is a UDT structure with a type hash (not an ID) of `97 4e`.   Then we get the length, `07 00 00 00`.   That is 7 in little-endian format as a 4-byte DINT.   Then we get the character data, `4e` = 'N', `6f` = 'o' etc.  The last part is just zero-ed out filler for the rest of the character bytes.

Try making sure that the write code is enabled and try that new tag attribute string.   It looks like it will work.

revereb, 2021-12-24:
Yes, the UDT is defined as  DINT LEN and SINT[12].  I changed the tag path as recommended and changed the write delay to 500ms.  I ran the test for about 2 minutes without any crashes before ending the debug session:
[log.txt](https://github.com/libplctag/libplctag/files/7774673/log.txt)

I had uncommented the write so perhaps it didn't recompile before the previous run.  The above log still shows the write thinks it succeeded yet the read value is still always 'Nothing'.  I modified the code to write "HelloX" where X is a counter and am outputting said value before calling plc_tag_set_string.  here is the updated code:

[TestPLCCrash.zip](https://github.com/libplctag/libplctag/files/7774682/TestPLCCrash.zip)

revereb, 2021-12-24:
Oh, I do know I can hold on to tags and do it often in my production code.  The C# version recreates the tags every time in order to get the garbage collector to kick off sooner.  I just kept that idea here for the same reason, a buffer overrun will become more apparent when the memory location has a higher chance of being in different locations.

kyle-github, 2021-12-24:
Sorry I missed this on the first round.  You are not calling `plc_tag_write()` anywhere.   After you set the string to "HELLO", add the following at about line 63:

```c++
    rc = plc_tag_write(tag, DATA_TIMEOUT);
    if (rc != PLCTAG_STATUS_OK)
    {
        std::cout << "Bad tag write status: " << plc_tag_decode_error(rc);
        plc_tag_destroy(tag);
        return false;
    }
```

You also do not have a read anywhere, but it still works because one of the steps of `plc_tag_create()` triggers a read so that you have data about the tag contents and the size before you do a write.

If you want to have the library do reads and writes automatically for you, [you can set that up](https://github.com/libplctag/libplctag/wiki/Tag-String-Attributes#auto-sync).  I suggest using callbacks to find out when data is changing in that case.

revereb, 2021-12-24:
oops, sorry about that.  Well, at least you know that the buffer overrun happens during the plc_tag_set_string function rather than during the actual write.  I ran it for several minutes without a crash.  Here is an updated log which shows about a minute of run time:
[log.txt](https://github.com/libplctag/libplctag/files/7774779/log.txt)

revereb, 2021-12-24:
And here's a version with the debug _detail turned on:
[log.txt](https://github.com/libplctag/libplctag/files/7774794/log.txt)

kyle-github, 2021-12-24:
Looks like it is working!

I will use your original test in the next few days to see if I can cause an overwrite in the library.   You may have found a spot where I am not checking the size correctly.   It should not ever cause a crash even if you get the tag attributes wrong.

kyle-github, 2021-12-28:
I am having a hard time triggering this.  The code appears to be checking all the right things.  I will need to duplicate your exact situation to see if I can cause it to happen.

revereb, 2021-12-28:
Do you have a PLC to test with?  If so I have attached an .ACD file for importing the tags to a PLC.
[UDT_Handshaking.zip](https://github.com/libplctag/libplctag/files/7784754/UDT_Handshaking.zip)
It is specifically the UDT strings that are created here that cause the problem.  In our real world application there are a set of 9 pallet locations that when the pallet is full of product I get a signal to tell the PLC to release the pallet with a specific LPN (barcode).  I proceed to write the tracking barcode to the PLC and then set its associated Release tag.  They are setup as an array of tag pairs [1-9].  For example, position one starts at offset 1:
Comms_FromNorthPallet.Build[1].LPN
Comms_FromNorthPallet.Build[1].Release
(North and South because there are two pallet build locations in the warehouse)
The buffer overrun occurs when the LPN is being written.  LPN is a UDT called palletString defined as a DINT LEN; and SINT[12] DATA;  Let me know if you need anything else.

kyle-github, 2022-01-07:
I tested this against a real PLC but with a standard 82-byte string.   I tried to write 83 bytes and it truncated it.  I am going to try to make a test program to force longer strings.

revereb, 2022-01-07:
A standard 82-byte string will not crash the dll.  You need to create a UDT string to see the problem. If you can upload the tags in the UDT_Handshaking.zip it would be your ideal test.  Once the tags are loaded try writing to:
Comms_FromNorthPallet.Build[1].LPN

kyle-github, 2022-01-07:
I found a PLC that has a UDT "string" type with a LEN element that is a DINT and a DATA element that is SINT[16] for a total of twenty bytes.   I tried to read and write it.   Leaving all the defaults on, I can get the string data and print it out.   When I try to write a long string to it, "01234567890123456789...", I just get an error back.   Under valgrind (memory check) there are no writes past the buffer end.

Here is the definition of the UDT:

```text
 UDT WO_PN (ID 223, 20 bytes, struct handle dba1):
    Field 0: LEN, offset 0, type (00c4) DINT: Signed 32-bit integer value.
    Field 1: DATA, offset 4, array [16] of type (20c2) SINT: Signed 8-bit integer value.
```

Here is the tag string I am using:

```text
protocol=ab-eip&gateway=10.206.1.40&path=1,0&plc=lgx&name=CB_Txt[0,0]
```

This is using version 2.4.9 that was just released.

So there must be something else that I need to do to trigger this.

I am testing under Linux and macOS.  I will try to set up the same test under Windows.   Do you have a good tool to check memory safety on Windows?   I've tried Dr. Memory, but it has been a while since I used it.  Valgrind (on Linux) is really amazing.

radrevere, 2022-01-07:
I use VisualStudio 2019 Community edition.  I had some time this morning to pull down your source and debug it in VS and found what I believe is your problem.  Look at line 3362 of lib.c:
`int string_capacity = (tag->byte_order->str_max_capacity ? (int)(tag->byte_order->str_max_capacity) : get_string_length_unsafe(tag, string_start_offset));`
the problem with this is that is assumes that the default max capacity at 82 is the size of the string.  I would guess that this is a PLC default field that it fills when you are trying to read a string.  As you know, the problem is that it is a UDT string and so we cannot make that assumption.  This UDT string in particular is a DINT and SINT[12] combo making it 16 bytes total.  We can verify that in the VS debugger:
[image]
You will note that the tag struct has a size that it's reporting which is actually the correct size of 16.  So on line 3381 when try to zero-pad the buffer you get a massive buffer overrun:
 `/* zero pad the rest. */
                for(int i = string_length; i < string_capacity; i++) {
                    size_t char_index = (((size_t)(unsigned int)i) ^ (tag->byte_order->str_is_byte_swapped)) /* byte swap if necessary */
                                    + (size_t)(unsigned int)string_start_offset
                                    + (size_t)(unsigned int)(tag->byte_order->str_count_word_bytes);
                    tag->data[char_index] = (uint8_t)0;
                }`
Rather than always presuming the max size, I would recommend checking against the actual size of the tag before padding or writing anything to that buffer (tag->data).

I will leave it to you and your team to code the fix for this.

radrevere, 2022-01-07:
Oh, as a side note my PLC guys tell me that the length of a standard string can be changed so relying on the default max would be problematic if someone decided to change that length anyways (which is very rare).
And if you decide to use tag->size don't forget to subtract the 4 bytes for the LEN ;)

kyle-github, 2022-01-07:
I tested the scenario: short actual UDT, default lengths and a long string value.  What I have not checked was a short string value but long string capacity (the value I checked was as long as the string capacity).  I will make sure that the check on 3367 actually catches that.   I think you may have found the problem but I need to do more testing and make sure, if this is indeed the bug, that I get this into the regression tests!

I need to split out this logic into smaller steps to make sure it is very clean.

closed 2022-01-09 by kyle-github

kyle-github, 2022-01-09:
I think I have fixed this.  My tests pass at least.   Thanks, @radrevere for all the help and testing and thought you put into this!  It looks like the core problem was triggered by the following:
1. have a non-standard string size.
2. set a value that fits in the non-standard size but leaves some room for zero-ing out the remainder.

Then the loop that zeros out the rest of the tag buffer will overwrite memory.   I put checks on all three parts of writing to the tag buffer data:
1. a check on each byte write of the string data into the tag buffer.
2. a check on each byte write when zeroing out the remainder of the buffer.
3. a check on the final end index when writing out the string count/size value.

I am making a release for this right now.  Please test it as soon as you see the release dropped.   Thanks again for all the help!

radrevere, 2022-01-10:
@kyle-github ,
I haven't tested it but I don't see any code that will fix the issue.  Has it been checked in?  line 3446 is as such:
`for(int i = string_length; i < string_capacity; i++) `
and before this string_capacity will be set to the full 82 characters on line 3362 by assigning it str_max_capacity and never checked against tag->size (which in my case is 16) or made smaller by any other checks.  The buffer overrun will still occur with the code that is in place.  I recommend putting a check in place right after line 3362:

`// put check in for UDT strings where string_capacity is set to default max value and actual UDT size is less than string_capacity
if(string_capacity > (tag->size-4))
{
    pdebug(DEBUG_WARN, "Bad configuration? String capacity/size is larger than the tag buffer! Adjusting string capacity.");
    string_capacity = tag_size - 4;
}`

kyle-github, 2022-01-10:
That does not correspond to the code I have.   Are you certain you got the latest code?

Here is the full routine starting at line 3314:

```c
LIB_EXPORT int plc_tag_set_string(int32_t tag_id, int string_start_offset, const char *string_val)
{
    int rc = PLCTAG_STATUS_OK;
    plc_tag_p tag = lookup_tag(tag_id);
    int string_length = 0;

    pdebug(DEBUG_DETAIL, "Starting with string %s.", string_val);

    if(!tag) {
        pdebug(DEBUG_WARN,"Tag not found.");
        return PLCTAG_ERR_NOT_FOUND;
    }

    /* is there data? */
    if(!tag->data) {
        pdebug(DEBUG_WARN,"Tag has no data!");
        tag->status = PLCTAG_ERR_NO_DATA;
        rc_dec(tag);
        return PLCTAG_ERR_NO_DATA;
    }

    /* are strings defined for this tag? */
    if(!tag->byte_order || !tag->byte_order->str_is_defined) {
        pdebug(DEBUG_WARN,"Tag has no definitions for strings!");
        tag->status = PLCTAG_ERR_UNSUPPORTED;
        rc_dec(tag);
        return PLCTAG_ERR_UNSUPPORTED;
    }

    if(!string_val) {
        pdebug(DEBUG_WARN, "New string value pointer is null!");
        tag->status = PLCTAG_ERR_NULL_PTR;
        rc_dec(tag);
        return PLCTAG_ERR_NULL_PTR;
    }

    /* note that passing a zero-length string is valid. */

    if(tag->is_bit) {
        pdebug(DEBUG_WARN, "Setting a string value on a bit tag is not supported!");
        tag->status = PLCTAG_ERR_UNSUPPORTED;
        rc_dec(tag);
        return PLCTAG_ERR_UNSUPPORTED;
    }

    string_length = str_length(string_val);

    critical_block(tag->api_mutex) {
        int string_capacity = (tag->byte_order->str_max_capacity ? (int)(tag->byte_order->str_max_capacity) : get_string_length_unsafe(tag, string_start_offset));
        int string_last_offset = string_start_offset + (int)(tag->byte_order->str_count_word_bytes) + string_capacity + (tag->byte_order->str_is_zero_terminated ? 1 : 0);

        /* initial checks - these must be done in the critical block otherwise someone could change the tag buffer size etc. underneath us. */
pdebug(DEBUG_WARN, "string_capacity=%d, string_last_offset=%d, tag_size=%d.", string_capacity, string_last_offset, (int)tag->size);

        /* will it fit at all? */
        if(string_capacity < string_length) {
            pdebug(DEBUG_WARN, "Passed string value is longer than allowed string capacity!");
            rc = PLCTAG_ERR_TOO_LARGE;
            tag->status = (int8_t)rc;
            break;
        }

        /* do we have a mismatch between what is in the tag buffer and the string or configuration? */
        if(string_last_offset > tag->size) {
            pdebug(DEBUG_WARN, "Bad configuration? String capacity/size is larger than the tag buffer!");
[... 85 lines cut ...]
            tag->status = (int8_t)rc;
            break;
        }

        /* if this is an auto-write tag, set the dirty flag to eventually trigger a write */
        if(rc == PLCTAG_STATUS_OK && tag->auto_sync_write_ms > 0) {
            tag->tag_is_dirty = 1;
        }

        /* set the return and tag status. */
        rc = PLCTAG_STATUS_OK;
        tag->status = (int8_t)rc;
    }

    rc_dec(tag);

    pdebug(DEBUG_DETAIL, "Done with status %s.", plc_tag_destroy(rc));

    return rc;
}
```

Note that there are guards on every location that updates the tag data buffer.  I just double checked to make sure that updating Git from GitHub gave me the about code.

Please let me know if this is somehow not in the release!

radrevere, 2022-01-10:
yes, i looked at the latest code.  Do a search in what you pasted for stirng_capacity (search on this page and it will highlight in yellow) and you will see what I am talking about.  It is never adjusted or checked against anything that represents the actual size of the tag buffer. So when the remaining buffer is zeroed out it will still cause the buffer overrun.

kyle-github, 2022-01-10:
I think you may be missing either some code or misunderstanding what it is doing.   The release branch code is [here](https://github.com/libplctag/libplctag/blob/release/src/lib/lib.c).

Right after the start of the mutex protected section, I calculate two values:

```c

        int string_capacity = (tag->byte_order->str_max_capacity ? (int)(tag->byte_order->str_max_capacity) : get_string_length_unsafe(tag, string_start_offset));
        int string_last_offset = string_start_offset + (int)(tag->byte_order->str_count_word_bytes) + string_capacity + (tag->byte_order->str_is_zero_terminated ? 1 : 0);
```

The first one gets the total capacity of the string.  In this case that is the default 82 bytes.   The next one calculates the position one past where the string would end including the count word, the total capacity and the terminating character.  This is relative to the starting offset.

Those are then used to check for overruns:

```c
        if(string_capacity < string_length) {
            pdebug(DEBUG_WARN, "Passed string value is longer than allowed string capacity!");
            rc = PLCTAG_ERR_TOO_LARGE;
            tag->status = (int8_t)rc;
            break;
        }

        /* do we have a mismatch between what is in the tag buffer and the string or configuration? */
        if(string_last_offset > tag->size) {
            pdebug(DEBUG_WARN, "Bad configuration? String capacity/size is larger than the tag buffer!");
            rc = PLCTAG_ERR_BAD_CONFIG;
            tag->status = (int8_t)rc;
            break;
        }

```

The first check either passes or punts out of the mutex-protected section completely.  If it passes we know that the string length is less than or equal to the string capacity.

In the next check, we determine that the offset one past the end of the required buffer space for the string (counting capacity, not the actual string size) is less than or equal to the size of the tag.   If it is not, then we punt out as well.

After these two checks we know that there is sufficient space for the maximum-sized string in the tag data buffer and that the string passed in is the same size or shorter than the string capacity.

Then we have the loop to copy the string data:

```c
        for(int i = 0; i < string_length; i++) {
            size_t char_index = (((size_t)(unsigned int)i) ^ (tag->byte_order->str_is_byte_swapped)) /* byte swap if necessary */
                            + (size_t)(unsigned int)string_start_offset
                            + (size_t)(unsigned int)(tag->byte_order->str_count_word_bytes);

            if(char_index < (size_t)(uint32_t)tag->size) {
                tag->data[char_index] = (uint8_t)string_val[i];
            } else {
                pdebug(DEBUG_WARN, "Out of bounds index generated during string copy!");
                rc = PLCTAG_ERR_OUT_OF_BOUNDS;

                /* note: only breaks out of the for loop, we need another break. */
                break;
            }
        }
```

The index range is safe because it is only as far as the string length which we already proved was shorter than or equal to the string capacity.   Just in case that is not enough, on each possible write, we also check that the string index is less than the tag buffer data size.

The same sort of check is done for zeroing out the unused data:

```c
        for(int i = string_length; i < string_capacity; i++) {
            size_t char_index = (((size_t)(unsigned int)i) ^ (tag->byte_order->str_is_byte_swapped)) /* byte swap if necessary */
                            + (size_t)(unsigned int)string_start_offset
                            + (size_t)(unsigned int)(tag->byte_order->str_count_word_bytes);

            if(char_index < (size_t)(uint32_t)tag->size) {
                tag->data[char_index] = (uint8_t)0;
            } else {
                pdebug(DEBUG_WARN, "Out of bounds index generated during string zero padding!");
                rc = PLCTAG_ERR_OUT_OF_BOUNDS;

                /* note: only breaks out of the for loop, we need another break. */
                break;
            }
        }
```

The index range of the for loop is safe because we proved that an index with a maximum offset of the starting offset, plus the count word, plus the full string capacity was within the tag's data buffer.  Again, just to be safe, we also check each individual index to make sure it is within the tag data buffer.

The string capacity is not modified in these checks.   It is part of the value that is used to determine the total length of the required space for the string argument and that, plus the offset, is used to determine whether there is space within the tag's data buffer.

Thanks for making me walk through this, again.   If there are PLCs that defined fixed-width strings with zero termination and the zero byte is not counted in the fixed width, then there may be an overwrite of one byte.  I am not sure that there are any PLCs like that (most would include the termination byte in the width), but it is worth noting.

radrevere, 2022-01-11:
I'm glad you went through this, however I understand your code just fine. 
The problem, I think, is that you misunderstand the bug. I think because you haven't loaded the tags from the ACD file in UDT_Handshaking.zip you don't see the values that are set by the PLC.  Tracking the values based on these UDT strings I get the following:
1. string_capacity is set to 82 because that is the value of str_max_capacity.  (the first value you calculate)
  - I believe this is set by default by the PLC even though the UDT string is much smaller
  - This false value given by the PLC is the problem
2. tag->size = 16.  this includes the DINT LEN which is 4 so the actual buffer size for the string is 12.
3. all your checks pass in this scenario because the string being written fits within the 12 bytes. Meaning that char_index (or string_length) is less than tag->size.  Your checks only compare the string being written against the tag->size and so the true problem is missed.  (keep in mind that this is also incorrect as well because tag->size includes the LEN which is 4 bytes)

Now, by the time you get to the padding loop you have the following values:
string_capacity = 82
tag->data = 12 bytes  (tag->size-4)
This means that you are over running the buffer by 70 bytes while padding the buffer with 0's.

radrevere, 2022-01-11:
Correction, 4 byte overrun because of this check in the padding loop:
if(char_index < (size_t)(uint32_t)tag->size)

Please remember that the tag is pointing to a UDT string structure which includes the DINT LEN and the SINT DATA buffer.  Your check should be changed to:
if(char_index < (size_t)(uint32_t)(tag->size-4))

radrevere, 2022-01-11:
I apologize for the massive edit in my previous post.  I am traveling this week, tired and I rushed through the code review not noticing the check in the padding loop.  
That being said, I would recommend establishing the size of the buffer at the top of the function. Doing this would make it more maintainable in the future.  You would have only one place to add your buffer size checks (such as the NULL termination situation you mentioned) and then you can use that value with confidence in the rest of the function.  Along with this I would also recommend using memset instead of a loop to pad the buffer once you have established the buffer size (zero padding doesn't care about byte order but may care about str_count_word_bytes).  memset is proven faster and more efficient than a loop:
```c
int string_capacity = (tag->byte_order->str_max_capacity ? (int)(tag->byte_order->str_max_capacity) : get_string_length_unsafe(tag, string_start_offset));
if(string_capacity > (tag->size-4))
{
  string_capacity = tag->size-4;
}
/* will it fit at all? */
/// don't forget to include the start offset when evaluating to see if the input string fits, makes for one check instead of two ///
if(string_capacity < (string_length+string_start_offset)) {
    pdebug(DEBUG_WARN, "Passed string value is longer than allowed string capacity!");
    rc = PLCTAG_ERR_TOO_LARGE;
    tag->status = (int8_t)rc;
    break;
}
memset(tag->data, string_start_offset, (string_capacity-string_start_offset));
```
Once memset is called you can then write the string in the buffer and have confidence that it is null terminated and padded properly, eliminating the need for your padding loop altogether.  Granted the above code isn't quite complete but I think you get the idea.  Feel free to take my advice or not, I am just making suggestions here.

kyle-github, 2022-01-12:
Ah, I think I see what you are thinking.  However, that is not going to happen in this case.   The first value calculated, `string_capacity`, is going to be the total string capacity regardless of the actual string length passed.   The trinary operation will return the fixed capacity if there is one or the string length if there is not.   That `string_capacity` is used in the second value to calculate the offset of the last byte of the string in the buffer (including the count word), plus one (it points just past the last byte).

In turn that last index value is compared in the second if check to the actual size of the string tag data buffer.   If it fits, we continue but if it does not, then we drop out with an error.  In my case the tag data buffer is 20 bytes but the last index calculated is well past that, so the check fails.

After these checks, we know that all indexes up to the maximum capacity of the string are safe.

However, I see that I missed the padding in the calculation.  So my if check could theoretically succeed if the tag data buffer was 86 bytes, but it should be 88 bytes.   It still will not overwrite in that case because the padding is never written and there is still the byte-by-byte check in each copy/pad loop.  That total length calculation needs to be refactored out.  There is already an API function for it and it should call an internal shared function to get the value.

As to why I use the byte-by-byte copy, there are a few reasons:

- bytes are not always placed in order due to the word-level byte swapping that is done on some PLCs (Micrologix).  So I would always need some amount of checking for the first/last byte.
- it allows me to have the check at each byte to make sure that everything is safe at all times.
- even if the compiler does not turn the loop into fast code, the overhead of this code compared to the latency of sending a request to the PLC and getting a response is definitely lost in the noise.

Now, one thought might be to use memset (or mem_set) before I do any copying.  I still think the lower safety on that makes it not worth it at this time.

All this code needs more refactoring.  Strings are really ugly and when you add in the ability of people to make their own string types and then not tell the library about the difference, it makes it worse.

radrevere, 2022-01-12:
OK, so help me understand something.  tag->size is the total size of everything in the tag.  In my case, the tag contains the DINT LEN and the SINT[12] making it a total of 16 bytes.  However, the total bytes available for the string is only 12.  
Is tag->data 16 bytes or 12 bytes in length?
If it is 16 bytes then won't it overwrite the LEN portion?
If it's only 12 bytes then your padding code would overrun the buffer by 4 bytes.
What am I missing here?

kyle-github, 2022-01-12:
In your case we have:

- tag size is 20 bytes.  That includes everything.  4 bytes for the count word and 16 bytes for the data.  But the library does not know anything about this.  It just gets the whole chunk of data from the PLC as is.
- the tag string capacity is still the default, which is 82, since there is no override of the defaults.
- the default has a 4-byte count word defined.
- the default has a 2-byte end padding defined.
- the last index calculation takes into account the count word, so comes up with a total last index of `offset+86` (this is where the padding count is missing).

Assume that the offset is 0, so our last index value is 86.   The second `if` check catches that because the total tag data buffer is 20.  

Now, assuming that we pass that if check then we know that the last index value includes both the string count word and the string data capacity.   So there will be room for both.

Both the copying loop and the zeroing loop calculate an index that includes the passed in offset and the count word size.   So the data being copied or zeroed starts after the count word.   Since there is a check on each byte index before anything is written against the tag data buffer size, we cannot go out of bounds (at least above).

The code order is a bit confusing since I copy the string data and then set the count word and then do the zeroing.  If I did it in index order, it would be count word first, then the string data, then the zeroing.

radrevere, 2022-01-12:
Ah, so the simple answer is the buffer size is the full tag size and I see now how you are accounting for the LEN size.   I do a lot of localization and so in my mind str_count_word_bytes was signifying possible multi-byte characters rather than the size of the LEN field.  Thank you for the explanation and your patience with me.

As I mentioned, I am out of town on business so I will test this as soon as I can (likely the beginning of next week) and let you know how it goes.

radrevere, 2022-01-17:
@kyle-github OK, I was able to test and there is no crash but it gets to line 3377 and exits with the "Bad configuration" error and, of course, does not write the string.  
With our focus on the code following this that checks tag->size in for loops I didn't think to consider that it wouldn't make it to the loops.  Exiting on that check makes the check in the loops unnecessary as the code will never reach that point.

That being said, I would like to request that the check on line 3377 log the warning and adjust the appropriate variables but still allow the string to be written into the buffer, which I thought was what the fix was doing.

The reason I would like this is because when a developer is working with a PLC, they are aware of the size limitations and typically will not write anything larger than the defined buffer size.  I would be so bold as to suggest that a string that is too long for the buffer simply be truncated and logged (using pdebug) rather than outright rejected as well.  It is far easier to see and troubleshoot a truncation than it is to interpret PLCTAG_ERR_BAD_CONFIG as a buffer size mismatch because there are many other bad configurations that could be causing a problem.

radrevere, 2022-01-17:
I took the time to rework the critical block section of the code.  I believe I have accounted for all of the variables that you did before as well as fix the null termination bug and allow longer strings to be truncated if they don't fit into the buffer rather than fail outright.
As I went through this a couple of questions came up.  
1. What is the purpose of string_start_offset (passed into the function)?  My understanding it that it is for being able to edit an existing string. 
2.  Why are you using string_start_offset when placing the length bytes?  This doesn't makes sense with how I have seen strings defined in a PLC.  Shouldn't it always write the length bytes at the start of the buffer?  If it doesn't then the tag doesn't fit the definition of a PLC string that I know.  Is this a bug or expected behavior that I am not aware of?

Here is the rework suggestion that mentioned above.  Yes it uses memset, not for efficiency but more for clarity and consistency when working with a buffer.  I have comments above the changes I made explaining the reason.  You will note that I have comments in the section writing the length which correspons to my above questions:
```c
critical_block(tag->api_mutex) {
        size_t string_capacity = (tag->byte_order->str_max_capacity ? (int)(tag->byte_order->str_max_capacity) : get_string_length_unsafe(tag, string_start_offset));
        if (string_capacity > ((size_t)tag->size - tag->byte_order->str_count_word_bytes))
        {
            pdebug(DEBUG_WARN, "Configuration mismatch: String capacity:%s tag size:%d ; using tag size configuration.",string_capacity,tag->size);
            string_capacity = ((size_t)tag->size - tag->byte_order->str_count_word_bytes);
        }
        size_t start_offset = (size_t)tag->byte_order->str_count_word_bytes + string_start_offset;
        // 0 pad the buffer from starting offset
        memset(tag->data + start_offset, 0, (size_t)tag->size - start_offset);

        // let's ensure we don't overflow with a null termination requirement
        // this presumes that the documentation is correct that a 1 means null termination required
        // so we subtract one byte from the total capacity after we memset the end of the buffer
        string_capacity -= tag->byte_order->str_is_zero_terminated;

        // string_last_offset is not necessary because it is the same as tag->data + tag->size
        // and we have ensured that string_capacity is accurate.
        //int string_last_offset = string_start_offset + (int)(tag->byte_order->str_count_word_bytes) + string_capacity + (tag->byte_order->str_is_zero_terminated ? 1 : 0);

        /* initial checks - these must be done in the critical block otherwise someone could change the tag buffer size etc. underneath us. */
        //pdebug(DEBUG_WARN, "string_capacity=%d, string_last_offset=%d, tag_size=%d.", string_capacity, string_last_offset, (int)tag->size);
        pdebug(DEBUG_WARN, "string_capacity=%d, tag_size=%d.", string_capacity, (int)tag->size);

        // this check is now to warn that the string will be truncated
        /* will it fit at all? */
        if (string_capacity < string_length) {
            pdebug(DEBUG_WARN, "Passed string value is longer than allowed string capacity, string will be truncated!");
            //rc = PLCTAG_ERR_TOO_LARGE;
            //tag->status = (int8_t)rc;
            //break;
        }

        // this check is now handled above
        /* do we have a mismatch between what is in the tag buffer and the string or configuration? */
        /*if (string_last_offset > tag->size) {
            pdebug(DEBUG_WARN, "Bad configuration? String capacity/size is larger than the tag buffer!");
            rc = PLCTAG_ERR_BAD_CONFIG;
            tag->status = (int8_t)rc;
            break;
        }*/

        /* copy the string data into the tag. */
        rc = PLCTAG_STATUS_OK;
// max index taking into account null termination
        int max_idx = tag->size - tag->byte_order->str_is_zero_terminated;
        for (int i = 0; i < string_length; i++) {
            size_t char_index = (((size_t)(unsigned int)i) ^ (tag->byte_order->str_is_byte_swapped)) /* byte swap if necessary */
                + start_offset;// use pre-calculated offset
                //+ (size_t)(unsigned int)string_start_offset
                //+ (size_t)(unsigned int)(tag->byte_order->str_count_word_bytes);

            // let's ensure we aren't overrunning the buffer using our pre-calculated max
// this will fix the null termination overrun by 1 byte that you mentioned before
            if (char_index < max_idx) {
                tag->data[char_index] = (uint8_t)string_val[i];
            }
            else {
                pdebug(DEBUG_WARN, "Out of bounds index generated during string copy, string will be truncated!");
                rc = PLCTAG_ERR_OUT_OF_BOUNDS;

                /* note: only breaks out of the for loop */
                break;
[... 60 lines cut ...]
            }
            else {
                pdebug(DEBUG_WARN, "Out of bounds index generated during string zero padding!");
                rc = PLCTAG_ERR_OUT_OF_BOUNDS;

                /// note: only breaks out of the for loop, we need another break. 
                break;
            }
        }*/

        /* break out of the critical block if bad status. */
        if (rc != PLCTAG_STATUS_OK) {
            tag->status = (int8_t)rc;
            break;
        }

        /* if this is an auto-write tag, set the dirty flag to eventually trigger a write */
        if (rc == PLCTAG_STATUS_OK && tag->auto_sync_write_ms > 0) {
            tag->tag_is_dirty = 1;
        }

        /* set the return and tag status. */
        rc = PLCTAG_STATUS_OK;
        tag->status = (int8_t)rc;
    }```

kyle-github, 2022-01-24:
Oops, sorry I missed this.

Thanks for the approach here.   There are a few things that need to be noted:

- The offset is for starting the string at any arbitrary position in the tag's data buffer.   This is to handle the cases of arrays where all elements except the first start at some offset, and why string has `plc_tag_string_get_string_total_length()`, and in cases where the tag that is read is an entire UDT and the string part is just one of the fields.  All of the data accessor function have the offset and use it in the same way.  
- I would rather not truncate strings to fit.   Whether or not to truncate is a decision that the application can make, but the library should not.  For instance, what if the string is some sort of ID, for instance a barcode that you are sending to a printer.  Truncating it would be very problematic.

I will be returning to strings in a while when I get the latest couple of problems out of the way to see if there are ways I can check that the definition of a string that the library knows and the definition in the PLC can be checked against each other without doing a full tag listing.  I might be able to do this and if I can, it might be worth doing so that I can more easily return some sort of error like PLCTAG_ERR_BAD_CONFIG earlier rather than later.

I added the string handling functions a lot later than almost everything else because they rely on a lot of information from the user rather than Just Working(tm) most of the time.  If I ever redo the API, I will be rethinking how to do strings again.  This is the best approach I have found so far, but it is still not great.

radrevere, 2022-01-24:
Thank you for the response.  I don't have a problem with failing if the passed in string is too long but perhaps make the error as a new error of "string too long" so that it isn't a nebulous 'bad config' message that is tougher to track down.

The main thing I would like to see is the string getting written even though the buffer from the PLC is smaller than the default max buffer size.  Your documentation states that the str_max_capacity member is filled with a default value. ( [Tag-String-Attributes](https://github.com/libplctag/libplctag/wiki/Tag-String-Attributes) ) The PLC will not know to change this on a UDT because it thinks a standard string is being accessed based on the API being used.  This will not change from the PLC side so the library needs to handle it and still allow the string to be written even when there is a mismatch between tag->size and str_max_capacity.  The only exception being if the string to be written is too long.

The way you have it written it will fail to write **ALL** UDT strings that have a buffer smaller than str_max_capacity.  This is the new bug that needs to be fixed please.

kyle-github, 2022-01-24:
A user created UDT is not a string.   It may look like a string, but it is not a STRING type.

I like the idea of returning PLCTAG_ERR_TOO_LONG instead of a config error.

The default value comes from the library based on the type of the PLC.  The PLC returns essentially zero metadata about strings.   Especially in the case where a tag is an entire UDT instance and the string part is one of the fields.   In that case we do not get any indication that the field is a string.   This has nothing to do with the API used to access the tag.  The CIP protocol does not use the type when reading.  The "type" information returned when reading a tag is not the actual type of the value.   It is sometimes related, but not consistently (it is often a hash of the UDT fields and thus not unique).

It is up to the user to correctly configure the tag attribute string with the right values.  That is not great, but there is no way around it.  Even with listing out tags and looking for "string-like" UDTs we cannot solve this in the library.   

As to allowing strings to fit, I considered that, but in my experience this is a bad compromise.  It seems to work but what it really does is hide the fact that the user misconfigured the tag string attributes.   At some point in the future, the user will attempt to do something that won't work (i.e. the application developer will forget that the string is not a STRING and try to use a longer value) and only then will the error come up.  I have been the "lucky" developer who found that kind of problem long after the original developer left the company and I do not want anyone else to have to live with that pain at 2am :-/

Thanks for the thought and effort on this.   You have helped me think of a use case where it might be better for the library to stop defaulting string configuration entirely.   Here's the scenario:

- We have a UDT with a string-like UDT field followed by some other field.  The string-like field is shorter than a standard STRING type.  
- The application uses the string functions to write to the string.   Due to the other field following the string-like field, the total space left in the tag data buffer starting at the string-like field is at least as large as the default string type needs (88 bytes).
- The library will happily take the string data passed to it and overwrite the field following the string-like field.

A similar problem occurs when the count word is different from DINT.  If an INT is used in the string-like UDT, the first four bytes of the UDT will be overwritten with the count.   Since AB is little endian (mostly) this would mean that the count would look right but the string data would start off with two zero bytes and the string's character data would go past the end of the actual section of the buffer reserved for the string (if the total was a multiple of 4 bytes).

This is not solvable with the current library by using defaults at all.  In these cases, the defaults are dangerous.

This is definitely a real problem with the existing use of defaults.   However, there is no way for the library to get the real configuration values from the PLC without significant help from the user.

The right answer here is to drop all the defaults.  However, that is a large breaking change for users.  I am going to need to think about this and poll the wrapper owners.

radrevere, 2022-01-24:
You make some very good points.  I have only worked with a limited number of PLC's to this point so your experience becomes invaluable to this project.  Thank you for considering my suggestions and I will talk to my PLC experts here about UDT tag configuration.

kyle-github, 2022-01-24:
I opened issue #344 to track the problematic use of default values.  

Thanks again for continuing to push on this issue.  Without this discussion I would not have realized that this big flaw existed!

commit be85eaf2 (2022-01-09): Fix #328 (#336)

* Checkin for testing with Valgrind.

* Update string test to catch errors.
commit 12cfebc6 (2022-01-09): Merge prerelease with fix #328. (#337)

* Adding string test.

* Fix #328 (#336)

* Checkin for testing with Valgrind.

* Update string test to catch errors.

* Add overwrite check for string count word.