libplctag/libplctag #340: Compiling example file; undefined reference 'funct'

DAZ-NZ-Eng, 2022-01-18:
Hi, 

I've tried to compile the example file plc5.c and get the errors below. Has anyone come across this? It appears to be an issue with the plc_tag_.... reference.

~/libplctag/src/examples$ gcc -o plc5 plc5.c
/usr/bin/ld: /tmp/cc3cam6c.o: in function `main':
plc5.c:(.text+0x2d): undefined reference to `plc_tag_check_lib_version'
/usr/bin/ld: plc5.c:(.text+0x7c): undefined reference to `plc_tag_get_int_attribute'
/usr/bin/ld: plc5.c:(.text+0x95): undefined reference to `plc_tag_get_int_attribute'
/usr/bin/ld: plc5.c:(.text+0xad): undefined reference to `plc_tag_get_int_attribute'
/usr/bin/ld: plc5.c:(.text+0xd9): undefined reference to `plc_tag_set_debug_level'
/usr/bin/ld: plc5.c:(.text+0xea): undefined reference to `plc_tag_create'
/usr/bin/ld: plc5.c:(.text+0xfd): undefined reference to `plc_tag_decode_error'
/usr/bin/ld: plc5.c:(.text+0x12f): undefined reference to `plc_tag_status'
/usr/bin/ld: plc5.c:(.text+0x13d): undefined reference to `plc_tag_status'
/usr/bin/ld: plc5.c:(.text+0x144): undefined reference to `plc_tag_decode_error'
/usr/bin/ld: plc5.c:(.text+0x16c): undefined reference to `plc_tag_destroy'
/usr/bin/ld: plc5.c:(.text+0x185): undefined reference to `plc_tag_read'
/usr/bin/ld: plc5.c:(.text+0x198): undefined reference to `plc_tag_decode_error'
/usr/bin/ld: plc5.c:(.text+0x1c3): undefined reference to `plc_tag_destroy'
/usr/bin/ld: plc5.c:(.text+0x1ec): undefined reference to `plc_tag_get_float32'
/usr/bin/ld: plc5.c:(.text+0x237): undefined reference to `plc_tag_get_float32'
/usr/bin/ld: plc5.c:(.text+0x294): undefined reference to `plc_tag_set_float32'
/usr/bin/ld: plc5.c:(.text+0x2ad): undefined reference to `plc_tag_write'
/usr/bin/ld: plc5.c:(.text+0x2c0): undefined reference to `plc_tag_decode_error'
/usr/bin/ld: plc5.c:(.text+0x2eb): undefined reference to `plc_tag_destroy'
/usr/bin/ld: plc5.c:(.text+0x304): undefined reference to `plc_tag_read'
/usr/bin/ld: plc5.c:(.text+0x317): undefined reference to `plc_tag_decode_error'
/usr/bin/ld: plc5.c:(.text+0x342): undefined reference to `plc_tag_destroy'
/usr/bin/ld: plc5.c:(.text+0x368): undefined reference to `plc_tag_get_float32'
/usr/bin/ld: plc5.c:(.text+0x39e): undefined reference to `plc_tag_destroy'
collect2: error: ld returned 1 exit status

kyle-github, 2022-01-19:
C's compilation model requires that you need to compile each compilation unit (C file) independently then link them all together to make an executable.   In this case you need to tell the compiler that you are going to use the pthread and libplctag libraries.  The standard C library is automatically included.

```text
gcc -o plc5 plc.c -pthread -lplctag
```

This assumes that you have both libraries in your library search path.

kyle-github, 2022-01-27:
Please reopen if you have additional questions.

closed 2022-01-27 by kyle-github