libplctag/libplctag #341: Segmentation fault in tag_tickler() after several hours

heppth, 2022-01-20:
Hi,

I am using libplctag version 2.4.9 on ARM. I have only a single tag. The PLC is not available in the network. After several hours my application breaks with SIGSEGV.

Single tag: `protocol=ab_eip&gateway=192.168.99.225&cpu=micro800&elem_size=1&name=SIM_BYTE`

backtrace from gdb:
```
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x0009dd1c in tag_tickler ()
[Current thread is 1 (LWP 25732)]
(gdb) bt
#0  0x0009dd1c in tag_tickler ()
#1  0x0008a798 in tag_tickler_func ()
#2  0x0012d1b4 in start ()
#3  0x001373b4 in __clone ()
#4  0x001373b4 in __clone ()
#5  0x001373b4 in __clone ()
#6  0x001373b4 in __clone ()
#7  0x001373b4 in __clone ()
#8  0x001373b4 in __clone ()
#9  0x001373b4 in __clone ()
#10 0x001373b4 in __clone ()
#11 0x001373b4 in __clone ()
#12 0x001373b4 in __clone ()
#13 0x001373b4 in __clone ()
#14 0x001373b4 in __clone ()
#15 0x001373b4 in __clone ()
#16 0x001373b4 in __clone ()
#17 0x001373b4 in __clone ()
#18 0x001373b4 in __clone ()
#19 0x001373b4 in __clone ()
#20 0x001373b4 in __clone ()
...
```

kyle-github, 2022-01-20:
Thanks for bringing this up, @heppth.   Can you please try to find the line of code with gdb?   Stack Overflow has a question about that [here](https://stackoverflow.com/questions/8545931/using-gdb-to-convert-addresses-to-lines).  The tickler routine is probably the one in src/protocols/ab/eip_cip.c.

Are you using callbacks?

When you say that the PLC is not on the network does that mean that the library is not connected at all?

Can you rerun the test with debugging enabled (PLCTAG_DEBUG_DETAIL)?   

This sounds like some sort of race condition, but may be platform specific.   

What kind of ARM platform is this?  32-bit?  64-bit?   Because you are using gdb, I assume a Linux distro.   Which one?

heppth, 2022-01-20:
I am afraid that I build libplctag without debug symbols. So I won't have any further information.
```
(gdb) info line *0x0009dd1c
No line number information available for address 0x9dd1c <tag_tickler+280>
```
No, I am currently not using any callbacks.

Yes, it was a test with interrupted network connection. No connection to the PLC.

This test ran on: `Linux raspberrypi 4.19.66-v7l+ #1253 SMP Thu Aug 15 12:02:08 BST 2019 armv7l GNU/Linux`

Yes, this sounds like a race condition. I will try to repead this test isolated with debugsymbols and `PLCTAG_DEBUG_DETAIL` enabled this week.

kyle-github, 2022-01-20:
Can you please describe the test outline?   Do you start with the PLC connected and then take it off the network?   How often are you repeating the attempt to read?  Is this test multi-threaded (in the application)?

Is there any test code you can share to help repro this?  

I do have a RPi running 32-bit ARM Raspian.   I have to dust it off and see if I can repro this.   I also have a M1-based Mac, but that is 64-bit and running macOS and who knows if that will repro.   It is also at least an order of magnitude faster than the RPi so I have found that race conditions are much harder to hit there because everything completes so fast.

heppth, 2022-01-21:
Thank you Kyle for that fast response.

I rebuild my application with debug symbols of libplctag. Now I have a little bit more information:
```
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x0009438c in lock_release (lock=0x0) at /root/libplctag/src/platform/posix/platform.c:962

warning: Source file is more recent than executable.
962         __sync_lock_release((int*)lock);
[Current thread is 1 (LWP 18841)]
(gdb) bt
#0  0x0009438c in lock_release (lock=0x0) at /root/libplctag/src/platform/posix/platform.c:962
#1  0x0009debc in check_read_status_connected (tag=0x76e61db8) at /root/libplctag/src/protocols/ab/eip_cip.c:1368
#2  tag_tickler (tag=0x76e61db8) at /root/libplctag/src/protocols/ab/eip_cip.c:247
#3  0x0008a8b8 in tag_tickler_func (arg=<optimized out>) at /root/libplctag/src/lib/lib.c:534
#4  0x0012d2d4 in start ()
#5  0x001374d4 in __clone ()
#6  0x001374d4 in __clone ()
#7  0x001374d4 in __clone ()
#8  0x001374d4 in __clone ()
#9  0x001374d4 in __clone ()
#10 0x001374d4 in __clone ()
#11 0x001374d4 in __clone ()
#12 0x001374d4 in __clone ()
#13 0x001374d4 in __clone ()
...
```

What happened inside my application: Not much! The PLC was never reachable. Basically the following code is called cyclically.
```
plc_tag_create("protocol=ab_eip&gateway=192.168.99.225&cpu=micro800&elem_size=1&name=SIM_BYTE", 2000);
usleep(20000);
```
This always returns `PLCTAG_ERR_TIMEOUT`.

kyle-github, 2022-01-21:
I think I see what is happening here.   The lock var is not in the tag.  It is in a request.  That request is being aborted and destroyed.   The spin lock loop here succeeds in locking the lock, but on exit from the lock-protected block, the request has been deleted out from underneath it and thus the lock is null (I usually set things to zero or null to cause this sort of error rather than randon weirdness).

This would probably be almost impossible to trigger on a faster system as the timing would have to be perfect.

Thanks very much for finding this.  This is my focus now as this is a definite crash bug.

kyle-github, 2022-01-21:
It looks like the fix here is to attempt to increment the reference count on the request.   Hmm, that might not be enough...   I need to think through this.   It might involve different ordering in the tag destructor too.

kyle-github, 2022-01-23:
This is going to take a bit longer than I thought.  I am not seeing how the request buffer is being cleared without the tag API mutex being held.  Another oddity is that the lock variable in this case is not allocated separately.   If there is a request, it has a lock value as a field.  So it simply is not possible to get a null pointer for a lock field.   I wonder if taking an address of a field from a pointer that itself is null returns null in some cases?  Normally that dereference itself causes a segfault.

Something is not quite right here.   I noted that the output from GDB that you copied has this line in it:

```text
warning: Source file is more recent than executable.
```

Is it possible that the address at which the SEGV is happening is not actually in lock_release()?  Is there something weird going on with the clocks on your RPi?

I have some ideas that will help protect against any null dereferencing here and I will be trying those.

heppth, 2022-01-23:
The warning is explainable: The problem does not occur on my RaspberryPi, but on that of my colleague. I then copied the executable, dump and source code on my Pi to run gdb there.

I have analyzed your source code a bit. I think the problem is that `*rc_dec_impl(const char *func, int line_num, void *data)` returns `NULL` even if the reference count is greater than 0. So in `check_read_status_connected` the `tag->req` poitner is non-zero, it acquires a lock, but in the same time an other function calls `rc_dec`. Although `check_read_status_connected` is working on the request buffer, the `tag->req` pointer is set to zero.

### Two thoughts (which are not fully thought out): 
1. If the `tag->req` pointer can be set to zero at any time, wouldn't the pointer then simply have to be copied to a local variable to fix the problem?
2. Is it logical and reasonable that `rc_dec` returns a null pointer even if the buffer is preserved?

### Off-topic: 
Your lock implementation, like mine, is based on atomic operations. Thereby I noticed a code passage: 
https://github.com/libplctag/libplctag/blob/3f67cee6f931da92ca573532290ca3bf50907d84/src/platform/posix/platform.c#L952-L957
I am calling [sched_yield](https://linux.die.net/man/2/sched_yield) on Linux or [Sleep(0)](https://docs.microsoft.com/en-us/windows/win32/api/synchapi/nf-synchapi-sleep#parameters) on Windows in the body of the while loop, because I think I can save CPU time with it. What do you think about this?

kyle-github, 2022-01-23:
As best as I can tell the problem is roughly this:

- The tickler thread calls the tickler for a tag which has a request in flight.  That calls through to check the status of that request.   That check acquires the spin lock.   Then the thread is rescheduled with the spin lock held.
- Another thread, likely the application, finally gives up and times out calling `plc_tag_destroy()`.   That in turn aborts all requests in flight.   Aborting will end up calling `rc_dec()` on the request and using the result to null out the request reference in the tag.  However, internally, aborting takes the same spin lock as the previous step.
- We then return to the tickler thread and it continues on and eventually tries to release the spin lock which then fails because the reference to the request is gone out from underneath it.  Though this part is weird because the lock functions take a pointer to an atomic integer and that integer is part of the request.  It is not allocated separately, so it could not be null.

Where things get odd for me is that if the spin lock was acquired, the abort process should have hung before it cleared out the tag reference to the request because it takes the same spin lock and tries to set the abort flag in the request.  And because that spin lock uses atomic access, that provides a memory barrier so the compiler should not be able to hoist the reference change above the spin lock.  Thus access should be protected there.   This was designed fairly carefully to avoid this exact situation.  Clearly not carefully enough, but...

So I can see what must have somehow happened, but I cannot figure out how it could happen.  As I said in an earlier comment, that part has me stumped.   

The behavior of `rc_dec_impl()` is intentional.   If you are releasing a reference you cannot keep it around.   C does not have move semantics but this would normally be a move in C++ or Rust.  The pattern:

```c
x = rc_dec(x);
```

Is an idiom I use to make sure that the reference is released and no remaining copy of the pointer is left behind for me to accidentally use.

The implementation of spin locking was based on the assumption that spin locks would be rarely used.   I am slowly getting rid of all uses of spin locks.   As can be seen in this bug, it is possible for the thread to be rescheduled between grabbing the lock and releasing it.   Thus anything waiting is going to be busy waiting.  Potentially for a really long time.

Making the scheduler yield does relieve the CPU, but it causes huge delays.   On Windows since about a year ago, the default schedule task "tick" is now about 15ms.   So every time you call `Sleep(0)`, you are basically guaranteed that that thread will not be rescheduled for at least 15ms.  I found this out a few months ago and had to do some serious surgery to libplctag to make it perform well on Windows.  The difference in overall performance was enormous.

The best option is to not use spin locks at all.  All modern operating systems (and many of the embedded ones) have really good implementations of mutexes where you often do not even enter the kernel if the mutex is not contested (the common case).  This involves some atomic accesses, but so do spin locks.

This is all pushing me to change how I deal with shared objects.   I implemented the reference counting to ensure that the logic for when shared objects were deleted was sane.   However, it is not enforcing any exclusive access.   So each object needs a mutex as well.  Unfortunately, that is getting complicated because there are situations where changes can be made without the mutex being held (see how `rc_dec()` in the `plc_tag_destroy()` is called and thus calls `ab_tag_abort() which removes the request in flight).

I have come up with a new way (for the library, not new in programming!) that will use non-pointer handles (so I can't accidentally cheat and use them directly as pointers) and combine both the reference counting and the exclusive access so that I can pull macro tricks like I use in `critical_block(mutex) { }`.   That little macro trick has caught and prevented so many mutex bugs!

However, this is not going to be fast to do.   So in the short term, I am going to do the following:

1. Remove the spin locks in the request.   The more I learn about the behavior and the clear indication in this bug that the scheduler can hit between the lock and unlock means that they can be hugely expensive.  And the more I research about mutexes, the more I find that they are really lightweight in the common case of no contention.  Maybe I will remove spin locks completely.
2. Guard all uses of the existing request reference with a call to `rc_inc()`.   If the request is in the process of being destroyed, this will return NULL.  It will ensure that the request reference is valid if it succeeds.   This is going to touch a lot of lines of code, unfortunately.  It is not difficult coding, just annoying.  The danger here is that I will be relying on the atomicity of access to pointer-sized values.   Any platform/CPU where that is not true will possibly break in a similar way to the bug you have found.  Luckily ARM and x86 both have the required property as do most "mainstream" CPUs.  I do not use pointers in packed structures, so the compiler _should_ be aligning things for natural access.  I do not like relying on should.

Sorry this is taking so long!

heppth, 2022-01-23:
First of all, thank you for your detailed answer. You are doing a great job.

Maybe I didn't understand the problem completely or I'm a bit too naive, but wouldn't we solve the problem by simply copying the request pointer? Here for illustration:

https://github.com/heppth/libplctag/commit/e38bc3cb5ef161eb6339159133a774daa4a0b8a2#diff-175e0a95942781edd48f4343969b109d1fb902054a2f0d1096bdd6df880d475f

kyle-github, 2022-01-23:
Unfortunately, it is not that easy.   The request is a reference counted object.   It can have the reference count dropped in other threads.  If that happens and triggers destruction of the request, then you would be causing a use-after-free error if all you do is copy the value of the pointer.   

What I have landed on is this:

```c
    request = rc_inc(tag->req);
    rc = check_write_request_status(tag, request);
    if(rc != PLCTAG_STATUS_OK)  {
        pdebug(DEBUG_DETAIL, "Write request status is not OK.");
        rc_dec(request);
        return rc;
    }

    ... use request ...

    request = rc_dec(request);
```

I refactored out the request checking into separate functions for read and write (write is shown).  The first line increments the reference count on the request.  If the request's reference count is 0, then `rc_inc()` will return NULL.   If the request pointer is NULL then `rc_inc()` will return NULL.  The check routine checks that condition as well as internal state within the request.

Because `rc_inc()` is smart enough to handle the case that the object is in the middle of being destroyed, it is safe here.  The reason this is safe is as following:

- If we get a NULL pointer for `tag->req`, then we get NULL for `request` and everything works fine.
- If we get a non-null pointer for `tag->req` then we have some sub-cases:
   - if the request is not in the process of being destroyed, then we get a good reference and prevent the request from being deleted out from under us by incrementing the ref count.
   - if the request pointer is already NULL, then we get NULL as a result and we never go further because we did not have a request.
   - if the request is in the process of being destroyed then we either get there before the ref count hits zero in which case we bump it up and the in-process `rc_dec()` in some other thread does not cause it to drop to zero and we push off the destruction until we run `rc_dec()`.   Or, if we do find the count to be zero, then we return NULL from `rc_dec()` and act as if we never had the request in the first place.

So I think this is now safe as long as we have atomic accesses to the request pointer.  Since most CPUs have atomic native word sized accesses, we are OK on all mainstream CPUs.   However, deeply embedded systems might not be OK.

kyle-github, 2022-01-24:
I have a possible fix in the prerelease branch.  I have tested it against Valgrind and made sure that memory is correctly released.  Could you check that out and try it?  The binaries are attached below.

The way to simplify this is really to do the major refactoring of the AB code to use a structure similar to that used in the Modbus protocol support.  This queues tags with outstanding requests up and does not use a separate list of request buffers.   This has a lot of good points:

- There is no request queue.   In some circumstances, you could potentially DoS the library internals and the application by overwhelming the system with requests faster than it can purge them (that would take effort).  The request queue is essentially unbounded (I put a lot of effort into controlling this so that it is not a problem except in extreme circumstances).
- There is no allocation or deallocation in the common case.   The request system used in the AB code allocates and frees a request buffer per individual request.  For embedded systems with weak allocation libraries, it is not good to allocate and deallocate a lot.   Leaks are more possible and many C allocation libraries will eventually fragment memory.
- It makes it easier to remove the library-wide tickler thread that is where many of these kinds of bugs reside.   There are at least three threads that every tag and request interact with: the application, the tickler thread, and a thread for the PLC IO communication.  A single thread takes a lot of resources.   Again, the removal of the thread is mostly useful for embedded or smaller systems.
- Request buffers are large.  By their nature they have to be the maximum possible size a request can be for the specific PLC.  For a modern ControlLogix (L80+), that could be 4kB.  Per request!  Get a few of those in flight and you start to use up a lot of memory.  The way the Modbus code works is that there is a send buffer and receive buffer per PLC and that is it.   So in the case of AB PLCs, that would be a maximum of 8kB of buffer.
- Request buffers limit the amount of packing that can be done.   Right now I pack whatever fits.  For read requests, that works OK (but would not for Omron since the request has to be packed based on both sending and receiving sizes).   For write requests, it is not as good.   When the code is creating the write request, it is not clear how much space would be available in a packed request packet.   So packing would be closer to optimal if the amount of already used space was an input to the process.

The more these kinds of bugs show up, the more I am certain that I will also need to simplify the way that shared data is handled.   Having separate functions and features to handle reference counting and to control access is not good.  Having the base for the reference count be a valid pointer is also not great because the compiler will not catch you if you forget to acquire a reference.  I could solve a lot of this by moving to C++ for the core of the library (the external API would remain C compatible) but every time I've looked into that, the size of the library increases by almost an order of magnitude and that really impacts embedded systems.

Luckily these two changes are mostly orthogonal, so I can figure out which one is most important and do that first.  It looks like moving the AB code to use the Modbus model would pay the most dividends right now.

I attached the ARM files that the standard CI system builds below.

[libplctag_2.4.11_linux_arm7l.zip](https://github.com/libplctag/libplctag/files/7922475/libplctag_2.4.11_linux_arm7l.zip)
[libplctag_2.4.11_linux_arm6hf.zip](https://github.com/libplctag/libplctag/files/7922476/libplctag_2.4.11_linux_arm6hf.zip)
[libplctag_2.4.11_linux_aarch64.zip](https://github.com/libplctag/libplctag/files/7922477/libplctag_2.4.11_linux_aarch64.zip)

heppth, 2022-01-24:
Thanks Kyle. I immediately created a new version with the prerelease branch this morning and applied it to two test devices (one with active connection and one without connection).
Both devices have now been running for 6 hours without crashing, memory leak or other issues.
Do you already know when you plan the next version?

kyle-github, 2022-01-24:
That's great!  Please let me know if anything pops up.

I am still quite concerned that I cannot repro this locally.

I'll target Wednesday for a release (that would be Thursday your time I suspect).

kyle-github, 2022-01-26:
@heppth, any signs of problems?

heppth, 2022-01-26:
Good morning @kyle-github,
All systems run smoothly. No crashes, no memory leaks, very low CPU load and no other abnormalities.

kyle-github, 2022-01-26:
OK, I'll release this today.

kyle-github, 2022-01-26:
Fix released in 2.4.11.

commit 6069f39e (2022-01-27): Fix #341 - use after free (#346)

* VS 2022 does not define WIN32 and only defines _WIN32.   This commit fixes shared examples. (#339)

* Fix 341 - fix segfault, use-after-free issue with requests (#342)

* Don't call the tag's abort function if it doesn't have one.

* In process.   Fix up handling of requests when checking request status.   Requests could be deleted out from underneath us.

* Refactor request status checking into common code.

* Remove refactored common request status checking code.

* Add common request status checking code.

* Removed commented out old code.

* Added missin [... cut]
closed 2022-01-26 by kyle-github