libplctag/libplctag #344: String defaults are dangerous

kyle-github, 2022-01-24:
This issue is a follow up to #328, where ongoing discussion surfaced the following use cases that allow data overwrites (not out side the buffer) because of the use of string attribute defaults.

Scenario 1:

- We have a UDT with a string-like UDT field followed by some other field. The string-like field is shorter than a standard STRING type.
- The application uses the string functions to write to the string. Due to the other field following the string-like field, the total space left in the tag data buffer starting at the string-like field is at least as large as the default string type needs (88 bytes).
- The library will happily take the string data passed to it and overwrite the field following the string-like field.

Scenario 2:

A similar problem occurs when the count word is different from DINT. If an INT is used in the string-like UDT, the first four bytes of the UDT will be overwritten with the count. Since AB is little endian (mostly) this would mean that the count would look right but the string data would start off with two zero bytes and the string's character data would go past the end of the actual section of the buffer reserved for the string (if the total was a multiple of 4 bytes).

This is not solvable with the current library by using defaults at all. In these cases, the defaults are dangerous.

The correct behavior would have been to not have any defaults and force the user to input string attribute values when constructing the tag attribute string.  While annoying, this would at least prevent the problems above from happening accidentally.

More discussion is needed on this as changing the existing behavior would be a breaking change.

kyle-github, 2022-01-24:
Please discuss in the comments.

timyhac, 2022-01-25:
Yikes! Does this negate the benefit of having string setter/getters entirely?

radrevere, 2022-01-25:
I had a good discussion on this topic with one of the PLC programmers at work.  Because UDT strings are created to mimic the standard PLC string they can be read and written like one, with the exception of the buffer length being different.  The need here is if a PLC dev doesn't want to use the full 86 bytes to 'save space' or to fit it into a larger UDT structure.  The other need, which is much more rare, is if the string needs to be longer than 82 characters.  In both cases it is very convenient to use the built in string accessors that you have because it saves time and extra coding required to handle the raw buffer.  But in both cases the buffer is handled as if it is a default string.
That being said he made the suggestion of specifying the size information to the library.  May I then suggest that you have a new set of accessor functions for UDT strings?
int plc_tag_create_ex(const char *attrib_str, int timeout, int max_buffer_len);
this new create function would allow the tag attributes to be setup appropriately for UDT types.  Or possibly:
int plc_tag_create_udtstr(const char *attrib_str, int timeout, int len_byte_count, int max_buffer_len);
for strings specifically.
If done right the new create function will setup the necessary boundaries for interacting with the tag and the normal string read and write functions can be used.  
The alternative would be to specify expected buffer sizes in the accessor functions:
int plc_tag_get_udtstring(int32_t tag_id, int len_byte_count, int max_string_len, char *buffer, int buffer_length);
int plc_tag_set_udtstring(int32_t tag_id, int len_byte_count, int max_string_len, const char *string_val);
no start offset needed because it is expected to be only the string that is to be read.  This is more cumbersome in the long term so a new create function would be preferable.
What are your thoughts on this?

radrevere, 2022-01-25:
In thinking about it, the new create function would be ideal and easy to implement:
```c
int plc_tag_create_udtstr(const char *attrib_str, int timeout, int len_byte_count, int max_buffer_len)
{
    int id = plc_tag_create(attrib_str, timeout);
    if(id != PLCTAG_STATUS_OK)
    {
        return id;
    }
    plc_tag_p tag = lookup_tag(id);
    if(!tag) {
        pdebug(DEBUG_WARN,"Tag not found.");
        return PLCTAG_ERR_NOT_FOUND;
    }
    tag->byte_order->str_max_capacity = max_buffer_len;
    tag->byte_order->str_count_word_bytes = len_byte_count;
    return PLCTAG_STATUS_OK;
} 
```
Of course any other checks or adjustments needed could be done here.  Everything else is standard fare and already handled.

kyle-github, 2022-01-25:
There are already special tag attributes to set these values when creating a tag.   See the [Optional Tag Attributes](https://github.com/libplctag/libplctag/wiki/Tag-String-Attributes#optional-generic-attributes) of the project wiki section, specifically the attributes that start with `str_`.

Having fixed functions with special arguments is a non-starter as it would require API changes every time a new kind of string was supported.  And due to the problems of passing structured data from wrappers, I cannot make this an option struct that gets filled in by the application.   At least not without a lot of heavy lifting on the part of the wrapper/application.

Just Allen-Bradley has the following _standard_ string types:

- PLC-5 strings.  These have a 2-byte count word followed by 82 bytes of character data.   You cannot define your own string types so the defaults work well here.
- MicroLogix strings.   These have a 2-byte count word followed by 82 bytes of character data.   Unlike PLC-5, each 16-bit word is sent in big-endian memory order so every word is effectively byteswapped.   The string "Hello" is sent on the wire as "eHll<nul>o" where <nul> is a null byte.  The defaults are always accurate here.
- Standard ControlLogix strings.  These have a 4-byte count word, 82 bytes of character data followed by 2 bytes of padding.  This is the current default if the PLC is Control- or CompactLogix.
- Tag listing name strings use a 2-byte count word followed by the character data.   However, they are not fixed size.  Only the characters in the string are included.   There is no nul character termination.  If you use the magic "@tags" tag name, this is the default.
- UDT name and field name strings.   These are C-style strings with no count word and are terminated with a nul character byte.   If you use the magic "@udt" tag name, this is the default.
- Micro800 and Omron have yet different string definitions.

These are just the standard string types AB uses.   If your strings fall into the above, the defaults always work and the existing string functions in the library API work great.

I hope this explains a bit why the existing defaults are a problem.   It took years (literally) to come up with the current string API and string attribute combination and I still got it wrong!

Modbus PLCs have no defaults today because I have no idea what they should be.  All the string functions will fail if you call them on a Modbus tag _and_ you have not explicitly set the string configuration attributes in the tag attribute string at tag creation time.

Where we get to problems is the case where someone made a UDT to save space (99% of the cases) on a Control/CompactLogix.   Unfortunately, it is not clear to me how this could be handled automatically within the library.  Even when tags are listed, you would need to somehow know that a specific UDT type was a special string.   How could the library figure that out?

What I should have done (hindsight is always 20/20) is to have no default string values  for Control/CompactLogix.   That would cause the string functions to fail immediately unless the application opened the tag while specifying the special string attributes to define what the string looks like.  The user experience would not be great, but at least it would not be dangerous.

kyle-github, 2022-01-26:
The question here is, what to do next?  Leaving the existing defaults in place guarantees some broken behavior as noted in the problem description.  However, dropping the defaults will break a lot of code that works today when standard STRING types are used.

kyle-github, 2022-01-26:
@radrevere Good points on the docs.  Maybe that is the right way to handle this.   Just document it to death and put all the caveats in bold print everywhere.  

The fact is that these settings are only dangerous if you are using string-like UDTs.  And if you use those, you need to know you are using those in general.

The goal of the library is to make it easy to do things the "right" way and lower friction on more difficult things.  It cannot protect the user from everything.

I do think that silently storing the data is not the right way to do things.  If we know that something is off, we should report it even if the exact operation requested just happens to be safe based on the data in the call.  We cannot get to 100% with this, but I do think that reporting the error when we can is better than trying to make things work.

The way I look at it is like JavaScript's solution to handling equality.   Making things "just work" has ended up being a serious problem and now you have "===" etc.   Something that would have stayed relatively simple if they had not bent over backward to make semi-incorrect code work has become a huge source of pain.

The more I think about this, the more I like the idea of just documenting this to the point where anyone running into problems clearly has done zero effort to understand the requirements and limitations.  That is just bad programming then.

Anyone have objections to simply documenting this to death?  @jkoplo? @timyhac?

kyle-github, 2022-01-26:
I can go so far as to rename the examples "standard_string.c", "user_udt_string.c" etc.

radrevere, 2022-01-26:
I am good with the documentation solution.  To your point, building band-aids to 'fix' outlier problems isn't a good solution in the long term as it leads to more complicated and less maintainable code.  I like the idea of renaming "string.c" to "standard_string.c" for clarity.  And perhaps start a "udt_string.c" with examples for handling UDT strings for different PLC types.

kyle-github, 2022-01-27:
Any other comments on this?   Otherwise I will start on the documentation changes and renaming.  I think I will copy the forum post I wrote into the Wiki.

jkoplo, 2022-01-27:
I think solving this on the documentation/user-side makes sense. Doing custom string types on a PLC should be somewhat outside of the norm. I think in any language, when you start deviating from the built-in, standard types you shouldn't be too surprised when the rest of the language and tooling doesn't work without change. If I decided to not use the built in `string` type in C# and instead implemented it as a fixed size byte array I wouldn't be surprised that `String.Trim` doesn't work on the new type I created...

@kyle-github how do you feel about moving documentation out of the wiki and into markdown files in a `doc` folder? I've seen the light on that and I think it would improve things a bit. For one thing, anyone checking out the source code would have the docs in the same directory right in their face.

kyle-github, 2022-01-27:
I think the documentation approach is the right way to do this.  I agree with your reasoning, @timyhac. 

On the wiki issue, I am not sure.   Right now, the wiki gets good search hits.   But your point about having the docs right in the download is a good one.   It is really too bad that GitHub made the wiki a separate repository.

I suppose I could just write them up as standard Markdown and then render them and keep both the .md and .html files in the main repo....  Hmm.  Maybe render them straight to PDF?

I'll treat the wiki issue separately.   Look for some example renaming and updates to the existing wiki in the next few days.

jkoplo, 2022-01-27:
If we do things with markdown files in a docs folder I am pretty confident we can wire up a github action to build those docs to pdf and include them in the releases.

I think you started a github.io page? Maybe we host wiki there from the docs folder - that should get pretty good hits too.

kyle-github, 2022-01-29:
There is a github.io page.  I created it a while back.  It is yet another repo (like the wiki).  When you create a repo, you more or less get three repos:

1. Your code.
2. Your wiki.
3. A reservation for xyz.github.io.   I think the site does not exist until you push something to it, but it is reserved.

Unfortunately, they all seem to be separate from each other.   I'll do some research to find out if there are good ways to tie them together more.

kyle-github, 2022-01-30:
This is in progress. I added a [new wiki page](https://github.com/libplctag/libplctag/wiki/Strings-and-String-Handling) for string handling.  I broke up the tag attributes into parts and made [string attributes](https://github.com/libplctag/libplctag/wiki/Tag-String-Attributes#optional-generic-string-attributes) their own section.

I will tackle merging the wiki into the general code later.   I think that generating a PDF from the wiki and including that into the build may be my first step.

kyle-github, 2022-02-02:
I think a some GitHub actions will help here:

1. store the whole wiki in the main repo.
2. copy the wiki pages to the wiki repo on commit to release.
3. transform the docs into pdf on release.

commit 1b329d1e (2022-02-14): Release for 2.5.0 (#354). Fix #275.  Fix #338. Fix #344. Fix #352.

* Don't call the tag's abort function if it doesn't have one.

* Remove refactored common request status checking code.

* Add common request status checking code.
* Fix duplicate slash in path.

* Bump version for request use-after-free fix.

* callback with user data (#345)

* add callback with user data

* add test_callback2

* rename plc_tag_register_callback2 to plc_tag_register_callback_ex

* rename test_callback2 to test_callback_ex

* refer to each other

* start fix ups of version.

* Break apart the ifdef checks as s [... cut]
closed 2022-02-14 by kyle-github