libplctag/libplctag #352: Clarification about callback unregistration

kpreisser, 2022-02-09:
Hi, sorry if this is not the appropriate place to open this. I have a question about unregistration of callbacks which is not clear from the documentation:

The [Wiki](https://github.com/libplctag/libplctag/wiki/API#callbacks) mentions the `plc_tag_unregister_callback` to unregister a callback, and also mentions that a callback might be called from a library thread.
Is the previously registered callback guaranteed to **not** be still called/executed once `plc_tag_unregister_callback` returns? E.g. in case a callback would be called in a library thread, does this function wait until the callback was executed (if the callback is just about to be called or is currently executing), or could it happen that the callback is still called after `plc_tag_unregister_callback` returns?

This would is important to know when using the callback registration with user data (`plc_tag_register_callback_ex`), as for example in .NET you can use a [function pointer](https://devblogs.microsoft.com/dotnet/improvements-in-native-code-interop-in-net-5-0/#function-pointers) to a static method (for optimal performance), and then use a `GCHandle` to map a .NET object to a pointer (`IntPtr`) which you can then use in the static callback method to map the pointer back to the .NET object. For this, you would allocate the `GCHandle` before calling `plc_tag_register_callback_ex`, and free it after calling `plc_tag_unregister_callback`. However, if the callback could still be called after that function returns, the `GCHandle` might already be freed and this might cause invalid memory access in the callback function.

Thank you!

kpreisser, 2022-02-09:
From what I understand when looking at the C code (but please note I'm not a C programmer so I might not interpret it correctly), the `plc_tag_unregister_callback` function locks at the tag's `api_mutex` in order to clear  `tag->callback`:

https://github.com/libplctag/libplctag/blob/6069f39eb5109ed9aca49c978731957e3a23dca1/src/lib/lib.c#L1130-L1156

However, in `THREAD_FUNC` (which I assume is called from a library thread), it looks like it first locks on the tag's `api_mutex`, but then releases the lock and then calls the callbacks (outside of the lock):
https://github.com/libplctag/libplctag/blob/6069f39eb5109ed9aca49c978731957e3a23dca1/src/lib/lib.c#L570-L574

https://github.com/libplctag/libplctag/blob/6069f39eb5109ed9aca49c978731957e3a23dca1/src/lib/lib.c#L385-L397

Do I understand correctly that this can mean that the callback could still be called after `plc_tag_unregister_callback` returns?

For example, if the thread that is currently executing `plc_tag_generic_handle_event_callbacks()`  is at the point where it has read the `tag->callback` value (that is not yet `NULL`) and is just about to call that function, and before that happens, another (application) thread calls `plc_tag_unregister_callback`, which enters the mutex, clears the callback, and returns. If then the library thread continues, it would still call the callback function (through the previously read `tag->callback` value) even though `plc_tag_unregister_callback` already returned previously in the other thread.

Thanks!

kpreisser, 2022-02-09:
**Edit:** Sorry, I noticed I was looking at the wrong branch previously - in branch `prerelease` which contains the new `userdata` functionality, it looks like the callbacks are executed within the tag's `api_mutex`. Because `plc_tag_unregister_callback` also locks on that mutex, it should be guaranteed that once `plc_tag_unregister_callback` returns, the callback can no longer be executed, so this should be safe.

I think this should be clarified in the documentation. Thank you!

kyle-github, 2022-02-09:
It all should be safe, but I still need to walk through all the possible code paths before I release 2.5.0.  I had to make the mutexes recursive which means that it is possible for a single thread of control to call the removal of the callback while within the callback.  

This is one of the many things that need to be refactored.  There are too many areas where such access is not enforced and should be.   I have some ideas here but need to do some testing first.

And yes, it is possible that events can be raised somewhat out of order in some circumstances.  That is the nature of the asynchronous core.

kyle-github, 2022-02-09:
If you do find a location that is not guarded by the mutex, please let me know!  Hopefully I have not missed one.  More eyes on the code is always welcome!

I will be writing some tests to use a lot of threads and add/remove callbacks often.

kyle-github, 2022-02-09:
The change to use recursive mutexes prevents other threads from calling these API functions in dangerous ways but it open up the possibility of unregistering the callback from within a callback call chain.   

I will add some checks to help catch that case, but there are instances where this is going to be a problem.   I will look through the call chains possible to see if I can come up with ways to call the API functions, generate a callback and call through NULL or call with a NULL user data pointer.

kyle-github, 2022-02-09:
It looks like I need to figure out possible compiler changes to code order.   In some circumstances, the compiler can legally reorder the execution.  If that could result in a case where the function pointer was not NULL but the user data pointer had been set to NULL, we might have a problem.  I might need another guard there.

kyle-github, 2022-02-11:
I have updated the prerelease branch to handle all the instances where the callback was called directly.

I convinced myself that the compiler will not reorder the stores within a single thread.   It is not clear to me whether a recursive mutex acts as a memory barrier within a single thread but if the compiler reordered stores in that circumstance, it would break regular single threaded code.

closed 2022-02-14 by kyle-github

kpreisser, 2026-09-29:
Hi! Just a heads-up question to this topic if you don't mind:
With current versions of libplctag (v2.7.2), when you have created a tag with `plc_tag_create_ex` and passed a `tag_callback_func` and `userdata` pointer, is it **guaranteed** that after destroying the tag with `plc_tag_destroy`, the library will not call the specified `tag_callback_func` with the specified `userdata` pointer again (e.g. from another thread)?
Is this also guaranteed after calling `plc_tag_unregister_callback`?

(This is important for lifecycle management, e.g. when passing a pinned object handle from a runtime such as .NET, that you don't free the handle until the callback will no longer be called with the specified pointer.
If the library guarantees that after the call to `plc_tag_destroy` or `plc_tag_unregister_callback` returns, the callback will no longer be called with the specified pointers, it should be safe to free the memory/handle after calling those functions.)

Thanks!

kpreisser, 2026-09-29:
Another question: If `plc_tag_create_ex` **fails** (returns a value < 0), is it guaranteed that the specified `tag_callback_func` is never called with the specified `userdata` again after `plc_tag_create_ex` returns?

As API consumer, I would assume this is guaranteed, as otherwise it wouldn't be possible to correctly free the memory/handle pointed to by the `userdata` pointer (because it wouldn't be possible to know whether the callback will be called at all, and whether the memory would needed to be freed in the `PLCTAG_EVENT_DESTROYED`).

However, I asked this an AI (Claude), which came to the conclusion that this is not always guaranteed in the current v2.7.2. Can you check whether this report is accurate, and whether this can be fixed? Thanks!

---
**Title:** Callback may be invoked with stale `userdata` after `plc_tag_create_ex()` fails

**Version:** v2.7.2

### Description

`plc_tag_destroy()` guarantees that the tag callback is not invoked after the function returns: `destroy_tag_common()` in `src/libplctag/lib/lib.c` clears `tag->callback` and `tag->userdata` under `tag->api_mutex`, and `plc_tag_generic_handle_event_callbacks()` checks the callback under the same mutex. This allows wrappers to safely free the memory referenced by `userdata` once `plc_tag_destroy()` has returned.

However, the failure paths in `plc_tag_create_impl()` that run after the tag has been published to the tag hashtable do not provide this guarantee. This affects the case where the initial status check returns an error, as well as (with `timeout > 0`) a failed `cond_wait()` and a timeout or error status after waiting. These paths only remove the tag from the hashtable and call `rc_dec(tag)`. `tag->callback` and `tag->userdata` are not cleared.

Since the tag is already visible in the hashtable at that point, the tickler thread may have acquired its own reference (`rc_inc()` in `tag_tickler_func()`) and can call `plc_tag_generic_handle_event_callbacks()` after `plc_tag_create_ex()` has returned an error. Pending events can exist in this situation; for example, the AB protocol code raises `PLCTAG_EVENT_CREATED` with an error status (`ab_common.c`, `eip_cip.c`).

A caller that frees its `userdata` after `plc_tag_create_ex()` returns an error, which seems like the natural thing to do, can therefore see the callback invoked with a dangling pointer. In our case (a .NET wrapper passing a `GCHandle` as `userdata`), this results in undefined behavior and potentially a crash.

**Note:** This was found by code review, not reproduced. The window is timing-dependent.

### Suggested fix

In the failure paths of `plc_tag_create_impl()` that run after the tag has been added to the hashtable, clear the callback the same way `destroy_tag_common()` does before releasing the reference:

```c
critical_block(tag->api_mutex) {
    tag->callback = NULL;
    tag->userdata = NULL;
}
```

Alternatively, factor this into a common helper used by both `destroy_tag_common()` and the create failure paths. It would also be helpful to document in `libplctag.h` that no callback is invoked after either `plc_tag_destroy()` returns or `plc_tag_create_ex()` returns an error, so wrappers can rely on this contract.

kyle-github, 2026-09-29:
This looks possible but I need to go through the code carefully.  You should be able to capture the tag destruction event if you want more safety, but I need to check that too.

kpreisser, 2026-10-27:
Thanks! I opened issue #684 to track the remaining callback issue in `plc_tag_create_ex`.

commit 1b329d1e (2022-02-14): Release for 2.5.0 (#354). Fix #275.  Fix #338. Fix #344. Fix #352.

* Don't call the tag's abort function if it doesn't have one.

* Remove refactored common request status checking code.

* Add common request status checking code.
* Fix duplicate slash in path.

* Bump version for request use-after-free fix.

* callback with user data (#345)

* add callback with user data

* add test_callback2

* rename plc_tag_register_callback2 to plc_tag_register_callback_ex

* rename test_callback2 to test_callback_ex

* refer to each other

* start fix ups of version.

* Break apart the ifdef checks as s [... cut]