libplctag/libplctag #355: add parameters to Makefile

ronytesler, 2022-02-21:
I have a code in c that I now compile like this:
gcc -std=c11 -pthread test.c -o test `pkg-config --cflags --libs libsystemd`

I want to modify one of the example files to include my code, let's say simple.c.
How can I add these settings to the existing Makefile?

kyle-github, 2022-02-22:
If you want to include libplctag in your build process, add libplctag to the libraries list and include libplctag.h in the files that use the library functions.

```text
gcc -std=c11 -pthread -lplctag test.c -o test `pkg-config --cflags --libs libsystemd`
```

The library needs to be installed where gcc can find it such as /usr/local/lib or you can pass flags to gcc to tell it where to search, and the include file needs to be installed where it can be found too.

ronytesler, 2022-02-22:
I just manged to do it. I though I needed to do something with the make file, but I just copied the libplctag output files to /usr/lib and compiled with gcc -std=c11 -pthread -lplctag test.c -o test `pkg-config --cflags --libs libsystemd` and it worked.
Thanks!

closed 2022-02-22 by ronytesler