libplctag/libplctag #356: CompactLogix 5069 error C#:  PLCTAG_ERR_BAD_GATEWAY

igal92, 2022-02-27:
Hello dear community,

I'm new at this, so thank you in advance for all the help you can provide: 

I'm trying to read a tag (TestDINTArray[0], first element DINT) on my CompactLogix 5069-L320ER, but I keep receiving the error PLCTAG_ERR_BAD_GATEWAY after the line trying to create the tag: "client.AddTag(tag);"
so at first i'm trying to check if the connection is established.

In addition, I watched the following video:
https://www.youtube.com/watch?v=s7sMrVXfaE4&t=25s 

after implementing the relevant code type for the .LGX type, still can't receive communication.
Is there anything wrong with my code? I think the path is ok the 5069-L320ER is located in the first slot and the port used is A2, therefore the "1, 1", or i don't understand it quite well?.. also tried the default "1, 0".

The IP address is definitely OK, and the controller was tested while in RUN and REM PROGRAM mode.

any ideas what i might doing wrong?

the following is the code:

```

using System;
using System.Collections.Generic;
using System.Linq;
using System.Text;
using System.Threading;
using System.Threading.Tasks;
using LibplctagWrapper;

namespace MyProject
{
    class Program
    {
        static void Main(string[] args)
        {
            try
            {
                //reads the first 5 elements in array TestDINTArray[0]
                var tag = new Tag("192.168.2.100", "1, 0", CpuType.LGX, "TestDINTArray[0]", DataType.Int32, 1);

                using (var client = new Libplctag())
                {
                    //adds the tag
                    client.AddTag(tag);

                    //checks if tag was added, if returns pending - retry.
                    while (client.GetStatus(tag) == Libplctag.PLCTAG_STATUS_PENDING)
                    {
                        Thread.Sleep(100);
                    }
                    //if status is not ok, need to handle the error
                    if (client.GetStatus(tag) != Libplctag.PLCTAG_STATUS_OK)
                    {
                        Console.WriteLine($"Error setting up tag internal state. Error {client.DecodeError(client.GetStatus(tag))}\n");
                        return;
                    }
                }
            }
            finally
            {
                Console.ReadKey();
            }

        }
    }
}

```

kyle-github, 2022-02-28:
Can you use the command line tool tag_rw2 to see if the PLC is actually at that address?   Generally a bad gateway error comes when you cannot create a new connection to a PLC.

igal92, 2022-02-28:
> Can you use the command line tool tag_rw2 to see if the PLC is actually at that address? Generally a bad gateway error comes when you cannot create a new connection to a PLC.

Thank you for answering.
I'm not sure how to use the tool, but here is more info about the setup:

The PLC is connected through Ethernet cable to the workstation, the IP of the PLC is 192.168.2.100 and he has dual ports.
I setup as Linear/DLR so it doesn't matter if you connect to A1 or A2.
therefore i changed the line to:
`                var tag = new Tag("192.168.2.100", "2, 0", CpuType.LGX, "TestINT", DataType.Int16, 1);
`
The path "2, 0" as "2" is for Ethernet, right? according to Tag.cs file
Also tried different path, i'm sure i miss something...
not sure how to proceed here :( is it possible to even establish connection to this one?

`
        /// <summary>
        /// Creates a tag. If the CPU type is LGX, the port type and slot has to be specified.
        /// </summary>
        /// <param name="ipAddress">IP address of the gateway for this protocol. Could be the IP address of the PLC you want to access.</param>
        /// <param name="path">Required for LGX, Optional for PLC/SLC/MLGX IOI path to access the PLC from the gateway.
        /// <para></para>Communication Port Type: 1- Backplane, 2- Control Net/Ethernet, DH+ Channel A, DH+ Channel B, 3- Serial
        /// <para></para> Slot number where cpu is installed: 0,1.. </param>
        /// <param name="slot"></param>
        /// <param name="cpuType">AB CPU models</param>
        /// <param name="name">The textual name of the tag to access. The name is anything allowed by the protocol. E.g. myDataStruct.rotationTimer.ACC, myDINTArray[42] etc.</param>
        /// <param name="elementSize">The size of an element in bytes. The tag is assumed to be composed of elements of the same size. For structure tags, use the total size of the structure.</param>
        /// <param name="elementCount">elements count: 1- single, n-array.</param>
        /// <param name="debugLevel"></param>`
and for start trying to read single tag of "INT" type, size=16.

kyle-github, 2022-02-28:
Are you using the .Net wrapper from [libplctag.Net](https://github.com/libplctag/libplctag.NET)?  That's the main supported wrapper for the library.

Paths are specific to ControlLogix-class PLCs.   The path is in pairs of port/node.  The first is the port.   Port 1 is the backplane on almost all (perhaps all) modules.   So if you are entering a PLC via an Ethernet link, the first number is almost always a "1".   The next is the slot/node ID.   If you are trying to get to a CPU module in a ControlLogix chassis, you need to put in the slot number next.   Slot zero is the left-most slot.   Slots count from left to right starting from zero.

Is the CPU module in slot zero?

igal92, 2022-02-28:
Hi Kyle, thanks for answering.
I created a new path in RsLinx.
apparently I had a wrong path, just up to the IP address without /Backplane/0 .
Now the code can pass but still seeing wrong value.

I will work on this tomorrow and if needed will open a new issue.

Thank you very much!

kyle-github, 2022-03-03:
Hi @igal92, changing the path in RSLinx does not change the actual path.  You need to look at the chassis.  The hardware determines the path.  If the CPU module is in slot 0, then your path is "1,0".  If your CPU module is in slot 4, then the path is "1,4".

What slot is the CPU module in?

igal92, 2022-03-03:
Hi kyle,

The CPU module is in slot 0. It's located physically at the most left slot, between the controller to the I/O modules.
Can also be confirmed as the CPU slot is [0] in Studio 5000, and the program runs.
Are there any configurations that needs to be done such as to "open" incoming connections?

igal92, 2022-03-03:
EDIT: 
Apparently the issue was with the "1, 0". i had a space between the coma and the number so the UniqueKey had a space in the middle.. that's why it didn't recognize the path.
i removed the spaces in the string, and now it works.
 
before:
in var tag = new Tag(“192.168.2.100”, “1, 0”, CpuType.LGX, “TestDINT”, DataType.Int32, 1)

after: (note the space between the coma and the "0" in "1,0".
in var tag = new Tag(“192.168.2.100”, “1,0”, CpuType.LGX, “TestDINT”, DataType.Int32, 1)

Also the scope of the tags needs to be in the Controller, and not in the Program.

Thank you for the assistance!

kyle-github, 2022-03-06:
Grr.   You are not the first person to find that problem.   I will call this a bug.   It has hit too many people.  I will probably still throw a warning, but it is just too common and the library should gracefully handle this.

kyle-github, 2022-11-14:
This is, finally, mostly fixed in version [2.5.4](https://github.com/libplctag/libplctag/releases/tag/v2.5.4).  I say "mostly" because I do not strip all spaces out of the tag attribute string.  I leave them in attribute values.   I also string spaces out of path values.

closed 2022-11-14 by kyle-github