libplctag/libplctag #357: Problem with the libPLCTag .Net Framework example Project

BartGuy1215, 2022-03-05:
This is kind of a strange one, and I REALLY need some help figuring this out!  I have a VS2019 .NET4 framework based project that needs to now communicate to an AB CLX L81E PLC.  

I compiled the .NET Core example and it runs fine on .NET5.  Perfect!  But my project is based on .NET4 Framework and I was having issues with references to the other projects in the solution.  Not being an EXPERT VB guy, I saw the example project that used .NET4.7.2 and decided to try it.  

Here's the code:
        public static void Run()
        {
            Console.WriteLine($"\r\n*** ExampleRW ***");

            const int TIMEOUT = 5000;

            //DINT Test Read/Write
            var myTag = new Tag<DintPlcMapper, int>()
            {
                //Name of tag on the PLC, Controller-scoped would be just "SomeDINT"
                Name = "MyTESTDint",
                
                //PLC IP Address
                Gateway = "172.16.1.2",

                //CIP path to PLC CPU. "1,0" will be used for most AB PLCs
                Path = "1,0", 

                //Type of PLC
                PlcType = PlcType.ControlLogix,

                //Protocol
                Protocol = Protocol.ab_eip,

                //A global timeout value that is used for Initialize/Read/Write methods
                Timeout = TimeSpan.FromMilliseconds(TIMEOUT), 
            };
            myTag.Initialize();

            //Read tag value - This pulls the value from the PLC into the local Tag value
            Console.WriteLine($"Starting tag read");
            myTag.Read();

            //Read back value from local memory
            int myDint = myTag.Value;
            Console.WriteLine($"Initial Value: {myDint}");

            //Set Tag Value
            myDint++;
            myTag.Value = myDint;

            Console.WriteLine($"Starting tag write ({myDint})");
            myTag.Write();

            //Read tag value - This pulls the value from the PLC into the local Tag value
            Console.WriteLine($"Starting synchronous tag read");
            myTag.Read();

            //Read back value from local memory
            var myDintReadBack = myTag.Value;
            Console.WriteLine($"Final Value: {myDintReadBack}");

For the life of me, I cant get it to work.  The code is exactly the same as the .NET Core example on .NET 5, but in the framewok example project.  Every time I run it I get the following when executing the initialize function:

System.IO.FileNotFoundException  HResult=0x80070002
  Message=Could not load file or assembly 'libplctag.NativeImport, Version=1.0.17.0, Culture=neutral, PublicKeyToken=null' or one of its dependencies. The system cannot find the file specified.
  Source=libplctag StackTrace:
   at libplctag.NativeTag.plc_tag_create(String lpString, Int32 timeout) in C:\Users\jbart\JDB Team Dropbox\JDB Team Data\Digi\AB PLC AGC\NEW VB HMI\libplctag.NET-master\src\libplctag\NativeTag.cs:line 12
   at libplctag.NativeTagWrapper.Initialize() in C:\Users\jbart\JDB Team Dropbox\JDB Team Data\Digi\AB PLC AGC\NEW VB HMI\libplctag.NET-master\src\libplctag\NativeTagWrapper.cs:line 184
   at libplctag.Tag.Initialize() in C:\Users\jbart\JDB Team Dropbox\JDB Team Data\Digi\AB PLC AGC\NEW VB HMI\libplctag.NET-master\src\libplctag\Tag.cs:line 146
   at libplctag.Tag`2.Initialize() in C:\Users\jbart\JDB Team Dropbox\JDB Team Data\Digi\AB PLC AGC\NEW VB HMI\libplctag.NET-master\src\libplctag\TagOfT.cs:line 102
   at CSharpDotNetFramework.ExampleRW.Run() in C:\Users\jbart\JDB Team Dropbox\JDB Team Data\Digi\AB PLC AGC\NEW VB HMI\libplctag.NET-master\src\Examples\CSharp DotNetFramework\ExampleRW.cs:line 38
   at CSharpDotNetFramework.Program.Main(String[] args) in C:\Users\jbart\JDB Team Dropbox\JDB Team Data\Digi\AB PLC AGC\NEW VB HMI\libplctag.NET-master\src\Examples\CSharp DotNetFramework\Program.cs:line 9

The other example projects work fine, but I REALLY need the one that runs on .NET4 Framework.

I'm SURE it's something on my laptop, just can't figure out what!!  Any help is appreciated.

John

kyle-github, 2022-03-05:
Not to deflect here, but Jody and Tim support the .Net wrapper over on the [libplctag.NET issues](https://github.com/libplctag/libplctag.NET/issues) page.

This issues page is for the core C library and I am not a .Net programmer.   Your question sounds like one that they can probably help with much better than I can!

BartGuy1215, 2022-03-06:
Thanks Kyle.  I'll move it over there.

I've written a ton of .net programs bot don't feel that I'm on your level is where I was going with that.

John

closed 2022-03-06 by BartGuy1215